micropython/micropython #7302: mbedtls compilation error with GCC >= 11: 'mbedtls_sha512_finish_ret' accessing 64 bytes in a region of size 48

SpotlightKid, 2021-05-24:
I enabled `MICROPY_MBED_TLS` in my board definition and when my distro (Manjaro) updated to GCC 11.x, I now get this error when compiling the firmware:

```
CC ../../lib/lwip/src/core/tcp_in.c
../../lib/mbedtls/library/ssl_tls.c: In function 'ssl_calc_finished_tls_sha384':
../../lib/mbedtls/library/ssl_tls.c:6408:5: error: 'mbedtls_sha512_finish_ret' accessing 64 bytes in a region of size 48 [-Werror=stringop-overflow=]
 6408 |     finish_sha384( &sha512, padbuf );
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../lib/mbedtls/library/ssl_tls.c:6408:5: note: referencing argument 2 of type 'unsigned char *'
In file included from ../../lib/mbedtls/include/mbedtls/ssl_internal.h:74,
                 from ../../lib/mbedtls/library/ssl_tls.c:73:
../../lib/mbedtls/include/mbedtls/sha512.h:165:5: note: in a call to function 'mbedtls_sha512_finish_ret'
  165 | int mbedtls_sha512_finish_ret( mbedtls_sha512_context *ctx,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
```

I tried updating the `lib/mbedtls` submodule to the HEAD of the `development` branch, but then I get this error:

```
make: *** No rule to make target 'lib/mbedtls/library/certs.c', needed by 'build-VCC_GND_F407VE/genhdr/qstr.i.last'.  Stop.
```

Related upstream issue: https://github.com/ARMmbed/mbedtls/issues/3782
And merged fix: https://github.com/ARMmbed/mbedtls/pull/3848

SpotlightKid, 2021-05-24:
Workaround, of course is:

    CFLAGS="-Wno-stringop-overflow" make BOARD=...

dpgeorge, 2021-05-24:
Thanks for the report.  We are on the v2.16.x branch (do `git checkout mbedtls-2.16`) and #7303 updates to the latest commit on that.  I don't yet have gcc11 so couldn't test it, but it at least builds with gcc10.

SpotlightKid, 2021-05-24:
Yes, I can confirm that the build now succeeds again with #7303 and GCC 11.1.0 and without `-Wno-stringop-overflow`.

dpgeorge, 2021-05-25:
Fixed by 0aa01b0205adc7544315fed602029e792a235b2e

closed 2021-05-25 by dpgeorge