micropython/micropython #7304: `tuple + class MyTuple(tuple)` performs a wrong type cast

dimpolo, 2021-05-24:
```python
class MyTuple(tuple):
    pass

print((1,2) + MyTuple((3,4)))  # prints (1, 2, (nil))
```

Relevant code:
https://github.com/micropython/micropython/blob/3b950ed2959c603ff30dd6052c157e7981c4d2d6/py/objtuple.c#L145-L153
`mp_obj_tuple_binary_op` checks rhs with `mp_obj_is_subclass_fast` and then casts it to `mp_obj_tuple_t`

peterhinch, 2021-05-25:
Subclassing built-in types is not fully supported: see [differences from CPython](https://github.com/micropython/micropython/wiki/Differences).

dimpolo, 2021-05-25:
As far as I can tell, the pointer cast and subsequent memcopy (`mp_seq_cat`) are undefined behaviour.

dimpolo, 2021-05-25:
```python
class MyTuple(tuple):
    pass

t = MyTuple((3,4))
t.field = 1
print((1,2) + t)  # crashes
```

dimpolo, 2023-09-28:
This still reproduces.
See https://wokwi.com/projects/377113575387211777 for a `IllegalInstruction` crash

peterhinch, 2023-09-28:
See [differences doc](https://github.com/micropython/micropython/wiki/Differences):
> Subclassing of builtin types is partially implemented, but there may be various differences and compatibility issues with CPython. [#401](https://github.com/micropython/micropython/issues/401)

jepler, 2026-04-05:
I was curious about the situation with lists. It's rejected with a type error...

```
>>> class L(list): pass
... 
>>> [1,2] + L((3,4))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __add__: 'list', 'list'
```

closed 2026-04-15 by dpgeorge