micropython/micropython #7328: ports/unix: compile error on macOS 11.4-x86_64

k-ronny, 2021-05-28:
Hello,
I get this error message:
```
CC mpthreadport.c
mpthreadport.c:129:94: error: cast to smaller integer type 'int' from 'pthread_t' (aka 'struct _opaque_pthread_t *') [-Werror,-Wpointer-to-int-cast]
    snprintf(thread_signal_done_name, sizeof(thread_signal_done_name), "micropython_sem_%d", (int)thread->id);
                                                                                             ^~~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/secure/_stdio.h:57:62: note: expanded from macro 'snprintf'
  __builtin___snprintf_chk (str, len, 0, __darwin_obsz(str), __VA_ARGS__)
                                                             ^~~~~~~~~~~
1 error generated.
```
... and got it to work with this patch:
```
diff --git a/ports/unix/mpthreadport.c b/ports/unix/mpthreadport.c
index cbc4901f6..14a0307ae 100644
--- a/ports/unix/mpthreadport.c
+++ b/ports/unix/mpthreadport.c
@@ -126,7 +126,7 @@ void mp_thread_init(void) {
     thread->next = NULL;
 
     #if defined(__APPLE__)
-    snprintf(thread_signal_done_name, sizeof(thread_signal_done_name), "micropython_sem_%d", (int)thread->id);
+    snprintf(thread_signal_done_name, sizeof(thread_signal_done_name), "micropython_sem_%ld", (intptr_t)thread->id);
     thread_signal_done_p = sem_open(thread_signal_done_name, O_CREAT | O_EXCL, 0666, 0);
     #else
     sem_init(&thread_signal_done, 0, 0);
```
Is that the right approach?

dlech, 2021-05-28:
I don't thing this would work on all OSes since the definition of `pthread_t` and `intptr_t` is OS dependent. Maybe the `id` field could be defined as a union of `int` and `pthread_t`? Or do the cast to int like this: `*(int *)&thread->id`?

k-ronny, 2021-05-28:
oh, I thought it might work that way because it is already enclosed by `#if defined(__APPLE__)`.

dlech, 2021-05-28:
I missed that part. If we don't care about 32-bit builds on Mac anymore, I guess it is OK.

closed 2021-06-13 by k-ronny

dpgeorge, 2021-06-15:
Should be fixed by 95048129b1d93854c25f501c02801929aeeb23f0

dpgeorge, 2021-06-15:
See #7393.