micropython/micropython #7330: Test suite produces unexpected output

peterhinch, 2021-05-29:
Running this against a Pyboard 1.1 produced this output which struck me as odd. The Pyboard is running firmware built from today;s source.
```bash
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/tests
$ ./run-tests.py --device /dev/ttyUSB0
skip  basics/class_inplace_op2.py
pass  basics/builtin_abs_intbig.py
pass  basics/builtin_id.py

[SNIP]

pass  basics/int_big_mod.py
pass  basics/string_partition.py
basics/is_isnot.py:1: SyntaxWarning: "is" with a literal. Did you mean "=="?
  print(1 is 1)
basics/is_isnot.py:2: SyntaxWarning: "is" with a literal. Did you mean "=="?
  print(1 is 2)
basics/is_isnot.py:3: SyntaxWarning: "is not" with a literal. Did you mean "!="?
pass  basics/self_type_check.py
  print(1 is not 1)
basics/is_isnot.py:4: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  print(1 is not 2)
pass  basics/slice_intbig.py
pass  basics/is_isnot.py

[SNIP]

pass  basics/int_constfolding.py
pass  basics/int2.py
basics/op_error.py:33: SyntaxWarning: 'int' object is not subscriptable; perhaps you missed a comma?
  1[0]
basics/op_error.py:41: SyntaxWarning: str indices must be integers or slices, not str; perhaps you missed a comma?
  ''['']
basics/op_error.py:55: SyntaxWarning: 'int' object is not callable; perhaps you missed a comma?
  1()
pass  basics/subclass_native2_tuple.py
pass  basics/set_iter_of_iter.py

[SNIP]

skip  extmod/vfs_userfs.py
798 tests performed (20968 individual testcases)
798 tests passed
57 tests skipped: builtin_help builtin_next_arg2 builtin_range_binop class_inplace_op2 cmd_parsetree extra_coverage framebuf1 framebuf16 framebuf2 framebuf4 framebuf8 framebuf_subclass heap_locked import_mpy_invalid import_mpy_native_gc import_mpy_native_x64 io_buffered_writer machine_timer math_factorial_intbig memoryview_itemsize namedtuple_asdict repl_words_move resource_stream schedule sys_getsizeof sys_settrace_features sys_settrace_generator sys_settrace_loop uasyncio_threadsafeflag ucryptolib_aes128_ctr urandom_extra urandom_extra_float urandom_seed_default ure_debug ure_groups ure_span ure_sub ure_sub_unmatched vfs_basic vfs_blockdev vfs_fat_fileio1 vfs_fat_fileio2 vfs_fat_finaliser vfs_fat_more vfs_fat_mtime vfs_fat_oldproto vfs_fat_ramdisk vfs_fat_ramdisklarge vfs_lfs vfs_lfs_corrupt vfs_lfs_error vfs_lfs_file vfs_lfs_mount vfs_lfs_mtime vfs_lfs_superblock vfs_posix vfs_userfs
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython/tests
```

stinos, 2021-05-29:
It has been like this for a while now, at least since CPython 3.8

dlech, 2021-05-29:
Should we drop those tests since they "only work by accident" (according to https://bugs.python.org/issue34850)?

dpgeorge, 2021-05-29:
These are mostly for coverage.  I'd suggest that the tests which produce such warnings be moved to separate tests with .exp files provided.

dpgeorge, 2021-05-29:
See #7332 for a fix.

dpgeorge, 2021-05-30:
Fixed by 025e4b6fbc32cd3118a5519cf6a18aa04b045abe

closed 2021-05-30 by dpgeorge