micropython/micropython #7334: Could ujson.loads be improved faster?

bhcuong2008, 2021-05-30:
Hi,

This is just my question to improve ujson load time, not an issue.
I test ujson.loads, then found it takes about 2-3ms in my cases. Here are my tests. It seems to be load time depending on input length rather than nested dictionary. Could it be improved under 1ms?

```
>>> 
>>> d
'{"a": 2990, "b": {"c": 2, "c2": [[10,100], 1]}, "d": ["OFF", {"foo": "ON", "foo2": "OFF"}]}'
>>> test_json(d)
dt = 2.593ms
{'a': 2990, 'd': ['OFF', {'foo2': 'OFF', 'foo': 'ON'}], 'b': {'c': 2, 'c2': [[10, 100], 1]}}

>>> 
>>> d2
'{"battery":93,"battery_low":false,"contact":false,"linkquality":127,"tamper":false,"voltage":2900}'
>>> test_json(d2)
dt = 2.769ms
{'linkquality': 127, 'contact': False, 'battery': 93, 'battery_low': False, 'tamper': False, 'voltage': 2900}
>>>
```

Thank you.

stinos, 2021-05-30:
> This is just my question to improve ujson load time, not an issue.

The best place for questions is forum.micropython.org, not the issue tracker.

bhcuong2008, 2021-05-30:
Sorry I dont know it.
Thank @stinos

closed 2021-05-30 by bhcuong2008