micropython/micropython #7338: Syntax error with * in function argument list

peterhinch, 2021-05-31:
This works as expected under CPython but throws a syntax error in MicroPython (RP2 and Unix build).
```python
def bar():
    return 1, 2

class Foo:
    def __init__(self, x, y, z):
        print(x, y, z)

foo = Foo(*bar(), 3)
```
It seems that the * is interpreted as the specifier for kwonly args. Outcome:
```
Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
SyntaxError: non-keyword arg after */**
```
A further observation. When I pasted this code in an `mpremote` session, a few seconds after the syntax error was reported `mpremote` fell over. Note that `mpremote` is as installed with pip3. Here is the complete session.
```
$ mpremote mount .
Local directory . is mounted at /remote
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] to exit this shell
>
MicroPython v1.15 on 2021-04-29; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== Traceback (most recent call last):
===   File "<stdin>", line 8, in <module>
=== SyntaxError: non-keyword arg after */**
=== 
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> Traceback (most recent call last):
  File "/home/adminpete/.local/bin/mpremote", line 8, in <module>
    sys.exit(main())
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 1037, in main
    do_repl(pyb, args)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 934, in do_repl
    do_repl_main_loop(pyb, console, console_out_write, file_to_inject)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 874, in do_repl_main_loop
    pyb.soft_reset_with_mount(console_out_write)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 644, in soft_reset_with_mount
    self.exec_(fs_hook_code)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 458, in exec_
    raise PyboardError("exception", ret, ret_err)
mpremote.pyboard.PyboardError: ('exception', b'\x18\x01\x03\x00\x00\x00/os\x18\x01\x06\x00\x00\x00/os.py\x18\x01\x07\x00\x00\x00/os.mpy\x18\x01\x03\x00\x00\x00/io\x18\x01\x06\x00\x00\x00/io.py\x18\x01\x07\x00\x00\x00/io.mpy\x18\x01', b'\x00\x00\x00/sys\x18\x01\x07\x00\x00\x00/sys.py\x18\x01\x08\x00\x00\x00/sys.mpy')
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython-micro-gui
```

stinos, 2021-05-31:
*It seems that the * is interpreted as the specifier for kwonly args*

It's actually just that MicroPython doesn't support normal arguments after unpacking other arguments (bey it keyword or not), ran into this many times already,always assuming it was documented behavior :) I.e. should be mentioned in the differences but isn't, as far as I can tell. See https://github.com/micropython/micropython/commit/1e70fda69fcb4991eb60ed43e610f664ea1319e6

peterhinch, 2021-05-31:
MicroPython seems to support subsequent keyword args, but barfs on positional ones. This example is of working code:
```python
Button(writer, *self.locn(row, col), height = buttonheight, width = buttonwidth,
       textcolor = BLACK, bgcolor = color,
       text = text, shape = RECTANGLE,
       callback = back, args = (text,))
```

dlech, 2021-05-31:
This is one of the Python 3.5 features (namely PEP 448) that hasn't been implemented yet. See #1329 and #5807.

dlech, 2021-05-31:
So technically it documented:

> MicroPython implements Python 3.4 and some select features of Python 3.5.

But of course it would be better if it was specifically called out. 😉

peterhinch, 2021-05-31:
OK, thanks both. I'd forgotten the PEP448 issue.

The behaviour of `mpremote` remains outstanding.

closed 2021-05-31 by peterhinch