micropython/micropython #7342: mimxrt: Time sleep functions stall.

robert-hh, 2021-06-01:
With the new master branch for mimxrt, all timer sleep functions stall. That did not happen before. When interrupting the code in the debugger, it seems that the GPT times used for the timer ticks is not running.

dpgeorge, 2021-06-01:
I confirm this.  It looks like it's commit 4c407c790fd8c9efb89f91ab4805fc459970a833 that broke it, but not sure why.

robert-hh, 2021-06-02:
OK. I made no attempt yet to understand the clock machine of these devices. It looks very extensive.

dpgeorge, 2021-06-02:
Yeah, it's a pretty complicated chip.  I'd suggest removing that line and then adjusting the clock divider for the PIT (the following line in the code) so that the PIT runs at the required frequency.

robert-hh, 2021-06-02:
Setting the divider to 5 causes utime.sleep() times to be increased by 25. It seems that the Gpt also depends on that clock source. So the only way would be adjusting the timer delay times.

robert-hh, 2021-06-02:
Got it. I had to change the clock source for the PIT us calculation to the IPG clock. Then both work fine.

dpgeorge, 2021-06-03:
Fixed by d79105d7c0f68e65ff175b752d9895241e355c08

closed 2021-06-03 by dpgeorge