micropython/micropython #7346: stm32: eth.c wrong registers used in eth_link_status.

iabdalkader, 2021-06-02:
https://github.com/micropython/micropython/blob/4c407c790fd8c9efb89f91ab4805fc459970a833/ports/stm32/eth.c#L783

Hello, this doesn't look right at all, the link status bit is in BSR register (address 1), while registers 0 and 0x10 are the BCR (Basic Control Register) and the NLP/CROSSOVER TIMEREGISTER respectively, which don't have any relevant info, see the LAN8742 datasheet. I think it should be something like this instead:
```
int eth_link_status(eth_t *self) {
...
    } else {
        if (!(eth_phy_read(1) & PHY_BSR_LINK_STATUS)) {
            return 0; // link down
        } else {
            return 1; // link up
        }
    }
}
```
I tested the above code (after initializing low level pins, clocks etc... and before calling active()) and I can detect if the cable is connected or disconnected I get 1 or 0 Vs the old code which always returns 1 .. Is there any reason why it's not implemented like that ?

Ref:
[image]

dpgeorge, 2021-06-05:
> Is there any reason why it's not implemented like that ?

I'm not sure... most of my testing of the initial ETH code was on an STM32F769DISC.  I'll need to go and retest it.  But your code does look much better than what's there...

iabdalkader, 2021-06-05:
>  I'll need to go and retest it.

To test this it needs to be called before active(True) with some basic initialization done first (configure MDC, MDIO, REF clock GPIOs and enable ETH1MAC clock) otherwise it takes the other path if it's called after active(True) because it sets NETIF flags and I'm not sure how to clear these flags with no way to bring the interface down. Actually this is something else I wanted to discuss, maybe some of the basic initialization (GPIOs, clocks) should be moved to a separate function so functions like link_status (and possibly enter/exit low-power mode) can work before calling active(True). So for example you can check if the cable is connected or not without blocking on active(True) or switch eth to low-power mode in early board init functions, but to read/write registers the minimal initilization above is needed first.

iabdalkader, 2021-06-06:
@dpgeorge I sent a PR which fixes this for testing and also adds low power mode.. It's easier to test link_status() with that PR because low_power must do some basic initialization to read/write registers, so all you need to do is:

```Python
lan.config(low_power=False) # This also does some basic initialization which allows link_status to work without active(True).
print(lan.status())
lan.config(low_power=True)
```
With the cable connected/disconnected.

commit 66115a37 (2021-06-13): stm32/eth: Fix eth_link_status function to use correct BSR bit.

Fixes #7346.
closed 2021-06-13 by dpgeorge