micropython/micropython #7352: micropython 1.15 esp32 reboot at SD deinit

emard, 2021-06-03:
After unmounting SD card I need to release it's pins (to input or hi-z)
for this I want to deinit SD card and set pins to input, but at deinit
ESP32 reboots

```
>>> import os
>>> from machine import SDCard, Pin
>>> import ecp5
>>> sd = SDCard(slot=3)
>>> os.mount(sd,"/sd")
>>> os.listdir("/sd")
['KERNEL.SYS', 'COMMAND.COM', 'INVADERS.COM', 'CTMOUSE.EXE', 'kick.rom', 'OSD_CA01.sys', 'amiga', 'colecovision', 'hdl4fpga', 'linux', 'm68k', 'mp3', 'msx', 'nes', 'next186', 'orao', 'ql', 'scope', 'sega_sg1000', 'rootfs.cpio.uboot', 'uImage', 'uboot.env.defunct', 'dtb', 'slides', 'ti99_2', 'ti99_4a', 'vic20', 'zxspectrum', 'snes', 'rtc', 'myamoto', 'apple2', 'speak', 'c64']
>>> os.umount("/sd")
>>> sd.deinit()
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4016412d  PS      : 0x00060530  A0      : 0x801620e6  A1      : 0x3ffccf60  
A2      : 0x0e32f602  A3      : 0x3ffc43d0  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffcd110  A7      : 0x3ffcd130  A8      : 0x3ffcd110  A9      : 0x3ffcd070  
A10     : 0x3f81ad81  A11     : 0x3ffcd128  A12     : 0x00000004  A13     : 0xffffff29  
A14     : 0x3f81ad82  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0e32f606  LBEG    : 0x40092b1e  LEND    : 0x40092b29  LCOUNT  : 0x00000000  

Backtrace:0x4016412a:0x3ffccf60 0x401620e3:0x3ffccf90 0x4016238b:0x3ffccfd0 0x401d244a:0x3ffccff0 0x400dbf46:0x3ffcd010 0x400e25e9:0x3ffcd030 0x400e273a:0x3ffcd050 0x400e4edd:0x3ffcd070 0x400dc0a8:0x3ffcd110 0x400e25e9:0x3ffcd160 0x400e2612:0x3ffcd180 0x400f5afe:0x3ffcd1a0 0x400f5e55:0x3ffcd230 0x400d5c38:0x3ffcd270

ELF file SHA256: e47b1b0d0f482c6e

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:5636
load:0x40078000,len:13136
load:0x40080400,len:4356
entry 0x400806b0
Started webrepl in normal mode
ssid: ra chan: 1 rssi: -40 authmode: WPA2-PSK
MicroPython v1.15 on 2021-04-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
```

dpgeorge, 2021-06-17:
Thanks for the report.  I can reproduce the problem, and created #7412 to fix it.

dpgeorge, 2021-06-18:
Fixed by 4eb13c50cd87eff8374b9bffc3d7c15b1bf5e35f

closed 2021-06-18 by dpgeorge

emard, 2021-06-18:
Great, it doesn't crash after umount and deinit, but it won't mounting it again?

E (513301) spi: spi_bus_initialize(462): SPI bus already initialized.
E (513301) sdspi_host: spi_bus_initialize failed with rc=0x103