micropython/micropython #7361: REPL not working

kdschlosser, 2021-06-07:
I have an ESP32 that I loaded MicroPython onto. What I have is a LilyGo TTGO T7 and this is the binary I loaded esp32spiram-idf3-20210202-v1.14.bin onto it.

I erased the flash first without any problems and then I used this command to load the firmware

```esptool.py --chip esp32 --port COM6 --baud 500000 write_flash -z 0x1000 esp32spiram-idf3-20210202-v1.14.bin```

This board has an additional 8mb of ram.

When I connect to the USP32 using putty this is the output I get
```
ets J]
      ▒▒▒▒▒ 00:2&▒▒▒H▒HH▒▒▒0xL▒POWE▒▒▒%UMQ),boo▒▒x13 ▒*▒}MQ_FLAS▒%==Q▒
confi֥▒▒ 0       ▒A%]A▒0xee▒▒,k▒E▒v:0x0bq_dr▒▒x00,▒E▒▒▒0▒0,cs▒▒▒▒0A0,hd_▒▒▒0x▒▒wp_d.▒▒▒▒0C▒둕▒DIO, c▒k▒▒di▒▒
▒▒+▒▒0Mfff0▒L,▒▒▒▒5008
▒
 ▒▒▒▒l 12 r▒
            ▒j
              load:0▒00780▒▒len:LL▒j
                                    ho 0 tXZ
                                            ▒▒▒▒om 4▒!녑▒0x4008▒00,lYK▒▒▒▒j
                                                                           ent▒▒0x4▒806b▒H▒MicroPython v1.14 on 2021-02-02; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>>
```

I am unable to type any commands. I am not sure what is going on with it because the firmware loads with no errors.

Any assistance is really appreciated.

dpgeorge, 2021-06-08:
I would say it is a hardware error, or maybe the UART for the REPL is connected to different pins.

kdschlosser, 2021-06-09:
Its something with the encoding. If I restart putt then it works. Something with the Ansi escape characters. It started behaving nicely for a bit and when O powered it down went back to not working right again. I have other ESP32's running MicroPython and they work properly I am using a different version of MicroPython on those units because they do not have the additional memory.

The other thing I noticed is that I only have 4mb of the 8mb of ram the esp comes with.

pumelo, 2021-08-21:
> The other thing I noticed is that I only have 4mb of the 8mb of ram the esp comes with.

This already came up several times:
https://github.com/micropython/micropython/issues/5428
https://github.com/micropython/micropython/issues/6460

Citing @dpgeorge from #5428:
> There's only enough address space to map up to 32Mbit. 64MBit is only supported using a special "himem API" and bank switching.

closed 2025-03-03 by kdschlosser