micropython/micropython #7362: support for UART rx interrupt

embhobbb, 2021-06-07:
I am using raspberry pi pico for my new project.
I was checking for example for UART receive interrupt. Do we have support for the same?
How can we achieve the UART RX interrupt?
I checked the micropython documentation but I didn't see support for RPi pico.

robert-hh, 2021-06-07:
If you ask for something like a callback handler on UART receive: that is not implemented. As far as I know, not for any of the ports. Received data goes into a ring buffer of configurable size, and you can request the amount of data with uart.any(). Implementing a callback is surely possible and not even complicated, but it has to be considered on how. It should be configurable, like at what buffer level it should be called.

dpgeorge, 2021-06-07:
> As far as I know, not for any of the ports.

stm32 does support a UART RX callback, and it's called when the RX line is "idle" (as defined by the hardware itself, when there's been a gap in received data).

robert-hh, 2021-06-07:
Ahh. I looked into the IMXRT Manuals and I see it there too. The receiver considers to be idle after a full character time with no activity after having received data. That may in deed be useful to get a notice at the end of  message.