micropython/micropython #7363: Pin implicit In doesn't work on stm32f4

TecDroiD, 2021-06-07:
Hi,
i compiled mp for stm32f407. After powering the system I wanted to read an input which didn't work (just connected the pin with 3.3V) and read value 0. When I explicitely set the pin to Pin.IN, it works.
`Type "help()" for more information.
>>> pin = pyb.Pin('IN1')
>>> pin
Pin(Pin.cpu.E9, mode=Pin.IN)
>>> pin.value()
0
>>> pin = pyb.Pin('IN1',pyb.Pin.IN)
>>> pin
Pin(Pin.cpu.E9, mode=Pin.IN)
>>> pin.value()
1
` 
I'm basically using a copy of STM32F4DISC with some additional Pins defined. 
Maybe this could also be the reason why my sdcard isn't found..

TecDroiD, 2021-06-07:
just found out that this only happens for the first time I create a Pin on Port E. After, all input pins with implicitely set mode work as input.. At least, Port A works without problems.

MPY: can't mount SD card
MicroPython v1.15-181-g2f365d234 on 2021-06-07; MILLISCAN with STM32F407
Type "help()" for more information.
>>> pin = pyb.Pin('C8')
>>> pin.value()
1
>>> pin = pyb.Pin('C8',pyb.Pin.IN)
>>> pin.value()
0

dpgeorge, 2021-06-08:
By default on the stm32 port, constructing a pin object without any arguments (just the pin name) will not do any initialisation on that pin.  This is useful, eg to find out the current state of a pin without changing it.

If you want a pin to be an input pin, then it needs to be explicitly constructed like:
```python
from machine import Pin
pin = Pin('A0', Pin.IN)
```

TecDroiD, 2021-06-08:
I basically understand what you mean but

- i only need to use Pin.IN on the first Pin on port E. After that, all pins are automatically valid inputs
- stm32 initializes all pins as floating inputs on reset. [(STM32F405 reference manual](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwich4qbqofxAhUFdBQKHffkB7QQFjACegQIBhAE&url=https%3A%2F%2Fwww.st.com%2Fresource%2Fen%2Freference_manual%2Fdm00031020-stm32f405-415-stm32f407-417-stm32f427-437-and-stm32f429-439-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf&usg=AOvVaw2x8tbTRz8d9PfqXBk3qZ74) , 8.3.1, p269)

dpgeorge, 2021-06-09:
Ah, the issue might simply be that the GPIOE peripheral is not automatically turned on at start up, but ports A-D are.

The fix for this would be one of:
1. call `__HAL_RCC_GPIOE_CLK_ENABLE()` unconditionally at start up
2. call it only when a pin on that port is constructed

I think option (2) is better because it scales to other pins (ports F, G etc) and doesn't use power if the port is never used.

dpgeorge, 2021-09-02:
Fixed by 05cd17e36fbc5d3606e34e430dc72549bd1d6708

closed 2021-09-02 by dpgeorge