micropython/micropython #7366: build error with thread enable

xhlinux, 2021-06-07:
when enable

```
#define MICROPY_PY_THREAD                 (1)
#define MICROPY_PY_THREAD_GIL             (1)
```
and I got this error：
```
../../py/mpthread.h:38:26: fatal error: mpthreadport.h: No such file or directory
 #include <mpthreadport.h>
                          ^
compilation terminated.
```

stinos, 2021-06-07:
Not all ports implement threading. Only ports which have mpthreadport.h in their directory. So unless soemthing is wrong with the build system: are you using a port which doesn't support threading perhaps?

xhlinux, 2021-06-07:
> Not all ports implement threading. Only ports which have mpthreadport.h in their directory. So unless soemthing is wrong with the build system: are you using a port which doesn't support threading perhaps?

use port: esp8266

stinos, 2021-06-07:
See duplicate: https://github.com/micropython/micropython/issues/3667 and info: https://forum.micropython.org/viewtopic.php?f=2&t=5920&p=56084&hilit=esp8266+threading#p56084

xhlinux, 2021-06-07:
> See duplicate: #3667 and info: https://forum.micropython.org/viewtopic.php?f=2&t=5920&p=56084&hilit=esp8266+threading#p56084

thanks for u help.

closed 2021-06-07 by xhlinux