micropython/micropython #7374: ESP32 Pin.PULL_UP does not work on all pins

peterhinch, 2021-06-09:
On the reference board, specifying pins that are broken out, `PULL_UP` works as expected.

However the TTGO T-Display brings out pins 36-39 (and others). These pins work as inputs, but the `PULL_UP` specifier has no effect: pins are floating. This issue can also be demonstrated on the reference board.

Is this a MicroPython issue or a hardware limitation?

robert-hh, 2021-06-09:
That's a limitation of the ESP32.

peterhinch, 2021-06-09:
OK, thanks for clarifying.

closed 2021-06-09 by peterhinch

IhorNehrutsa, 2021-06-09:
This fragment from https://randomnerdtutorials.com/esp32-pinout-reference-gpios/
```
GPIOs 34 to 39 are GPIs – input-only pins. 
These pins don’t have internal pull-ups or pull-down resistors. 
They can’t be used as outputs, so use these pins only as inputs:
GPIO 34
GPIO 35
GPIO 36
GPIO 39
```