micropython/micropython #7375: How do you make webrepl automatically listen to socket requests in the background without blocking?

Li-Lian1069, 2021-06-09:
你是怎么做到让 webrepl 在后台自动监听socket请求而不阻塞进程的?
How do you make webrepl automatically listen to socket requests in the background without blocking?

我注意到你使用了 setsockopt(socket.SOL_SOCKET, 20, callback) 这种方式来定义一个回调函数, 但它是如何工作的? 原理是什么, 常量20的含义是什么?使用你这种方法是否会打断目前正在进行的其他任务?
I noticed that you used `setsockopt(socket.SOL_SOCKET, 20, callback)`  to define a callback function, but how does it work? What is the principle and meaning of the constant 20? Will using your method interrupt other tasks currently in progress when a request come on?

或者, 是否有更好的方式在 socket 需要有请求被 accept 的时候才调用 accept 函数 而不阻塞主线程的方法?
Or, is there a better way to call the accept function when the socket have a request to be accepted without blocking the main thread?

这个问题困扰了我很久, 而且我目前急需这个解决方案 ,非常希望您能够回答我
This problem has troubled me for a long time, and I urgently need this solution at the moment, I hope you can answer me very much.

Thank you very much! I am grateful for you !

robert-hh, 2021-06-10:
`setsockopt(socket.SOL_SOCKET, 20, callback)` is a special function defined by the MP team to call a callback function upon events at the socket. The number 20 was chosen to identify the option, because it was not yet in use. I use that callback in the ftp server to accept and serve incoming requests. That interrupts other tasks in progress. And being a scheduled callback, there is no limit on actions you may take, besides time and memory. And obviously that action should not block. You may look at that as an example: https://github.com/robert-hh/FTP-Server-for-ESP8266-ESP32-and-PYBD

Li-Lian1069, 2021-06-10:
Thank you very much!

closed 2021-06-10 by Li-Lian1069