micropython/micropython #7376: py: micropython.schedule is unsafe on Signal Handler

amirgon, 2021-06-09:
I'm seeing this on the unix port when Micropython code is invoked in a signal handler, but this might apply also to Interrupt Handlers on other platforms.

Under signal/interrupt handler [it's illegal to allocate memory](https://docs.micropython.org/en/latest/reference/isr_rules.html#:~:text=avoid%20memory%20allocation%3A%20no%20appending%20to%20lists%20or%20insertion%20into%20dictionaries%2C%20no%20floating%20point.) but it's allowed to call `micropython.schedule` to schedule a function that would allocate memory.

However, sometimes `micropython.schedule` directly calls the function on the same context (a tail call?) instead of scheduling it, causing a deadlock.

Here is a stack trace that shows how this happens:
```
#0  __lll_lock_wait () at ../sysdeps/unix/sysv/linux/x86_64/lowlevellock.S:135
#1  0x00007ffff7bc3dbd in __GI___pthread_mutex_lock (mutex=0xab9df8 <mp_state_ctx+2520>) at ../nptl/pthread_mutex_lock.c:80
#2  0x000000000064a017 in mp_thread_mutex_lock (mutex=0xab9df8 <mp_state_ctx+2520>, wait=1) at mpthreadport.c:292
#3  0x000000000040544a in gc_alloc (n_bytes=<optimized out>, alloc_flags=0) at ../../py/gc.c:448
#4  0x0000000000404afa in m_malloc (num_bytes=16) at ../../py/malloc.c:86
#5  0x00000000006cdf9b in lv_to_mp_struct (type=0x847d80 <mp_lv_anim_t_type>, lv_struct=0x7ffff1470880) at build-dev/lvgl/lv_mpy.c:385
#6  0x00000000006d7cd6 in mp_read_ptr_lv_anim_t (field=0x7ffff1470880) at build-dev/lvgl/lv_mpy.c:4641
#7  0x00000000006f1e0b in lv_anim_t_path_cb_callback (arg0=0x7ffff1470880) at build-dev/lvgl/lv_mpy.c:21647
#8  0x000000000066cd7f in anim_timer (param=0x7ffff13a4280) at ../../lib/lv_bindings/lvgl/src/misc/lv_anim.c:363
#9  0x0000000000668b17 in lv_timer_exec (timer=0x7ffff13a4280) at ../../lib/lv_bindings/lvgl/src/misc/lv_timer.c:306
#10 0x00000000006686fd in lv_timer_handler () at ../../lib/lv_bindings/lvgl/src/misc/lv_timer.c:97
#11 0x00000000006c5f1a in lv_task_handler () at ../../lib/lv_bindings/../../lib/lv_bindings/lvgl/src/lv_api_map.h:36
#12 0x00000000006f544d in mp_lv_task_handler (mp_n_args=0, mp_args=0x7fffffffbf18, lv_func_ptr=0x6c5f11 <lv_task_handler>) at build-dev/lvgl/lv_mpy.c:25949
#13 0x00000000006c6303 in lv_fun_builtin_var_call (self_in=0x8590e0 <mp_lv_task_handler_obj>, n_args=0, n_kw=0, args=0x7fffffffbf18) at build-dev/lvgl/lv_mpy.c:94
#14 0x00000000004a79a9 in mp_call_function_n_kw (fun_in=0x8590e0 <mp_lv_task_handler_obj>, n_args=0, n_kw=0, args=0x7fffffffbf18) at ../../py/runtime.c:652
#15 0x00000000004a9731 in mp_call_method_n_kw (n_args=0, n_kw=0, args=0x7fffffffbf08) at ../../py/runtime.c:668
#16 0x00000000005cdb7d in mp_execute_bytecode (code_state=0x7fffffffbee0, inject_exc=<optimized out>) at ../../py/vm.c:1090
#17 0x00000000005071d8 in fun_bc_call (self_in=0x7ffff13aed80, n_args=2, n_kw=0, args=0x7fffffffc030) at ../../py/objfun.c:288
#18 0x00000000004a79a9 in mp_call_function_n_kw (fun_in=0x7ffff13aed80, n_args=2, n_kw=0, args=0x7fffffffc030) at ../../py/runtime.c:652
#19 0x00000000004f3a95 in mp_call_method_self_n_kw (meth=0x7ffff13aed80, self=0x7ffff13aef40, n_args=1, n_kw=0, args=0x7fffffffc150) at ../../py/objboundmeth.c:70
#20 0x00000000004f3b0d in bound_meth_call (self_in=0x7ffff13af000, n_args=1, n_kw=0, args=0x7fffffffc150) at ../../py/objboundmeth.c:83
#21 0x00000000004a79a9 in mp_call_function_n_kw (fun_in=0x7ffff13af000, n_args=1, n_kw=0, args=0x7fffffffc150) at ../../py/runtime.c:652
#22 0x00000000004a78f1 in mp_call_function_1 (fun=0x7ffff13af000, arg=0x1) at ../../py/runtime.c:630
#23 0x00000000004c6e8b in mp_call_function_1_protected (fun=0x7ffff13af000, arg=0x1) at ../../py/runtime_utils.c:33
#24 0x00000000004c6fef in mp_handle_pending_tail (atomic_state=4294967295) at ../../py/scheduler.c:92
#25 0x00000000005cd2be in mp_execute_bytecode (code_state=0x7fffffffc300, inject_exc=<optimized out>) at ../../py/vm.c:1387
#26 0x00000000005071d8 in fun_bc_call (self_in=0x7ffff13aeac0, n_args=2, n_kw=0, args=0x7fffffffc470) at ../../py/objfun.c:288
#27 0x00000000004a79a9 in mp_call_function_n_kw (fun_in=0x7ffff13aeac0, n_args=2, n_kw=0, args=0x7fffffffc470) at ../../py/runtime.c:652
#28 0x00000000004f3a95 in mp_call_method_self_n_kw (meth=0x7ffff13aeac0, self=0x7ffff13aef40, n_args=1, n_kw=0, args=0x7fffffffc6d8) at ../../py/objboundmeth.c:70
#29 0x00000000004f3b0d in bound_meth_call (self_in=0x7ffff13af020, n_args=1, n_kw=0, args=0x7fffffffc6d8) at ../../py/objboundmeth.c:83
#30 0x00000000004a79a9 in mp_call_function_n_kw (fun_in=0x7ffff13af020, n_args=1, n_kw=0, args=0x7fffffffc6d8) at ../../py/runtime.c:652
#31 0x00000000004a9731 in mp_call_method_n_kw (n_args=1, n_kw=0, args=0x7fffffffc6c8) at ../../py/runtime.c:668
#32 0x00000000005cdb7d in mp_execute_bytecode (code_state=0x7fffffffc6a0, inject_exc=<optimized out>) at ../../py/vm.c:1090
#33 0x00000000005071d8 in fun_bc_call (self_in=0x7ffff13ae6c0, n_args=1, n_kw=0, args=0x7fffffffc840) at ../../py/objfun.c:288
#34 0x00000000004a79a9 in mp_call_function_n_kw (fun_in=0x7ffff13ae6c0, n_args=1, n_kw=0, args=0x7fffffffc840) at ../../py/runtime.c:652
#35 0x00000000004f3a95 in mp_call_method_self_n_kw (meth=0x7ffff13ae6c0, self=0x7ffff13af040, n_args=0, n_kw=0, args=0x7fffffffc960) at ../../py/objboundmeth.c:70
#36 0x00000000004f3b0d in bound_meth_call (self_in=0x7ffff13afca0, n_args=0, n_kw=0, args=0x7fffffffc960) at ../../py/objboundmeth.c:83
#37 0x00000000004a79a9 in mp_call_function_n_kw (fun_in=0x7ffff13afca0, n_args=0, n_kw=0, args=0x7fffffffc960) at ../../py/runtime.c:652
#38 0x00000000007451b6 in call_py_func (cif=0x7ffff13afd40, ret=0x7fffffffcb60, args=0x7fffffffc9e0, func=0x7ffff13afca0) at modffi.c:251
#39 0x00007ffff79b7c2f in ffi_closure_unix64_inner () from /usr/lib/x86_64-linux-gnu/libffi.so.6
#40 0x00007ffff79b7fa8 in ffi_closure_unix64 () from /usr/lib/x86_64-linux-gnu/libffi.so.6
#41 <signal handler called>
#42 0x0000000000405205 in gc_sweep () at ../../py/gc.c:314
#43 gc_collect_end () at ../../py/gc.c:361
#44 0x0000000000649577 in gc_collect () at gccollect.c:47
#45 0x00000000005bb87c in py_gc_collect () at ../../py/modgc.c:35
```

In this trace we can see that:
- `gc.collect()` was called (frame 45)
- During `gc_sweep`, a signal handler was invoked and called a micropython function through FFI (frames 38 - 41)
- The function, running under a signal handler, calls `micropython.schedule` which doesn't schedule the function as I expected, but calls it directly (frames 24-25)
- The function tries to allocate gc memory and hangs in a deadlock, since the lock is held by gc_sweep on the same call stack (frames 0-3)

amirgon, 2021-06-09:
Actually I was using non-up-to-date Micropython.
After updating Micropython the problem does not reproduce.
Possibly solved by https://github.com/micropython/micropython/commit/b6b39bff47b5ff1ffc322a6e795d57451d4215a8

Closing.

closed 2021-06-09 by amirgon

dpgeorge, 2021-06-09:
> Actually I was using non-up-to-date Micropython.
> After updating Micropython the problem does not reproduce.
> Possibly solved by [b6b39bf](https://github.com/micropython/micropython/commit/b6b39bff47b5ff1ffc322a6e795d57451d4215a8)

Yes that commit would definitely change the behaviour of schedule on the unix port.  Not sure if it's 100% correct but I think the bug is captured by #7373.