micropython/micropython #7377: array constructor does not check for overflow

v923z, 2021-06-10:
This issue came up during a test with `ulab`, and in short, the problem is that `mp_binary_set_val_array` does not check, whether the value that one wants to set fits the container. As pointed out in the comment https://github.com/v923z/micropython-ulab/pull/408#issuecomment-857166133, this is `CPython`-incompatible behaviour, while `numpy` seems to ignore the problem, generating outputs identical to `micropython` (and `ulab`). 

An example is the following snippet:

```python
>>> import array
>>> a = array.array('b', [257])
>>> print(a)
array('b', [1])
```
while `CPython` throws an exception:

```python
>>> import array
>>> a = array.array('b', [257])
>>> print(a)

OverflowError 
Traceback (most recent call last)
<ipython-input-5-c436b4b4a3aa> in <module>
      1 import array
----> 2 a = array.array('b', [257])
      3 print(a)
OverflowError: signed char is greater than maximum
```

@dpgeorge If you think this incompatibility warrants a fix, I am more than happy to submit a PR.

dpgeorge, 2021-06-12:
This has come up a few times before, but I can't find the issues/PRs right now.

I prefer MicroPython's behaviour because truncation is usually what you want, and doing it implicitly (the current behaviour) saves doing it explicitly in the code (which costs bytecode size and VM execution time).

It is incompatible with CPython.  But it is still very easy to write code that is compatible with both CPython and MicroPython, ie just apply an explicit mask like `a[0] = value & 0xff`.

There are a fair few things like this in MicroPython, where MicroPython is more liberal than CPython, so there is precedence to keep the current behaviour, and my opinion is to just leave it as-is because the code is simpler and faster.

v923z, 2021-06-12:
I don't insist on breaking from `micropython`'s current behaviour. If you want me to, I can implement the required code, otherwise, just close the issue. I also tend to agree with your sentiments about the impact on code size, and execution time. We face(d) similar issues in `ulab`, and almost always concluded that, since a microcontroller is approx. 10 000 - 100 000 times smaller than a computer, we have to choose the cheaper solution, even if it leads to small incompatibilities. 

What could help, however, is, if such differences were spelt out in a prominent place, in either the documentation, or the readme. (The code usually comments on these cases.) You could even turn the list of differences to `CPython` into a sticky issue, so that people don't submit them over and over again.

dpgeorge, 2021-06-12:
> What could help, however, is, if such differences were spelt out in a prominent place, in either the documentation, or the readme.

We have `tests/cpydiff` which is a way to document differences to CPython which include code snippets.  We can add something to that for this issue of array assignment and truncation of int values.

v923z, 2021-06-12:
I have submitted a PR,  but it fails with the commit message format. I promise, I will watch out for this next time.