micropython/micropython #7386: Possible race condition failure in uasyncio wait_for/wait_for_ms

jonathanhogg, 2021-06-12:
I'm seeing an issue using `wait_for_ms()` on reads from a socket, on ESP32 devices with recent HEAD, where occasionally I get an uncaught task `CancelledError` logged to the terminal and the `wait_for_ms()` never returns, locking up my code.

I'm using UDP sockets, so I'm not using the built-in `Stream` stuff, instead I'm rolling my own using the same underlying `yield core._io_queue.queue_read()` trick.

I suspect that it's triggered by a packet arriving at the same time as the wait ending, but I've had a gander at the `uasyncio` code and I can't see any obvious logic errors.

I'm using an ugly workaround at the moment (ditching the IOQueue and just repeatedly attempting to read from a non-blocking socket) as I have to have this code working this week, but I'll try to get back to it next week and see if I can reduce it to a smaller failing case and work out what's going wrong.

dpgeorge, 2021-06-13:
What is the exact error printed out?  Might be related to #7289.

If you can get a reproduction (even a large amount of code) that would allow it to be investigated.

jonathanhogg, 2021-06-13:
OK, I have a reliable repro for this.

`receiver.py` to run on the (ESP32) device:

```python
import socket
import uasyncio as asyncio
import network

async def async_recvfrom(sock, count):
    yield asyncio.core._io_queue.queue_read(sock)  # noqa
    return sock.recvfrom(count)

async def receive_loop(ssid, password, listen_port=49000):
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.connect(ssid, password)
    while wlan.status() != network.STAT_GOT_IP:
        await asyncio.sleep_ms(500)
    print("IP address: {}".format(wlan.ifconfig()[0]))
    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    sock.bind(('', listen_port))
    while True:
        try:
            data, (host, port) = await asyncio.wait_for(async_recvfrom(sock, 1500), 1)
            print("Received {} bytes from {}:{}".format(len(data), host, port))
        except asyncio.TimeoutError:
            pass

def run(ssid, password):
    asyncio.run(receive_loop(ssid, password))
```

And `sender.py` to run on something else:

```python
import socket
import time

def run(host, port=49000, data=b'Hello world!'):
    sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    sock.connect((host, port))
    while True:
        sock.send(data)
        print("Sent {} bytes to {}:{}".format(len(data), host, port))
        time.sleep(1)
```

Log from the device:

```
>>> import receiver
>>> receiver.run('[***]', '[***]')
IP address: 192.168.178.47
Received 12 bytes from 192.168.178.20:49450
Received 12 bytes from 192.168.178.20:49450
Received 12 bytes from 192.168.178.20:49450
Task exception wasn't retrieved
future: <Task> coro= <Task>
<class 'CancelledError'>
>>> 
```

You can see that it doesn't typically take long to fail. I have deliberately made the wait on both ends the same as I'm pretty sure it's a race condition. In this case the scheduler has exited, realising that there are no runnable tasks. In my real code I have another task running so the networking loop just hangs up.

This is running with a bunch of patches (my outstanding PRs and Mike Teachman's i2s) on top of da8aad18a4508476aa041406c0b06f5df4a8e59b.

It doesn't feel like it's related to #7289.

jonathanhogg, 2021-06-13:
For good measure, I've also reproduced it on current HEAD (66115a37444350b166384ae8be95639aef5a5af4) with IDF 4.2 and with the `esp32-20210418-v1.15.bin` release firmware.

jonathanhogg, 2021-06-13:
Out of curiosity I tried taking current HEAD and reverting out b505971069333a373d9f0c12138b64dab83fed72, which looks to be the last thing that touched `wait_for()`.

I made a small tweak in my test code above to print out a message on timeout to confirm that both branches are being exercised and I've had this build running that code for a few minutes now without it failing.

So I suspect that the fix to #5797 introduced another bug.

I'd like to dig deeper into this code, but I really need to get back to the *actual* work I'm supposed to be doing today… 😀

dpgeorge, 2021-06-13:
Thanks for the reproduction.  I can get the above code to crash on ESP32 (TinyPICO using the GENERIC build) as of the latest commit 66115a37444350b166384ae8be95639aef5a5af4.  But it did not crash a PYBD-SF2.

```
Task exception wasn't retrieved
future: <Task> coro= <Task>
<class 'CancelledError'>
```
There is something wrong with this output: it should be printing an exception *instance* not a class.  So that's a hint as to what is wrong (and I have seen this before on an unrelated project, but it was not possible to reproduce it).

> So I suspect that the fix to #5797 introduced another bug.

The fix could have also uncovered another bug.  Investigation will tell.

dpgeorge, 2021-06-13:
I can reproduce the same issue with the following code that does not use sockets:
```python
import uasyncio as asyncio
import machine

async def wait_flag(flag):
    await flag.wait()

async def receive_loop():
    flag = asyncio.ThreadSafeFlag()
    machine.Timer(0).init(freq=99.9, callback=lambda t:flag.set())
    while True:
        try:
            await asyncio.wait_for(wait_flag(flag), 0.01)
            print("Flag was set")
        except asyncio.TimeoutError:
            print("TimeoutError")

def run():
    asyncio.run(receive_loop())
```
But I only see the error if WLAN is connected to an access point...

jonathanhogg, 2021-06-13:
> But I only see the error if WLAN is connected to an access point...

Curiouser and curiouser…

I'd love to get stuck in and help, but I'm aware that a) you know the code, and b) I *really* need to finish this proposal…

hoihu, 2021-06-13:
I can confirm that we see a very related issue (on a STM32F777 based system, no wlan), specifically the `<class 'CancelledError'>` error.

We have not seen this issue before we updated to 1.15 current head, coming from 1.13. We haven't (yet) tried to revert https://github.com/micropython/micropython/commit/b505971069333a373d9f0c12138b64dab83fed72 .

On our device the fault is showing up during integration testing, usually with higher communication loads (there are several UARTs and USB VCP).

dpgeorge, 2021-06-14:
I found a simple reproduction that does not use `wait_for()` (but is a reduction of the `wait_for` code):
```python
import time
import uasyncio as asyncio

TIMEOUT = 2

async def sleep_task():
    print(time.ticks_us(), "sleep_task sleep")
    await asyncio.sleep(0)

async def wait_task(t):
    print(time.ticks_us(), "wait_task wait")
    await t

async def main():
    waiting_task = asyncio.create_task(wait_task(asyncio.create_task(sleep_task())))

    print(time.ticks_us(), "main sleep")
    await asyncio.sleep_ms(TIMEOUT)

    waiting_task.cancel()
    print(time.ticks_us(), "main wait")
    await waiting_task

asyncio.run(main())
```
It errors out (most of the time, not always) with:
```
153758444 main sleep
153759156 sleep_task sleep
153759933 wait_task wait
153761033 main wait
Task exception wasn't retrieved
future: <Task> coro= <Task>
<class 'CancelledError'>
```
It will error on ESP32 without WLAN active.  It will error on PYBD-SF2 running at 48MHz, but not when running faster.

dpgeorge, 2021-06-14:
And here's a deterministic reproduction that also fails on the unix port:
```python
import uasyncio as asyncio

async def sleep_task():
    print("sleep_task sleep")
    await asyncio.sleep(0)
    print("sleep_task wake")

async def wait_task(t):
    print("wait_task wait")
    await t
    print("wait_task wake")

async def main():
    waiting_task = asyncio.create_task(wait_task(asyncio.create_task(sleep_task())))

    print("main sleep")
    await asyncio.sleep(0)
    await asyncio.sleep(0)

    waiting_task.cancel()
    print("main wait")
    await waiting_task
    print("main wait")

asyncio.run(main())
```

dpgeorge, 2021-06-14:
The issue seems to be when task A waits on task B, and there's a race between cancellation of task A and completion of task B.  If task B completes just before task A is cancelled then the cancellation of A does not work but rather prints the "Task exception wasn't retrieved" message.

dpgeorge, 2021-06-14:
See #7399 for a proposed fix.

dpgeorge, 2021-06-15:
Summary of the bug:
- occurs when using `wait_for(T, S)`, if the task T being waited on finishes at exactly the same time as the wait-for timeout S expires
- task T will have run to completion
- the `Task exception wasn't retrieved` message is printed with `<class 'CancelledError'>` as the error (ie no traceback)
- the `wait_for(T, S)` call never returns (it's never put back on the uasyncio run queue) and all tasks waiting on this are blocked forever from running
- uasyncio otherwise continues to function and other tasks continue to be scheduled as normal

jonathanhogg, 2021-06-15:
Can confirm that fix in #7399 works on my repro test case. I've sucked it into my branch and have switched my code back to the original `wait_for_ms()` and `yield ...queue_read()` combo and will run this today to see if it remains stable.

jonathanhogg, 2021-06-15:
Have had this patch running all day with my networking code without any obvious failures, so it looks good to me.

jonathanhogg, 2021-06-16:
(Updated to 514bf1a1911ac9173a00820b7e09dfb387e6b941 patch and that also seems solid today)

dpgeorge, 2021-06-17:
Thanks @jonathanhogg for testing.  Now merged in 514bf1a1911ac9173a00820b7e09dfb387e6b941

closed 2021-06-17 by dpgeorge

alessionossa, 2022-01-15:
I think I'm still facing this issue using _MicroPython v1.17 on 2021-09-02; ESP32 module with ESP32_.

peterhinch, 2022-01-16:
I think the loop beginning
```python
        while not foundPacket:
```
is hogging the scheduler. Try adding `await uasyncio.sleep_ms(0)` in the loop.

dpgeorge, 2022-01-16:
Indeed, `_getNatPunchResponse` never await's.

jonathanhogg, 2022-01-20:
@alessionossa: if you look at my repro code [way up in this thread](https://github.com/micropython/micropython/issues/7386#issuecomment-860200607) you'll see a simple mechanism for doing async `recv`s from a UDP socket. Something similar to that should solve your problem.