micropython/micropython #7387: mimxrt: Question about an SPI slave mode implementation.

robert-hh, 2021-06-12:
This is an implementation question, not an issue.
I have almost finished the implementation of Hard SPI for the MIMXRT devices. Adding slave mode I have questions. Unlike master mode, slave mode can fail if it does not receive data. I added a timeout mechanism, such that it does not block forever. But how should it tell the upper layers that it failed. Should it raise an exception? The transfer function is of type void, so there is no way of returning it there.

dpgeorge, 2021-06-13:
The API for slave/peripheral mode needs a bit of thought and discussion.  I think it should be a separate class like `SPIPeripheral` and `I2CPeripheral` (ie not how pyb.SPI does it).

Would be best to just implement master/controller mode on its own first.

robert-hh, 2021-06-13:
At the moment the code raises an OSERROR ETIMEDOUT exception, which works. And it support both Controller/Peripheral mode. For that I had to extend the dictionary in extmod/machine_spi.c to add the keywords. But I can also duplicate the dictionary in the port specific machine_spi.c. The code works fine with DMA and non-DMA for short messages, but needs some cleanup.

closed 2021-06-15 by robert-hh