micropython/micropython #7388: qstr_from_str can make qstr with length > 10

bhcuong2008, 2021-06-12:
Hi,

During my test with C-API MPY, I use qstr_from_str with string longer than 10, and it adds qstr successfully to QSTR pool, checked by micropython.qstr_info(1). Following the docs, it says that string length must be shorter than 10. I dont know whether it's code fault or not. So I report here.

I tested with MPY 1.15. Just in main C application, add 
`qstr_from_str("test_1234567890_1234567890");`

After that, in REPL, import micropython to check qstr pool.

Thank.

dpgeorge, 2021-06-13:
That's expected behaviour: certain strings must always be interned (converted to a qstr), eg when used as the name of a module.  And all identifiers are interned as well.

The limit of 10 characters is only when parsing Python code, and only applies to str/bytes objects.  See `MICROPY_ALLOC_PARSE_INTERN_STRING_LEN`.

closed 2021-06-13 by dpgeorge

bhcuong2008, 2021-06-13:
Thank @dpgeorge . 
I use mp_lexer_new_from_file, then it creates new qstr with full path filename. So it takes some memory per filename, often 20-30 chars. Is there any ways to limit it?

dpgeorge, 2021-06-13:
> I use mp_lexer_new_from_file, then it creates new qstr with full path filename. So it takes some memory per filename, often 20-30 chars. Is there any ways to limit it?

No, you can't limit that, the lexer requires a qstr because of how tracebacks work.

You could instead use the following custom function:
```c
mp_lexer_t *mp_lexer_new_from_file_with_name(const char *filename, qstr source_name) {
    mp_reader_t reader;
    mp_reader_new_file(&reader, filename);
    return mp_lexer_new(source_name, reader);
}
```
Then pass in a shorter `source_name`.

bhcuong2008, 2021-06-13:
It's great. Thank @dpgeorge .