micropython/micropython #7390: mpremote is missing os import on windows

stinos, 2021-06-13:
As reported on the forum:

[code]
File "c:\python39\lib\site-packages\mpremote\console.py", line 145, in set_conout_mode
fdout = os.open("CONOUT$", os.O_RDWR)
NameError: name 'os' is not defined
[/code]

dpgeorge, 2021-06-13:
Yes, thanks for the reminder, I was intending to fix this.  Is that the only fix to make it work on Windows?

stinos, 2021-06-13:
Not sure, I asked the one reporting it on the forum to chime in here.

haybarry, 2021-06-13:
Hi,  That would be me.  

The addition of the extra "import os" in console.py gets mpremote running in windows.- at least as far as getting console from my target (PRI PICO) to powershell.

Have not yet been able to test any other mpremote functionality as I don't seem to be able to interrupt the program running on the PICO.  The usual ctrl C terminates mpremote, not the RPI, so I can't get to repl on the PICO, or transfer files etc.

btw, I was tryin mpremote again because Pymakr and the latest version of VSCode do not play nocely :(

Barry

dpgeorge, 2021-06-13:
See #7395 for a fix for the os error, and also to handle ctrl-C on Windows.

dpgeorge, 2021-06-15:
Should now be fixed.  Latest version is on PyPI: https://pypi.org/project/mpremote/0.0.5/

closed 2021-06-15 by dpgeorge

joepassman, 2023-06-20:
Hi, I am trying to connect a TinyS3 board to Arduino Cloud via mpremote. I am following the instructions [here](https://docs.arduino.cc/arduino-cloud/getting-started/iot-cloud-micropython)

I am having this error and can't figure out how to move past it. Any ideas?

(base) C:\Users\JosephPassman>mpremote connect id:CD4557ACEF0F mip install github:arduino/arduino-iot-cloud-py

Install github:arduino/arduino-iot-cloud-py
Installing github:arduino/arduino-iot-cloud-py/package.json to /lib
Installing: /lib/arduino_iot_cloud/__init__.py
Traceback (most recent call last):
  File "C:\Users\JosephPassman\anaconda3\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\JosephPassman\anaconda3\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\JosephPassman\anaconda3\Scripts\mpremote.exe\__main__.py", line 7, in <module>
  File "C:\Users\JosephPassman\anaconda3\lib\site-packages\mpremote\main.py", line 502, in main
    handler_func(state, args)
  File "C:\Users\JosephPassman\anaconda3\lib\site-packages\mpremote\mip.py", line 183, in do_mip
    _install_package(
  File "C:\Users\JosephPassman\anaconda3\lib\site-packages\mpremote\mip.py", line 146, in _install_package
    _install_json(pyb, package, index, target, version, mpy)
  File "C:\Users\JosephPassman\anaconda3\lib\site-packages\mpremote\mip.py", line 109, in _install_json
    _download_file(pyb, _rewrite_url(url, version), fs_target_path)
  File "C:\Users\JosephPassman\anaconda3\lib\site-packages\mpremote\mip.py", line 79, in _download_file
    _ensure_path_exists(pyb, dest)
  File "C:\Users\JosephPassman\anaconda3\lib\site-packages\mpremote\mip.py", line 33, in _ensure_path_exists
    pyb.fs_mkdir(prefix)
  File "C:\Users\JosephPassman\anaconda3\lib\site-packages\mpremote\pyboard.py", line 640, in fs_mkdir
    self.exec_("import uos\nuos.mkdir('%s')" % dir)
  File "C:\Users\JosephPassman\anaconda3\lib\site-packages\mpremote\pyboard.py", line 496, in exec_
    raise PyboardError("exception", ret, ret_err)
mpremote.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nImportError: no module named \'uos\'\r\n')

jimmo, 2023-06-21:
@joepassman Can you connect to the REPL on the device and run:

```
>>> help("modules")
```