micropython/micropython #7394: RTC weekday meaning incosistences.

kadamski, 2021-06-13:
We have just moved one step closer to make the RTC.datetime() more consistent in fd4eec5555f93 but the reports quickly revealed that we still have a problem - the weekday has different range on different ports.

The RP2 port will accept any value betweeb 0-6 there, quite interestingly, the rp2040 datasheet mentions that dayw = 0 means Sunday but it also says:
"There  is  no  built-in  calendar  function.  The  RTC  will  not  compute  the  correct  day  of  the  week;  it  will  only  increment  the existing value.". So, basically, the meaning of the dayw = 0 can be whatever you want, it just depends on your interpretation. 

mimxrt port uses 0 = Monday. esp32, esp8266 uses the same interpretation as utime, so 0 = Monday. STM32 also considers Monday to be the first day of week but it uses 1-7, not 0-6 :/
Neither nrf nor cc3200 support RTC.datetime().

mimxrt, esp32, esp8266 will ignore the weekday value given when initializing the datetime and will compute its own value when returning it. STM32 and PR2 will put the given value to the hardware, **but only if it is a valid weekday**, otherwise the whole RTC setting will silently fail. See PR #7392 that fixes this for RP2.

To sum up:
- mimxrt, esp32, esp8266 will accept any value as weekday, it will use 0-6 (0=Monday) when returning the value
- rp2 will only accept 0-6, will return whatever was set before; if value is out of range, the whole datetime will not be set to hardware
- stm32 will only accept 1-7, will return whatever was set before; it uses assert_param to validate the value is in range but this macro is disabled by default, I don't know what will happen if the value of our range will be programmed to the hardware - it seems like in case of values >7, it will overwrite part of the year field, not sure what will happen if 0 is used - the datasheet says this value is *forbidden*. 

@dpgeorge Should we fix that? If so, how?

1. I think that we should, at least, create an equivalent to #7392 but for stm32. By default the HAL will not return an error if value out of range is used, but it seems that bad things may happen.
2. We have to decide if we want to change stm32 to use 0-6 (i.e. map the values so that they are consistent) or leave it at 1-7, 
3. We should decide if we really want to use the user-specified value for stm32/rp2 or maybe we should ignore it just like mimxrt, esp32, esp8266, and compute it internally based on the given Y/M/D, instead of forcing the user to do it.
4. In general, I think it doesn't make sense to use weekday when setting the datetime as this information is actually redundant as it can always be computed from Y/M/D and only brings confusion.

dpgeorge, 2021-06-13:
> In general, I think it doesn't make sense to use weekday when setting the datetime as this information is actually redundant as it can always be computed from Y/M/D and only brings confusion.

I agree with this, much easier for the user to just compute weekday automatically (because it's hard for them to compute it, and there's already C code to do it in uPy).  We could allow a value of `-1` for weekday (when setting it) to indicate that it should be automatically computed.

> We have to decide if we want to change stm32 to use 0-6 (i.e. map the values so that they are consistent) or leave it at 1-7,

It really needs to be consistent across all ports, but that means breaking code.  That has to be done, I guess, and it's not really possible to do it in a graceful way.

And it should be 0-6 maps to Mon-Sun to match `time.localtime()`.

[An alternative, which I don't know if I should bring up, is to abandon `datetime()` altogether, and use instead `now()` to get/set the format in a CPython compatible way... But note that `datetime.datetime.now()` is not actually a tuple, and does not even have the same format as `time.localtime()` (which does act like a tuple).]

peterhinch, 2021-06-14:
>  We could allow a value of -1 for weekday (when setting it) to indicate that it should be automatically computed.

Yes.

To minimise code breakage I would suggest making no changes to **pyb.RTC**.

Abandoning **machine.RTC.datetime** would break the code of surely everyone who has used the class. Consistency of **machine.RTC** across ports is crucial. Consequent code breakage seems unavoidable.

dpgeorge, 2021-06-14:
> To minimise code breakage I would suggest making no changes to **pyb.RTC**.

But that's necessary to make stm32 use 0-6 for the weekday value.

A way out could be to have `pyb.RTC` and `machine.RTC` behave differently: pyb.RTC remains unchanged and stm32's machine.RTC matches other ports and uses 0-6 for weekday (maybe that is what you meant?).

peterhinch, 2021-06-15:
> maybe that is what you meant?

Yes.

kevindawson, 2021-07-27:
My 2d

I agree with:
> In general, I think it doesn't make sense to use weekday when setting the datetime as this information is actually redundant as it can always be computed from Y/M/D and only brings confusion.

As for which port uses 0-6 or 1-7
had you considered telling people: -> by using your wiki

seems logical to me