micropython/micropython #7398: SPI master slave communication

embhobbb, 2021-06-14:
I am using pico in my project in which I am in need of using master-slave SPI communication.
What modifications are required to change the SPI master code to spi slave?
Is there any existing example for reference.

robert-hh, 2021-06-14:
I do not know whether a sample exists. But looking into the C-SDK API, there is a function called spi_set_slave(). It should not to be too hard to change the code in machine_spi.c to support both master and slave mode, at least in non-DMA mode.

embhobbb, 2021-06-14:
thanks, @robert-hh but i am looking for micropython implementation.

robert-hh, 2021-06-14:
That will only work with the PIO. Did you have a look at the PIO examples.

kadamski, 2021-06-16:
@embhobbb  the SPI slave is not currently supported on RP2 port. That would have to be added. It seems like a trivial task, just like Robert said, but someone would have to do it, and more importantly, test it. So you either have to do it yourself or wait for someone to do it.

peterhinch, 2021-06-16:
There is a difficulty with slave mode (SPI or I2C). If you have a blocking method, the application needs to ensure that it is executing the blocking method at the time when the master wants to access it. This can be quite hard to achieve, as (by definition) while the application is waiting, it can't do anything else. An interface is usually required to enable the slave to goad the master into action before the slave disappears down the rabbit hole of the blocking method. Manageable if you have control over the master code; with a hardware master, problematic.

This can be fixed by nonblocking methods, but to achieve the performance expected of SPI this would require C code and possibly DMA. Not quite so trivial.

kadamski, 2021-06-17:
@peterhinch Well, that depends - we have two cores after all. One can do its job (data acquisition etc), other can respond to SPI requests. While this isn't terribly efficient, it should work. Nonblocking (callback/DMA based) operation would be better but that doesn't mean the simple blocking API is of no use.

peterhinch, 2021-06-18:
Fair point, though as @robert-hh suggested, the PIO is probably the most efficient way to do this. It should be possible to implement a nonblocking slave without changes to the firmware. A general solution would need some thought to cope with the SPI initialisation options and to figure out the interface with Python. Maybe someone has already done this?

robert-hh, 2021-06-18:
> A general solution would need some thought to cope with the SPI initialisation options and to figure out the interface with Python. Maybe someone has already done this?

I have made a PR for the MIMXRT port that supports (a simple) hardware slave mode. That will be the real case for discussion.  The way it is implemented now, slave write or read is blocking, but a timeout can be set. Lacking an interface to pass pack the state, an exception is raised at timeout. Besides the external interface being blocking, the internal mechanism is non-blocking. So like UART the interface can be extended to support some level of non-blocking behavior. But that will be tricky.

It should be noted that the mimxrt port is still in an early phase. threading for instance is not yet supported, which may help here.

alan-rpi, 2021-07-21:
I would like to support this request and I've seen plenty of forum posts where others would like to do the same.

I come from a Pi/python background and the availability of micropython on the Pico and being a RPi Foundation development was really appealing. I am building an IOT sensor and such sensors usually work in a slave* mode.

Many thanks to all the people who have developed micropython and the Thonny IDE. Really looking good.

Alan 
*I've used the Master/Slave notation as the OP used that

PS It seems the I2C protocol on the Pico also only works in Master mode.

kwagyeman, 2023-10-09:
@robert-hh

We need to add this feature to the STM32 and MIMXRT so that can port our interface libraries to the machine module. What does it look like to make this happen sooner? E.g. SPI Slave mode. We also need I2C Slave mode.

alan-rpi, 2023-10-26:
In the end I used an Arduino Nano Every instead of a Pico. Did the job but I had to learn the new h/w, a new language and new development environment that had technical issues, so all-in-all was time consuming.