micropython/micropython #7403: Latest commit to ESP32 RMT code seems to cause panics with multiple channels

jonathanhogg, 2021-06-16:
The latest commit to the RMT code (8ff3520f67bb8f7e5509129f9f561a1217ababa4) – to zero out the newly initialised `rmt_config_t` object – is causing panics for me on ESP32 devices when I start a second looping RMT channel.

Reversing out this change causes the problem to go away for me. I'm unclear as to what is going on with it.

I'm afraid that I'm not working with RMT code today so I have limited ability to investigate further – I only noticed when trying to run some older code (which does use RMT) on a device to bisect a different problem.

Repro:

```
>>> from esp32 import RMT
>>> from machine import Pin
>>> 
>>> rmt0 = RMT(0, pin=Pin(12, Pin.OUT), clock_div=2)
>>> rmt0.loop(True)
>>> rmt0.write_pulses([16, 34])
>>> 
>>> rmt1 = RMT(1, pin=Pin(14, Pin.OUT), clock_div=2)
>>> rmt1.loop(True)
>>> rmt1.write_pulses([16, 34])
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4008bf08  PS      : 0x00060034  A0      : 0x40082d00  A1      : 0x3ffbe430  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ff56000  A5      : 0x00000008  
A6      : 0x00000004  A7      : 0x00000080  A8      : 0x00000040  A9      : 0x3ff56000  
A10     : 0x00000002  A11     : 0x3ffbe434  A12     : 0x3ffbd604  A13     : 0x3ffbd650  
A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
Core  0 was running in ISR context:
EPC1    : 0x401cd383  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008bf08

Backtrace:0x4008bf05:0x3ffbe430 0x40082cfd:0x3ffbe470 0x4000bfed:0x3ffcced0 0x40091eed:0x3ffccee0 0x40158b2a:0x3ffccf00 0x40108900:0x3ffccf30 0x400db356:0x3ffccf80 0x400e150d:0x3ffccfb0 0x400e163a:0x3ffccfd0 0x400e3c61:0x3ffccff0 0x400db3e4:0x3ffcd090 0x400e150d:0x3ffcd0f0 0x400e1536:0x3ffcd110 0x400f39ff:0x3ffcd130 0x400f3d11:0x3ffcd1c0 0x400d5394:0x3ffcd200
Core  1 register dump:
PC      : 0x401c727e  PS      : 0x00060034  A0      : 0x800d4b86  A1      : 0x3ffbc490  
A2      : 0x3ffc3bbd  A3      : 0x00060020  A4      : 0x00060020  A5      : 0x00060023  
A6      : 0x00000001  A7      : 0x00060723  A8      : 0x800d457d  A9      : 0x3ffbc480  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060020  A13     : 0x00000000  
A14     : 0x00060020  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401c727b:0x3ffbc490 0x400d4b83:0x3ffbc4b0 0x400924a0:0x3ffbc4d0

ELF file SHA256: 0ce3367c3cb0b316

Rebooting...
```

Back trace likely not useful here as I'm running a custom firmware build.

jonathanhogg, 2021-06-16:
OK, reproduced this on *master* (as of 66115a37444350b166384ae8be95639aef5a5af4) and confirmed that removing the initialiser makes it go away. Also confirmed that just adding `config.flags = 0` also causes it. I've also reproduced it on two different devices (both Heltec/MakerHawk ESP32 boards).

Out of curiosity I tried removing both and printing out the value of `config.flags` to see what it looks like unitialised and I get a `LoadProhibited` panic instead.

Nothing about this makes sense to me yet.

jonathanhogg, 2021-06-16:
I've tried a bunch of different ways of initialising this structure – including just following the latest IDF 4.2 examples using the `RMT_DEFAULT_CONFIG_TX` macro they've added – and all of them are exploding. However, it's *only* exploding when I have two channels both looping.

I'm leaning towards this being something else going wrong and the non-zero initialisation is somehow masking that other error.

jonathanhogg, 2021-06-16:
So can confirm that code doesn't explode on ESP32 v1.15 official release firmware, but the master clock is clearly 1MHz instead of 80MHz (hadn't gotten as far as working code to see what the output looked like before). Is this flags thing an IDF change? I mean, the code worked fine a year ago… I'd rebuild with different IDF versions, but it's such a staggering bore to switch.

I've also tried a few variations of using two looping RMT channels and it seems to be related somehow to how frequently they loop. I've seen the interpreter lock-up, but not panic, and also weird slow downs.

Running this:

```
from esp32 import RMT
from machine import Pin

rmt0 = RMT(0, pin=Pin(12, Pin.OUT), clock_div=20)
rmt0.loop(True)
rmt0.write_pulses([16, 34])

rmt1 = RMT(1, pin=Pin(14, Pin.OUT), clock_div=20)
rmt1.loop(True)
rmt1.write_pulses([16, 34])
```

is completely fine. As is:

```
rmt0 = RMT(0, pin=Pin(12, Pin.OUT), clock_div=10)
rmt0.loop(True)
rmt0.write_pulses([16, 34, 16, 34])

rmt1 = RMT(1, pin=Pin(14, Pin.OUT), clock_div=10)
rmt1.loop(True)
rmt1.write_pulses([16, 34, 16, 34])
```

and:

```
rmt0 = RMT(0, pin=Pin(12, Pin.OUT), clock_div=12)
rmt0.loop(True)
rmt0.write_pulses([16, 34])

rmt1 = RMT(1, pin=Pin(14, Pin.OUT), clock_div=12)
rmt1.loop(True)
rmt1.write_pulses([16, 34])
```

But:

```
rmt0 = RMT(0, pin=Pin(12, Pin.OUT), clock_div=10)
rmt0.loop(True)
rmt0.write_pulses([16, 34])

rmt1 = RMT(1, pin=Pin(14, Pin.OUT), clock_div=10)
rmt1.loop(True)
rmt1.write_pulses([16, 34])
```

panics.

The panic is in ISR mode according to the message. The thing is, if it's interrupt load, why are we even seeing interrupts? The looping in the RMT is hardware managed and we explicitly disable the TX interrupt when it's on (this is what I spent all that time last year working out).

dpgeorge, 2021-06-17:
I can confirm the crash, with `clock_div=10` and using `[16, 34]` for the pulses.  Using `clock_div=11` it doesn't panic but it does slow down the REPL/VM a huge amount.

dpgeorge, 2021-06-17:
I'm testing with a TinyPICO and using IDF 4.0.2 there is no crash.  Using IDF 4.1.1 there is a crash, and this IDF version does not have the `rmt_config_t.flags` entry.

dpgeorge, 2021-06-17:
IDF v4.4-dev-1404-gc13afea63 also shows the error.  Here is the dump:
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x4009395b  PS      : 0x00060034  A0      : 0x8008ca88  A1      : 0x3ffbe990
0x4009395b: xQueueGiveFromISR at esp-idf-v4.4/components/freertos/queue.c:1176

A2      : 0x3ffbb0b8  A3      : 0x3ffbe9b4  A4      : 0x3ffbb0b8  A5      : 0x3ffbb104
A6      : 0x00060021  A7      : 0x00000001  A8      : 0x80093958  A9      : 0x3ffbe970
A10     : 0x3ffc26a0  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x3ffbe980
A14     : 0x3ffc26a0  A15     : 0x00000008  SAR     : 0x00000020  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff
Core  0 was running in ISR context:
EPC1    : 0x400d3bd7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4009395b
0x400d3bd7: uart_hal_write_txfifo at esp-idf-v4.4/components/hal/uart_hal_iram.c:35
0x4009395b: xQueueGiveFromISR at esp-idf-v4.4/components/freertos/queue.c:1176

Backtrace:0x40093958:0x3ffbe9900x4008ca85:0x3ffbe9b0 0x400834f1:0x3ffbe9f0 0x400e6926:0x3ffcdd10 0x400e6b05:0x3ffcdd30 0x400ddf4a:0x3ffcdd50 0x400e6f1c:0x3ffcdd80 0x400e6f82:0x3ffcdda0 0x400e8ab7:0x3ffcddc0 0x400e0fbc:0x3ffcde60 0x400e70bd:0x3ffcdeb0 0x400e70e6:0x3ffcded0 0x400f9523:0x3ffcdef0 0x400f9835:0x3ffcdf80 0x400daf3c:0x3ffcdfc0

0x40093958: xQueueGiveFromISR at esp-idf-v4.4/components/freertos/queue.c:1176
0x4008ca85: rmt_driver_isr_default at esp-idf-v4.4/components/driver/rmt.c:772
0x400834f1: _xt_lowint1 at esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x400e6926: find_qstr at micropython/py/qstr.c:140
0x400e6b05: qstr_hash at micropython/py/qstr.c:260
0x400ddf4a: mp_map_lookup at micropython/py/map.c:212
0x400e6f1c: mp_load_global at micropython/py/runtime.c:176
0x400e6f82: mp_load_name at micropython/py/runtime.c:170
0x400e8ab7: mp_execute_bytecode at micropython/py/vm.c:368
0x400e0fbc: fun_bc_call at micropython/py/objfun.c:288
0x400e70bd: mp_call_function_n_kw at micropython/py/runtime.c:649
0x400e70e6: mp_call_function_0 at micropython/py/runtime.c:623
0x400f9523: parse_compile_execute at micropython/lib/utils/pyexec.c:116
0x400f9835: pyexec_friendly_repl at micropython/lib/utils/pyexec.c:661
0x400daf3c: mp_task at micropython/ports/esp32/build-GENERIC/../main.c:160

Core  1 register dump:
PC      : 0x401c0422  PS      : 0x00060234  A0      : 0x800d37f6  A1      : 0x3ffbd2f0
0x401c0422: esp_pm_impl_waiti at esp-idf-v4.4/components/esp_pm/pm_impl.c:830

A2      : 0x3ffc260d  A3      : 0x00060220  A4      : 0x00060220  A5      : 0x00060223
A6      : 0x00000001  A7      : 0x00060923  A8      : 0x800d6da9  A9      : 0x3ffbd2e0
A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060220  A13     : 0x00000000
A14     : 0x00060020  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x401c041f:0x3ffbd2f00x400d37f3:0x3ffbd310 0x40094008:0x3ffbd330
0x401c041f: cpu_ll_waiti at esp-idf-v4.4/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at esp-idf-v4.4/components/esp_pm/pm_impl.c:828

0x400d37f3: esp_vApplicationIdleHook at esp-idf-v4.4/components/esp_system/freertos_hooks.c:63
0x40094008: prvIdleTask at esp-idf-v4.4/components/freertos/tasks.c:3823
```

jonathanhogg, 2021-06-17:
> I'm testing with a TinyPICO and using IDF 4.0.2 there is no crash. Using IDF 4.1.1 there is a crash, and this IDF version does not have the `rmt_config_t.flags` entry.

Slightly mysterious then. I guess there's two different IDF changes been introduced here: the new clock selection flags and whatever change is causing the crash.

I'm super-busy this week and RMT is not something I need to have working for this current project (I've switched to SPI – as other bug might suggest 😉), but I'll try to return to this next week and see if I can reduce to a small C example using only the IDF. If I can make that crash then I'll raise with Espressif to see what they think.

jonathanhogg, 2021-06-22:
Right. So I've had some time to look at this in the studio with my oscilloscope and boiling down the code to a minimal plain C example:

```C
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "driver/rmt.h"

static const char *TAG = "rmt_test";

static const rmt_item32_t test_pulses[] = {
    {{{ 16, 1, 34, 0 }}},
};

void start_channel(rmt_channel_t channel, gpio_num_t pin)
{
    ESP_LOGI(TAG, "Start RMT channel %d on pin %d", channel, pin);
    rmt_config_t config = {0};
    config.rmt_mode = RMT_MODE_TX;
    config.channel = channel;
    config.gpio_num = pin;
    config.mem_block_num = 1;
    config.tx_config.loop_en = 0;
    config.tx_config.carrier_en = 0;
    config.tx_config.idle_output_en = 1;
    config.tx_config.idle_level = 0;
    config.clk_div = 2;
    ESP_ERROR_CHECK(rmt_config(&config));
    ESP_ERROR_CHECK(rmt_driver_install(channel, 0, 0));

    ESP_ERROR_CHECK(rmt_set_tx_intr_en(channel, false));
    ESP_ERROR_CHECK(rmt_write_items(channel, test_pulses, sizeof(test_pulses) / sizeof(test_pulses[0]), false));
    ESP_ERROR_CHECK(rmt_set_tx_loop_mode(channel, true));
}

void stop_channel(rmt_channel_t channel)
{
    ESP_LOGI(TAG, "Stop RMT channel %d", channel);
    ESP_ERROR_CHECK(rmt_set_tx_intr_en(channel, true));
    ESP_ERROR_CHECK(rmt_set_tx_loop_mode(channel, false));
    ESP_ERROR_CHECK(rmt_wait_tx_done(channel, portMAX_DELAY));
}

void app_main(void *ignore)
{
    start_channel(RMT_CHANNEL_0, 12);
    start_channel(RMT_CHANNEL_1, 14);

    vTaskDelay(10000 / portTICK_PERIOD_MS);

    stop_channel(RMT_CHANNEL_0);
    stop_channel(RMT_CHANNEL_1);

    while (true)
        vTaskDelay(1000 / portTICK_PERIOD_MS);
}
```

This is basically what's happening under the hood with the original MicroPython repro example. As expected, it explodes on IDF 4.2 and works on 4.0.2, but the interesting thing is that the 'scope shows the correct output signal on both channels when it hangs and crashes. The backtrace shows that it is exploding in `rmt_set_tx_loop_mode()`.

So I tried enabling looping mode in the config and removing this line, a la:

```C
void start_channel(rmt_channel_t channel, gpio_num_t pin)
{
    ESP_LOGI(TAG, "Start RMT channel %d on pin %d", channel, pin);
    rmt_config_t config = {0};
    config.rmt_mode = RMT_MODE_TX;
    config.channel = channel;
    config.gpio_num = pin;
    config.mem_block_num = 1;
    config.tx_config.loop_en = 1;
    config.tx_config.carrier_en = 0;
    config.tx_config.idle_output_en = 1;
    config.tx_config.idle_level = 0;
    config.clk_div = 2;
    ESP_ERROR_CHECK(rmt_config(&config));
    ESP_ERROR_CHECK(rmt_driver_install(channel, 0, 0));

    ESP_ERROR_CHECK(rmt_set_tx_intr_en(channel, false));
    ESP_ERROR_CHECK(rmt_write_items(channel, test_pulses, sizeof(test_pulses) / sizeof(test_pulses[0]), false));
}
```

and this runs fine without crashing. Now, I was never sure of the correct ordering of operations here – as the IDF documentation gives no information about how looping is meant to function – so I went back to configuring without looping enabled but then turning it on *before* writing the pulses instead of after:

```C
void start_channel(rmt_channel_t channel, gpio_num_t pin)
{
    ESP_LOGI(TAG, "Start RMT channel %d on pin %d", channel, pin);
    rmt_config_t config = {0};
    config.rmt_mode = RMT_MODE_TX;
    config.channel = channel;
    config.gpio_num = pin;
    config.mem_block_num = 1;
    config.tx_config.loop_en = 0;
    config.tx_config.carrier_en = 0;
    config.tx_config.idle_output_en = 1;
    config.tx_config.idle_level = 0;
    config.clk_div = 2;
    ESP_ERROR_CHECK(rmt_config(&config));
    ESP_ERROR_CHECK(rmt_driver_install(channel, 0, 0));

    ESP_ERROR_CHECK(rmt_set_tx_intr_en(channel, false));
    ESP_ERROR_CHECK(rmt_set_tx_loop_mode(channel, true));
    ESP_ERROR_CHECK(rmt_write_items(channel, test_pulses, sizeof(test_pulses) / sizeof(test_pulses[0]), false));
}
```

This also runs fine on both IDF 4.2 and 4.0.2.

So I've tried making a similar change in the RMT code in MicroPython:

```diff
diff --git a/ports/esp32/esp32_rmt.c b/ports/esp32/esp32_rmt.c
index 9619b1dd5..81a9bfe6f 100644
--- a/ports/esp32/esp32_rmt.c
+++ b/ports/esp32/esp32_rmt.c
@@ -239,17 +239,14 @@ STATIC mp_obj_t esp32_rmt_write_pulses(size_t n_args, const mp_obj_t *pos_args,
         if (loop_en) {
             check_esp_err(rmt_set_tx_intr_en(self->channel_id, true));
             check_esp_err(rmt_set_tx_loop_mode(self->channel_id, false));
+            check_esp_err(rmt_wait_tx_done(self->channel_id, portMAX_DELAY));
         }
-        check_esp_err(rmt_wait_tx_done(self->channel_id, portMAX_DELAY));
         check_esp_err(rmt_set_tx_intr_en(self->channel_id, false));
+        check_esp_err(rmt_set_tx_loop_mode(self->channel_id, true));
     }
 
     check_esp_err(rmt_write_items(self->channel_id, self->items, num_items, false /* non-blocking */));
 
-    if (self->loop_en) {
-        check_esp_err(rmt_set_tx_loop_mode(self->channel_id, true));
-    }
-
     return mp_const_none;
 }
 STATIC MP_DEFINE_CONST_FUN_OBJ_KW(esp32_rmt_write_pulses_obj, 2, esp32_rmt_write_pulses);
```

[Note that this also moves the `rmt_wait_tx_done()` so that it is only done if a previous looping chain of pulses is in progress – this feels more right than the way I was doing it before]

Excitingly, this seems to work fine against MicroPython as of HEAD today (b3a34dde36485581a89de105ca7c0dd0d0fe49bb).

The thing I'm not certain of is whether this has the semantics I want: which is to completely finish any previous chain of pulses and then start the new looping chain without any cruft from the old pulse buffer being jammed in-between. I'm going to do some tests with `clock_div` ratcheted way up and some long chains, but if they work OK then I'm going to open a PR for this change in a bit.

jonathanhogg, 2021-06-22:
Hmmm… two things:

- `rmt_wait_tx_done()` *absolutely needed to be where it was* or writing a looping train immediately after writing a non-looping train hangs (don't try to do two things at once, Jonathan);
- the semantics of this fix *are* broken: starting a looping train emits a single pulse from whatever was in the old buffer before the new train begins.

This is all feeling horribly familiar from last year…

I've stared a bit more at the IDF code to try and understand how looping works and I've got a plan: when looping is enabled, don't use `rmt_write_pulses()`. Instead, after stopping the previous train, fill the memory buffer by hand using `rmt_fill_tx_items()` and then enable hardware looping:

```diff
diff --git a/ports/esp32/esp32_rmt.c b/ports/esp32/esp32_rmt.c
index 9619b1dd5..5087246af 100644
--- a/ports/esp32/esp32_rmt.c
+++ b/ports/esp32/esp32_rmt.c
@@ -241,14 +241,17 @@ STATIC mp_obj_t esp32_rmt_write_pulses(size_t n_args, const mp_obj_t *pos_args,
             check_esp_err(rmt_set_tx_loop_mode(self->channel_id, false));
         }
         check_esp_err(rmt_wait_tx_done(self->channel_id, portMAX_DELAY));
-        check_esp_err(rmt_set_tx_intr_en(self->channel_id, false));
-    }
 
-    check_esp_err(rmt_write_items(self->channel_id, self->items, num_items, false /* non-blocking */));
+        check_esp_err(rmt_fill_tx_items(self->channel_id, self->items, num_items, 0));
+        rmt_item32_t stop_data = {0};
+        check_esp_err(rmt_fill_tx_items(self->channel_id, &stop_data, 1, num_items));
 
-    if (self->loop_en) {
+        check_esp_err(rmt_set_tx_intr_en(self->channel_id, false));
         check_esp_err(rmt_set_tx_loop_mode(self->channel_id, true));
     }
+    else {
+        check_esp_err(rmt_write_items(self->channel_id, self->items, num_items, false));
+    }
 
     return mp_const_none;
 }
```

Scarily, this just works without any actual call to start the RMT channel. Note that it requires inserting a sentinel value at the end of the filled items – `rmt_write_items()` does this automatically.

This fix seems to do everything I want:

- doesn't explode with the original example;
- stops an existing looping train before starting a new one;
- doesn't insert any junk at the beginning of a new looping train from the old memory buffer.

And, as a new bonus, it will throw an error if you try to start a looping train that won't fit in a single RMT memory block. The old code, now that I look at it, would have just done Bad Things in this scenario since hardware looping only works with a single block. I've had to be careful with the order that operations are done so that if it does raise an error it doesn't leave the channel in a half-way state.

It's a fairly ugly internal error:

```
>>> rmt.write_pulses([8000, 8000] * 100)
E (388029) rmt: rmt_fill_tx_items(633): RMT WR MEM OVERFLOW
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-258, 'ESP_ERR_INVALID_ARG')
```

@dpgeorge: is that bad? I could test the `rmt_fill_tx_items()` return code and raise some other error if you like? There is, unfortunately, no way that I can see to check the upper size of an RMT memory block without going below the official API and I'm loath to do that and risk making the code brittle to future IDF changes.

jonathanhogg, 2021-06-22:
Oh man! This is painfully close, but it breaks the TX semaphore logic. So `wait_done()` will now return `True` when a looping train is in motion. This is probably liveable with – although it upsets me – but worse is that the `rmt_wait_tx_done()` when switching between two different looping trains won't block and so the second will just rudely interrupt the first.

Looking at the code, only `rmt_write_items()` correctly handles the semaphore: it does a take/give at the beginning to wait for the last TX to end and then takes again before starting the new one – the TX interrupt then gives it back when it fires. The obvious/helpful thing would have been to put the semaphore take in `rmt_tx_start()`, which `rmt_write_items()` calls…

However, that `rmt_tx_start()` thing got me thinking as there's no equivalent stop after doing a `rmt_write_items()`. The TX complete interrupt doesn't seem to touch the same registers that `rmt_tx_start()` does. So I've tried going back to the old code, but putting a call to `rmt_tx_stop()` before enabling looping and calling `rmt_write_items()`.

```diff
diff --git a/ports/esp32/esp32_rmt.c b/ports/esp32/esp32_rmt.c
index 9619b1dd5..ceeb4d27c 100644
--- a/ports/esp32/esp32_rmt.c
+++ b/ports/esp32/esp32_rmt.c
@@ -218,6 +218,10 @@ STATIC mp_obj_t esp32_rmt_write_pulses(size_t n_args, const mp_obj_t *pos_args,
     mp_obj_get_array(pulses, &pulses_length, &pulses_ptr);
 
     mp_uint_t num_items = (pulses_length / 2) + (pulses_length % 2);
+    if (self->loop_en && num_items > 63) {
+        mp_raise_ValueError(MP_ERROR_TEXT("Too many pulses for loop"));
+    }
+
     if (num_items > self->num_items) {
         self->items = (rmt_item32_t *)m_realloc(self->items, num_items * sizeof(rmt_item32_t *));
         self->num_items = num_items;
@@ -242,14 +246,11 @@ STATIC mp_obj_t esp32_rmt_write_pulses(size_t n_args, const mp_obj_t *pos_args,
         }
         check_esp_err(rmt_wait_tx_done(self->channel_id, portMAX_DELAY));
         check_esp_err(rmt_set_tx_intr_en(self->channel_id, false));
-    }
-
-    check_esp_err(rmt_write_items(self->channel_id, self->items, num_items, false /* non-blocking */));
-
-    if (self->loop_en) {
+        check_esp_err(rmt_tx_stop(self->channel_id));
         check_esp_err(rmt_set_tx_loop_mode(self->channel_id, true));
     }
 
+    check_esp_err(rmt_write_items(self->channel_id, self->items, num_items, false /* non-blocking */));
     return mp_const_none;
 }
 STATIC MP_DEFINE_CONST_FUN_OBJ_KW(esp32_rmt_write_pulses_obj, 2, esp32_rmt_write_pulses);
```

This seems to work as far as I can make out: the old `wait_done()` semantics are restored and there are no extraneous pulses emitted when starting a new looping train. So I guess this is stopping the hardware firing back up when looping is enabled until `rmt_write_items()` calls `rmt_tx_start()` again. However, I'm back at home without scope, so I'm just testing with the slowest pulses I can do and an LED. I'll wait until I can test properly before declaring this done.

I've also added an explicit check on the length of the pulses list/tuple if looping is enabled. The size of the RMT memory blocks seems to be pretty clearly documented and hardware limited, so I reckon it's safe to hard-code in the upper limit.

jonathanhogg, 2021-06-24:
No, `rmt_tx_stop()` doesn't work – looks like that was poor squinting at the LED and wishful thinking.

However, I've taken the time to actually look at the IDF code and figure out what's happening and why it changed.

`rmt_write_items()` calls `rmt_tx_start()` at the end. This function has changed implementation between IDF v4.0.2 and v4.2 (presumably actually in v4.1): if looping is turned off, it now *enables the TX interrupt*. This is why turning on looping after calling `rmt_write_items()` crashes: the interrupt is re-enabled and this causes an interrupt storm that takes down the device.

So I've gone back to the old code (again) and just moved the TX interrupt disable call to *after* calling `rmt_write_items()`, at the same time as I enable the looping:

```diff
diff --git a/ports/esp32/esp32_rmt.c b/ports/esp32/esp32_rmt.c
index 9619b1dd5..14e3fa96d 100644
--- a/ports/esp32/esp32_rmt.c
+++ b/ports/esp32/esp32_rmt.c
@@ -218,6 +218,10 @@ STATIC mp_obj_t esp32_rmt_write_pulses(size_t n_args, const mp_obj_t *pos_args,
     mp_obj_get_array(pulses, &pulses_length, &pulses_ptr);
 
     mp_uint_t num_items = (pulses_length / 2) + (pulses_length % 2);
+    if (self->loop_en && num_items > 63) {
+        mp_raise_ValueError(MP_ERROR_TEXT("Too many pulses for loop"));
+    }
+
     if (num_items > self->num_items) {
         self->items = (rmt_item32_t *)m_realloc(self->items, num_items * sizeof(rmt_item32_t *));
         self->num_items = num_items;
@@ -241,12 +245,12 @@ STATIC mp_obj_t esp32_rmt_write_pulses(size_t n_args, const mp_obj_t *pos_args,
             check_esp_err(rmt_set_tx_loop_mode(self->channel_id, false));
         }
         check_esp_err(rmt_wait_tx_done(self->channel_id, portMAX_DELAY));
-        check_esp_err(rmt_set_tx_intr_en(self->channel_id, false));
     }
 
     check_esp_err(rmt_write_items(self->channel_id, self->items, num_items, false /* non-blocking */));
 
     if (self->loop_en) {
+        check_esp_err(rmt_set_tx_intr_en(self->channel_id, false));
         check_esp_err(rmt_set_tx_loop_mode(self->channel_id, true));
     }
 
```

This fixes the crashing behaviour, fully retains the old looping semantics, works under v4.0.2, and – more importantly – I actually understand *why* instead of just stabbing in the dark.

I'm going to open a PR.

jonathanhogg, 2021-06-24:
I've also just tested this fix against IDF 4.3b2 and it seems to work fine – no significant changes in the underlying API code that I can see other than a mass of internal renaming.

jonathanhogg, 2021-06-24:
@dpgeorge: I'm close to having a new version of #6274 ready taking into account your notes so I'm happy to roll this fix into that or you can take #7453 now since it's currently broken in mainline. A side-effect of this IDF interrupt-enable change is that looping will currently exhibit the old spurious-pulse bug that I put all that effort into fixing last year.

dpgeorge, 2021-06-25:
> I'm close to having a new version of #6274 ready taking into account your notes so I'm happy to roll this fix into that or you can take #7453 now since it's currently broken in mainline

I'll merge #7453 first.  It's a good independent fix.

> A side-effect of this IDF interrupt-enable change is that looping will currently exhibit the old spurious-pulse bug that I put all that effort into fixing last year.

Oh dear!  Maybe we need to make `loop` a kwarg in the `RMT` constructor so that it can be configured as looping right away, rather than doing it later via a separate method call?

dpgeorge, 2021-06-25:
Fixed by 30422ca7c24f02ea34fdbc61466d5fb4baa0af8a.  Thanks for the comprehensive debugging!

closed 2021-06-25 by dpgeorge

jonathanhogg, 2021-06-25:
> Oh dear! Maybe we need to make `loop` a kwarg in the `RMT` constructor so that it can be configured as looping right away, rather than doing it later via a separate method call?

This fix restores the correct behaviour so I'm happy leaving the API as-is for the moment. However, there'd be no harm in adding a new keyword argument to the constructor to save having to call the `loop()` method before `write_pulses()`… I'll add it to my list 😉