micropython/micropython #7404: esp32: Hardware SPI class initialiser doesn't support None for unused signals

jonathanhogg, 2021-06-16:
I'm using hardware SPI on ESP32 without the `sck` or `miso` signals (both interfaces). When initialising the peripheral, this is an error:

```
spi = SPI(1, baudrate=1000000, sck=None, mosi=Pin(12), miso=None)
```

because you're not allowed to specify `None` for the `sck` and `miso` parameters.

However this is A-OK:

```
spi = SPI(1)
spi.init(baudrate=1000000, sck=None, mosi=Pin(12), miso=None)
```

and does exactly what I want: internally the `None` is translated to a `-1` signal specification for the `spi_bus_config_t` structure to indicate that the signal is not used.

The problem with the latter is that it grabs and initialises the default pins first before reconfiguring. I would prefer that SPI left these pins alone as I need them for other purposes and this forces me into a careful initialisation order dance to make sure my pins are all correct at the end.

It doesn't look like a big fix, so I'm happy to put in my own PR if there's agreement on aligning these APIs.

robert-hh, 2021-06-16:
If you drop spi.init and provide the paramters in the initializer, the default pins are not touched.
```
spi = SPI(1, baudrate=1000000, sck=None, mosi=Pin(12), miso=None)
```

robert-hh, 2021-06-16:
You're right. Looking into the code, it seems not to follow the common make_new(), init() and init_helper() structure. In that case indeed a code change for your application is needed.

jonathanhogg, 2021-06-16:
I'm just prepping a PR now (having to pull up to HEAD to fix a build failure)

jonathanhogg, 2021-06-16:
Have just submitted PR #7405

This is sufficient to do what I need, but let me know if you want a bigger refactor of this code to fit with a different structure.

jonathanhogg, 2021-06-17:
Fix merged from #7405

closed 2021-06-17 by jonathanhogg