micropython/micropython #7406: Getting 00:00 when time.localtime() called inside main.py

ProgrammingPlanet, 2021-06-16:
When my esp8266 board boots and time.localtime() or utime.localtime() runs inside main.py, than the hh:mm is 00:00.

But when I call same function from webrepl console it give me actual time.

Board - ESP 12f, esp8266
Micropython version: 1.15

peterhinch, 2021-06-16:
This is really a query for [the forum](http://forum.micropython.org/index.php), not this bug tracker. 

After a power cycle the RTC will not be initialised.  Some tools such as **rshell** and (I guess) **webrepl** do this for you to aid in maintaining the filesystem. In a stand-alone application you need to do it in code, e.g. by querying an NTP server.

ProgrammingPlanet, 2021-06-16:
How to do it in my micropython code such as in boot.py file.

dpgeorge, 2021-06-17:
You need to initialise the RTC, see here for some examples: http://docs.micropython.org/en/latest/esp8266/quickref.html#real-time-clock-rtc

closed 2021-06-17 by dpgeorge

ProgrammingPlanet, 2021-06-17:
I have problem that it needs internet connection to setup actual clock time. Or when I set it custom I don't know which time I set on bootup.

I need to know that how webrepl is setting the time without internet connection.

Can anyone help please?

dpgeorge, 2021-06-17:
I'm pretty sure webrepl does not set the time.  But if you have webrepl working then it must have internet connection, so it's probably using `ntptime.settime()`

ProgrammingPlanet, 2021-06-17:
I make it sure last night that my board wasn't connected to internet and from webrepl console, I was getting correct time. But I don't know how webrepl setup the correct localtime without net connection.