micropython/micropython #7414: micropython 1.16 SD won't mount after umount

emard, 2021-06-18:
the SD card doesn't crash after umount but it won't mount again
```
>>> import os,machine
>>> sd=machine.SDCard(slot=3)
>>> os.mount(sd,"/sd")
>>> os.listdir("/sd")
['KERNEL.SYS', 'COMMAND.COM', 'INVADERS.COM', 'CTMOUSE.EXE', 'kick.rom', 'OSD_CA01.sys', 'amiga', 'colecovision', 'hdl4fpga', 'linux', 'm68k', 'mp3', 'msx', 'nes', 'next186', 'orao', 'ql', 'scope', 'sega_sg1000', 'rootfs.cpio.uboot', 'uImage', 'uboot.env.defunct', 'dtb', 'slides', 'ti99_2', 'ti99_4a', 'vic20', 'zxspectrum', 'snes', 'rtc', 'myamoto', 'apple2', 'speak', 'c64']
>>> umount("/sd")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'umount' isn't defined
>>> os.umount("/sd")
>>> sd.deinit()
>>> os.listdir("/sd")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
>>> sd=machine.SDCard(slot=3)
E (70971) spi: spi_bus_initialize(462): SPI bus already initialized.
E (70971) sdspi_host: spi_bus_initialize failed with rc=0x103

```

saul530, 2021-10-07:
I'm using Latest build from lv_micropython repo:
https://github.com/lvgl/lv_micropython

And I'm still seeing the same issue:
`
MicroPython v1.17-571-gec6a9fca1-dirty on 2021-10-06; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
>>> import uos
>>> from machine import SDCard
>>> sd = SDCard(slot=2)
>>> uos.mount(sd, '/sd')
>>> uos.listdir('/')
['sd', 'boot.py', 'button.py', 'lib', 'tabs.py']
>>> uos.umount('/sd')
>>> sd.deinit()
>>> sd = SDCard(slot=2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: (-259, 'ESP_ERR_INVALID_STATE')
>>> 
`
If I would like my device to handle a user removing an SD card then reinserting it. Is there a better way to re-initialize the SD card if I know I'm using the same SPI bus?

jonnor, 2024-09-08:
Is this issue still present on MicroPython 1.23 with latest sdcard driver?

projectgus, 2025-02-05:
I think this was most likely fixed in [#9147](https://github.com/micropython/micropython/pull/9147). I'm not able to reproduce any more.

closed 2025-02-05 by projectgus