micropython/micropython #7415: mpremote: incorrect recursive copying of files under WINDOWS

IhorNehrutsa, 2021-06-18:
I try to copy directory to device ESP32 MicroPython v1.15
python.exe mpremote.py connect port:COM8 cp -r IN88 :
```
>ver
Microsoft Windows [Version 10.0.19042.867]

>python.exe mpremote.py connect port:COM8 cp -r IN88 :
cp IN88\angles.py :IN88/IN88\angles.py
cp IN88\boot.py :IN88/IN88\boot.py
cp IN88\esp32_.py :IN88/IN88\esp32_.py
cp IN88\heli.py :IN88/IN88\heli.py
```
Destination file names are incorrect:
cp **IN88**\angles.py :**IN88/IN88**\angles.py

Note: Device os.sep is '/' and WINDOWS os.sep is '\\'

View in Thonny IDE: IN88 directory is Ok, file names are incorrect:
[image]

Thank you very much.

spacemanspiff2007, 2021-06-18:
I have the same problem: It's not possible to copy modules.

maruno, 2021-07-07:
I get the same behaviour, mpremote also can't delete the files anymore itself, getting on OSError

```
$ mpremote.exe ls
ls :
           0 .\lib\logging/
         139 boot.py
$ mpremote.exe rmdir ".\lib\logging/"
rmdir :.\lib\logging/
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
OSError: 39
```

dpgeorge, 2022-03-07:
Fixed by 1f84440538a017e463aaad9686831ce9527122b5

closed 2022-03-07 by dpgeorge