micropython/micropython #7416: Issues with STM32WB port

mtk11, 2021-06-18:
I am trying to use micropython with STM32WB55CE (512KB flash). Although I created a custom stm32wb55xe.ld with the following settings:

    FLASH (rx)      : ORIGIN = 0x08000000, LENGTH = 432K /* sectors 0-127 */
    FLASH_APP (rx)  : ORIGIN = 0x08004000, LENGTH = 416K /* sectors 4-127 */
    FLASH_FS (r)    : ORIGIN = 0x080A0000, LENGTH = 312K /* sectors 128-191 */

which suppose to leave space in the middle of the flash for FUS and wireless stack firmwares, just by pressing in STM32CubeProgrammer – Read FUS infos button, makes micropython not operational. As well as I tried changing the size of FLASH_FS, but it keeps creating a 239 KB flash drive. 

Any other config that needs to be changed ?

jimmo, 2021-07-05:
@mtk11   To flash BLE stack firmware the recommended address (e.g. for `stm32wb5x_BLE_HCILayer_fw.bin`) is 0x0806C000.

So you'll need to have FLASH_APP and FLASH_FS fit into that (which looks like what you're doing with the 432k, but you'll need to make FLASH_APP smaller and FLASH_FS also finish before sector 108).

I'm not quite sure why the "Read FUS Info" button is making MicroPython not work... what exactly do you see happen?

mtk11, 2021-07-05:
@jimmo "Read FUS Info" button changes the flash content in 0x0800000 - 0x0802000 region. Apparently there is a problem with the latest FUS - v1.1.2.0, because I am not able to flash any BLE stack firmware. Have you tried this FUS version ? Similar issues ?

jimmo, 2021-07-06:
@mtk11 Can you confirm which version -- you said 1.1.2, but the latest is 1.2.0.
Are you able to replicate this on WB55xG also? (i.e. on the Nucleo or UsbDongle boards from ST)

Note if you're using this chip, be aware of 2668337f36deef7fdd97f35e2efc68a5d2102192

We did most of our testing on FUS 1.1.0 (and our WS updater still expects this version). We haven't spent any more time looking at more recent firmware (FUS or WS) while we wait for a fix for the above issue.

mtk11, 2021-07-06:
@jimmo I can not install FUS v1.1.2 via STM32CubeProgrammer, it shows the following:

_Error: Could not execute firmware upgrade command, Impossible to install FUS V1.0.1 on 256/512KB Flash Parts_

Not even able to downgrade the FUS version:

_Error: The FUS version you're trying to install is older/equal to the FUS version already installed !_

Have no WB55xG, intend to order some …

I tried to use before rfcore_firmware.py, but it did not work. What suppose to be modified to update just the FUS ?

jimmo, 2021-07-06:
> @jimmo I can not install FUS v1.1.2 via STM32CubeProgrammer, it shows the following:

@mtk11 so which version of the FUS do you have installed?

Honestly as long as you've managed to get it to 1.1.0 there's not a lot of reasons to update past that. (I don't believe that any of the WS binaries require a later FUS version).

> I tried to use before rfcore_firmware.py, but it did not work.

What do you mean by "did not work" ?

You will need to modify at least the target addresses. `STAGING_AREA_START`, `_ADDR_FUS`, and `_ADDR_WS_BLE_HCI`.

But note that this has not been tested on 512k parts.

It is difficult to debug this script unfortunately... as you've already noticed you only get one attempt at the FUS update. We invested considerable time (and a large number of Nucleo boards) to get it working reliably and deal with all the edge cases.

> What suppose to be modified to update just the FUS ?

The updater will only update the files you give it. So if only the fus binaries are on the filesystem, then that's all that will be updated. However, unless you have external flash, I'm not sure you'll have enough space on the fileystem for the binaries on the 512k part.

FWIW, my personal opinion, unless you're prepared to invest a bit of time understanding this process and all the quirks of the way ST have implemented the WS updating, perhaps consider a different microcontroller. One good option is to use a separate nRF52xxx running Zephyr or NimBLE HCI firmware (which MicroPython can just use as an HCI UART).

mtk11, 2021-07-06:
@jimmo 
 
> @mtk11 so which version of the FUS do you have installed?

FUS v1.1.2 

> Honestly as long as you've managed to get it to 1.1.0 there's not a lot of reasons to update past that. (I don't believe that any of the WS binaries require a later FUS version).

[STM32WB55 Nucleo - FUS_STATE_IMG_NOT_FOUND error](https://electronics.stackexchange.com/questions/532179/stm32wb55-nucleo-fus-state-img-not-found-error) suggests downgrading the FUS.

> What do you mean by "did not work" ?

I had no serial connection to be more specific ...

> The updater will only update the files you give it. So if only the fus binaries are on the filesystem, then that's all that will be updated. However, unless you have external flash, I'm not sure you'll have enough space on the fileystem for the binaries on the 512k part.

STM32WB5xxE has actually 1MB flash, it is accessible, my FLASH_FS config:

    FLASH_FS (r)    : ORIGIN = 0x080A0000, LENGTH = 256K

If it is not too much trouble, can you describe how manually the FUS can be updated with rfcore_firmware.py ?

> FWIW, my personal opinion on the STM32WB family (after a reasonable amount of work implementing it into some products) -- unless the implementer is prepared to invest a bit of time understanding this process and all the quirks of the way ST have implemented the WS updating, I would recommend perhaps consider a different microcontroller. If you want to stick with STM32, one good option is to use a separate nRF52xxx running Zephyr or NimBLE HCI firmware (which MicroPython can just use as an HCI controller via the UART interface).

Do not really intend to stick with STM32 – currently most of the parts are not available as well as there is noticeable price increase. 

The product I am working on is a handheld spectrometer – really trying to use as much as possible compact components. Like the STM32WB55 ADC, but probably can also use nrf52840 or nrf52833 with external ADC. However there are some problems with nrf Micropython port. The USB serial port is shown under Linux after more than 40 s. Also, the nrf port implementation seems as not as well developed. In addition, I can not get the BLE working. For example, ubluepy_eddystone.py doesn’t show on my smartphone any Bluetooth device etc.

jimmo, 2021-07-06:
> @jimmo
>
>
> FUS v1.1.2
>
>
> [STM32WB55 Nucleo - FUS_STATE_IMG_NOT_FOUND error](https://electronics.stackexchange.com/questions/532179/stm32wb55-nucleo-fus-state-img-not-found-error) suggests downgrading [...]

I think this is a typo -- they meant 1.1.0 not 1.0.1 (there is no 1.0.1 version).

So I think they're saying you need to _upgrade_ the factory firmware first (not _downgrade_, which as far as I'm aware isn't possible for.the FUS). This also matches my experience -- you definitely need to get to FUS v1.1.0 before installing WS binaries (also to avoid bricking your chip).

> 
> > What do you mean by "did not work" ?
> 
> I had no serial connection to be more specific ...

How did you restore the device afterwards? Did you make any changes to rfcore_firmware.py first? (e.g. the target addresses)

> 
> STM32WB5xxE has actually 1MB flash, it is accessible, my FLASH_FS config:

The ST product page https://www.st.com/content/st_com/en/products/microcontrollers-microprocessors/stm32-32-bit-arm-cortex-mcus/stm32-wireless-mcus/stm32wb-series/stm32wbx5.html suggests that xE is all 512k.

> 
> If it is not too much trouble, can you describe how manually the FUS can be updated with rfcore_firmware.py ?

Can you let me know which parts of the existing instructions don't make sense or need to be clarified and I can update them. See the the top of rfcore_firmware.py and rfcore_makefirmware.py. You generate the firmware files from the ST provided files using rfcore_makefirmware.py (see the scripts for why this transformation of the files is necessary), copy the files you want to update to the device filesystem (e.g. using mpremote or USB), then invoke the updater from boot.py and your application. Note that this means you need enough space in your filesystem for the firmware images.

You can make this work OTA by copying the files over BLE (e.g. OTS or something simpler) from your phone app.

I think the key thing (other than updating the addresses) is that to just update the FUS you only need to copy the FUS binaries (if there is no WS binary it with just skip that step). Also you must provide both 1.0.2 and 1.1.0 (you can't go from factory firmware directly to 1.1.0). As I mentioned before, this script hasn't been updated for FUS v1.1.2 or v1.2 or later.

> However there are some problems with nrf Micropython port. The USB serial port is shown under Linux after more than 40 s. Also, the nrf port implementation seems as not as well developed. In [...]

Yeah my suggestion is to run MicroPython on a STM32L4 or F4 etc, then have a dedicated nRF (or other BLE HCI controller IC) just doing the HCI controller. Updating the nRF port is definitely on the to-do list though!

mtk11, 2021-07-06:
@jimmo

> How did you restore the device afterwards? Did you make any changes to rfcore_firmware.py first? (e.g. the target addresses)

I did make the requested changes to rfcore_firmware.py. At some point had the chip bricked, but managed to recover it with the OpenOCD script at the end of this topic: [Wireless Stack recovery for STM32WB](https://community.st.com/s/question/0D53W00000fnOhS/wireless-stack-recovery-for-stm32wb). Right after that installed a new FUS with STM32CubeProgrammer. 

> The ST product page https://www.st.com/content/st_com/en/products/microcontrollers-microprocessors/stm32-32-bit-arm-cortex-mcus/stm32-wireless-mcus/stm32wb-series/stm32wbx5.html suggests that xE is all 512k.

I presume, the whole STM32WB55 series uses the same silicon, STM just applied different software settings.

> Can you let me know which parts of the existing instructions don't make sense or need to be clarified and I can update them.

For example, _fus_run_fwupgrade(_ADDR_FUS) initially returned: (0,0), after that (255,0) ?

> I think the key thing (other than updating the addresses) is that to just update the FUS you only need to copy the FUS binaries (if there is no WS binary it with just skip that step). Also you must [...]

In the latest [STM32CubeWB](https://github.com/STMicroelectronics/STM32CubeWB) - [Release v1.11.1](https://github.com/STMicroelectronics/STM32CubeWB/releases/tag/v1.11.1) the size of stm32wb5x_FUS_fw_1_0_2.bin is 0 Bytes. STM [AN5185](https://www.st.com/resource/en/application_note/dm00513965-st-firmware-upgrade-services-for-stm32wb-series-stmicroelectronics.pdf): p.4 FUS version: V1.2.0 - Allows direct update from FUS V0.5.3 to FUS V1.2.0 (without installing intermediate FUS versions)

I guess what happened in my case is: [AN5185](https://www.st.com/resource/en/application_note/dm00513965-st-firmware-upgrade-services-for-stm32wb-series-stmicroelectronics.pdf): p.13: 

Before activating Anti-rollback, ensure that a wireless stack is correctly installed, and that it has not been deleted. If it is activated without any wireless stack installed, the FUS registers 0xFFFFFFFF as new version, and it is not possible to install any wireless stack.

Any way I can deactivate Anti-rollback ?

> Yeah my suggestion is to run MicroPython on a STM32L4 or F4 etc, then have a dedicated nRF (or other BLE HCI controller IC) just doing the HCI controller. Updating the nRF port is definitely on the to-do list though!

Regarding nrf port, [Circuitpython](https://github.com/adafruit/circuitpython) seems better maintained, but regarding BLE connectivity  - no module ubluepy ?

Will look at other microcontrollers and BLE ICs.

jimmo, 2021-07-07:
> @jimmo
>
>
> I did make the requested changes to rfcore_firmware.py. At some point had the chip bricked, but managed to recover it with the OpenOCD script at the end of this topic: [Wireless Stack [...]

No, they have different flash and RAM sizes.

> 
> > Can you let me know which parts of the existing instructions don't make sense or need to be clarified and I can update them.
> 
> For example, _fus_run_fwupgrade(_ADDR_FUS) initially returned: (0,0), after that (255,0) ?

You're not supposed to call `_fus_run_fwupgrade` directly. The whole process needs to be sequenced correctly, and this is just one of the stages.

See the instructions at the top of the file, starting with the heading "# To perform a firmware update:"... The only functions you should invoke is `rfcore_firmware.resume()` in your boot.py (step 3 in the instructions) and `rfcore_firmware.check_for_updates()` from your application (or the REPL) to start the update procedure (step 4).

> In the latest [STM32CubeWB](https://github.com/STMicroelectronics/STM32CubeWB) - [Release v1.11.1](https://github.com/STMicroelectronics/STM32CubeWB/releases/tag/v1.11.1) the size of [...]

Yep, the direct from factory-to-latest is new since 1.2.0. But like I said, our firmware updater was written and tested against FUS 1.0.2 and 1.1.0.

> I guess what happened in my case is: [AN5185](https://www.st.com/resource/en/application_note/dm00513965-st-firmware-upgrade-services-for-stm32wb-series-stmicroelectronics.pdf): p.13:
>
> Before [...]

Unfortunately not that I know of. This chip is difficult to work with and has a few traps (I have several bricked ones myself... like I said, we're very much looking forward to the firmware update that will fix 2668337f36deef7fdd97f35e2efc68a5d2102192).

> Regarding nrf port, [Circuitpython](https://github.com/adafruit/circuitpython) seems better maintained, but regarding BLE connectivity - no module ubluepy ?

`ubluepy` is a specific feature for the nRF port of MicroPython. On STM32, ESP32 and Unix, we have the low-level `ubluetooth` (and the high-level `aioble`) modules. We plan to move nRF to `ubluetooth` too.

CircuitPython have their own BLE modules (`_bleio` and `adafruit_ble` I think they're called)

> 
> Will look at other microcontrollers and BLE ICs.

mtk11, 2021-07-09:
> Unfortunately not that I know of. This chip is difficult to work with and has a few traps (I have several bricked ones myself... like I said, we're very much looking forward to the firmware update [...]

I also have support request regarding the raised issues, but STM so far has not provided solution. I expected much better quality and level of tech support for such relatively high priced product.

> `ubluepy` is a specific feature for the nRF port of MicroPython.

Managed to run properly the micropython nRF port on a nRF52840 module. Apparently, since I do not have an external 32 kHz crystal, in needs the following: CFLAGS += -DBLUETOOTH_LFCLK_RC.

The nRF port seems to work much faster – 784 pystone/sec vs 598  pystone/sec on STM32WB55.

Micropython support for the SPIM — Serial peripheral interface master with EasyDMA ? Also, looked at the [micropython examples](https://github.com/micropython/micropython/tree/master/examples), but was not able to find examples for controlling the peripherals in C/C++ - Pins, PWM, Timers … Any available ?

closed 2021-08-25 by mtk11