micropython/micropython #7426: RP2: inconsistent response to pin interrupts

peterhinch, 2021-06-19:
This was reported [here](https://forum.micropython.org/viewtopic.php?f=21&t=10642&start=10#p58898. The test script was the outcome of discussion and is a minimal test case. The user asked me to raise the issue as he lacks GitHub experience.

The test script uses PWM to produce a square wave on pin 16, which must be linked to pin 14. The latter has an ISR which pulses pin 13. The scope traces show that the square wave is uninterrupted, but that there are periods where the ISR is not being called.

kadamski, 2021-06-20:
Does the ISR start working again after some period of silence or it's more like - it works for some time, then it stops working?

peterhinch, 2021-06-20:
From the forum thread the interruptions are for a period of about 8ms and occur at unpredictable intervals. During these periods the RP2 is unresponsive to interrupts but PWM continues to run. The script does not stop running.

These measurements were done by @JWF, so he is the person to query about them; he is not a GitHub user but may be found in the forum.

kadamski, 2021-06-21:
Ok, i have reproduced the issue. The "dropout" happens from time to time and there is a short period of time when the events do not occur, after which it fixes itself for some time again. Now, look how the events look just after the drop out:

[image]

First 12 events are of the much higher frequency than the PWM signal, just like some bottleneck was unchoked and some number of queued events were popped up. Then, the signal comes back to one event per PWM cycle.

kadamski, 2021-06-21:
@dpgeorge any idea what may be blocking scheduling events for 8ms every ~10000 of interrupts on rp2 in this case?

kadamski, 2021-06-22:
Ok, that seems to be related to (*_) in the READ_ADC argument list. I didn't notice it at the first glance, but obviously this creates a tuple with 1 element every time the function (interrupt handler) is called. This tuple is called _ which by convention means we don't care about it, but in practice it is still a normal, working tuple that can be used and as such, requires allocating memory.

@dpgeorge  please correct me if I'm wrong here. Maybe we should update the isr_rules.rst to warn about usage of *args and **kwargs in the handler function?

peterhinch, 2021-06-22:
Well spotted! Arguably any allocation in an ISR should throw an exception as per STM. Please see https://github.com/micropython/micropython/issues/7433. Here is the outcome of similar code on a Pyboard:
```python
=== from pyb import Timer
=== def cb(*_):
===     pass
=== 
=== t = Timer(1)
=== t.init(freq=1000, callback=cb)
=== 
>>> uncaught exception in Timer(1) interrupt handler
MemoryError: 

```
Even I might have spotted the bug if this had been the outcome :)

kadamski, 2021-06-22:
But the Pin.irq() on RP2 by default runs with hard=0, meaning the actual handler is being run by mp_sched_schedule() (basically the same as the micropython.schedule() in python world), not in hard irq context, I'm not sure how that works on STM32, but I guess the allocation is not a problem if the IRQ is not hard.

peterhinch, 2021-06-23:
I have reported the conclusions back to the original poster in [this forum post](https://forum.micropython.org/viewtopic.php?f=21&t=10642&start=10#p59023). Namely set **hard=True** and change the ISR function declaration to avoid creating a tuple. Hopefully he will try both the test script and his application, and report back.