micropython/micropython #7427: micropython 1.16 Cannot connect to BLE in peripheral mode

vanminh0910, 2021-06-19:
I used this BLE example https://github.com/micropython/micropython/blob/master/examples/bluetooth/ble_uart_repl.py and it worked in v1.15 so far. After flashing the new release v1.16, I can not connect from my app or web any more. It cannot retrieve services. Here is the javascript code I used to connect (using web bluetooth).

`navigator.bluetooth.requestDevice({
      acceptAllDevices: true,
	  optionalServices: ["6e400001-b5a3-f393-e0a9-e50e24dcca9e"]})
  .then(device => {
    console.log('Connecting to GATT Server...');
    return device.gatt.connect();
  })
  .then(server => {
    // Note that we could also get all services that match a specific UUID by
    // passing it to getPrimaryServices().
    console.log(server);
	console.log('Getting Services...');
    return server.getPrimaryServices();
  })
  .then(services => {
	console.log(services);
    console.log('Getting Characteristics...');
    let queue = Promise.resolve();
    services.forEach(service => {
      queue = queue.then(_ => service.getCharacteristics().then(characteristics => {
        console.log('> Service: ' + service.uuid);
        characteristics.forEach(characteristic => {
          console.log('>> Characteristic: ' + characteristic.uuid + ' ' +
              getSupportedProperties(characteristic));
        });
      }));
    });
    return queue;
  })
  .catch(error => {
    console.log('Argh! ' + error);
  });`

The console log is as below:

Connecting to GATT Server...
VM393:11 BluetoothRemoteGATTServer {device: BluetoothDevice, connected: true}
VM393:12 Getting Services...
VM393:31 Argh! NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.

After I flashed back to v1.15, it works fine again.

dpgeorge, 2021-06-20:
What port and board are you using?

vanminh0910, 2021-06-20:
Sorry for missing that key info. I am using custom design esp32 board using esp-wroom-32 4mb and generic micropython port for esp32.

dpgeorge, 2021-06-20:
What IDF version do you use?  I tried with IDF 4.2 and indeed there seems to be a problem retrieving the GATT services (nRF Connect can connect but can't get the services/characteristics).  Using IDF 4.0.2 works fine.

vanminh0910, 2021-06-20:
I used the ready built firmware from the official link: https://micropython.org/resources/firmware/esp32-20210618-v1.16.bin. It just mentions "Firmware with ESP-IDF v4.x" in the download page so I don't know the exact IDF version.

dpgeorge, 2021-06-21:
Releases prior to v1.16 used IDF v4.0.2.  But v1.16 and above use IDF v4.2.  So the issue lies there, in the IDF version.

Carglglz, 2021-06-21:
@dpgeorge you are right https://github.com/espressif/esp-idf/issues/7026, [IDF v4.2.2](https://github.com/espressif/esp-idf/releases/tag/v4.2.2) seems to solve this 👍

vanminh0910, 2021-06-22:
So will we have a patch for the v1.16 download  build with the new IDF v4.2.2 for esp32? Thanks.

dpgeorge, 2021-06-23:
The build server now uses IDF v4.2.2, and the build for v1.16 has been redone to use this.

closed 2021-06-23 by dpgeorge

vanminh0910, 2021-06-23:
I have checked and it's working well. Thanks a lot.