micropython/micropython #7428: Documentation showing a missleading version overview

The-Bow-Hunter, 2021-06-19:
When switching to an older version in the documentation and then trying to switch bag it just shows all versions up to the selected one and latest. This mmight be confusing if you get a link, quickly check and see that the version you are looking at is the last one listed. This should be updated to show all available stable versions. Even worse is that https://docs.micropython.org/en/v1.4.4/pyboard/ up to version 1.6 it doesnt show any other versions (so it is not possible to select latest)

dpgeorge, 2021-06-20:
The problem is that the build server must rebuild all old docs when a new release comes out, so the old ones know about the new release.  That's probably not very easy to do (requires setting up a virtualenv for each version).

embeddedt, 2021-06-20:
I'm not sure if this works for MicroPython, but at LVGL, we got around this by storing the version list [separately from the docs themselves](https://github.com/lvgl/docs/blob/gh-pages/versionlist.txt), and then used a [custom page template](https://github.com/lvgl/lvgl/blob/b203167c7583905e2cb4006e57a16432841a2353/docs/_templates/page.html#L22-L53) which injects the version selector at runtime. This only requires building each of the old docs versions one more time, and then they are automatically aware of all future versions.

The-Bow-Hunter, 2021-06-22:
or at least al ittle note that it is the case to ensure people know that there might be another version

The-Bow-Hunter, 2021-06-22:
It should even be possible to fetch all versions from github via js and generate the corresponding links client sided