micropython/micropython #7430: Add ESP32-S2 boards firmware to micropython.org download

UnexpectedMaker, 2021-06-20:
Hi @dpgeorge - congrats on the 1.16 release!

Is it possible to add the ESP32-S2 boards (generic, TinyS2 and FeatherS2) to
https://micropython.org/download/

So I can stop building and hosting the firmware myself?

Thanks :)

UnexpectedMaker, 2021-06-20:
Is there a place I can add them myself to an action somewhere? I took a look at `ports_esp32` and `ci.sh` but can't find where builds for TinyPICO are specifically handled, to also add my other boards.

I did find this...

```
function ci_esp32_build {
    source esp-idf/export.sh
    make ${MAKEOPTS} -C mpy-cross
    make ${MAKEOPTS} -C ports/esp32 submodules
    make ${MAKEOPTS} -C ports/esp32
    make ${MAKEOPTS} -C ports/esp32 clean
    make ${MAKEOPTS} -C ports/esp32 USER_C_MODULES=../../../examples/usercmodule/micropython.cmake FROZEN_MANIFEST=$(pwd)/ports/esp32/boards/manifest.py
    if [ -d $IDF_PATH/components/esp32s2 ]; then
        make ${MAKEOPTS} -C ports/esp32 BOARD=GENERIC_S2
    fi
}
```
So have added this... (no PR yet)

```
function ci_esp32_build {
    source esp-idf/export.sh
    make ${MAKEOPTS} -C mpy-cross
    make ${MAKEOPTS} -C ports/esp32 submodules
    make ${MAKEOPTS} -C ports/esp32
    make ${MAKEOPTS} -C ports/esp32 clean
    make ${MAKEOPTS} -C ports/esp32 USER_C_MODULES=../../../examples/usercmodule/micropython.cmake FROZEN_MANIFEST=$(pwd)/ports/esp32/boards/manifest.py
    if [ -d $IDF_PATH/components/esp32s2 ]; then
        make ${MAKEOPTS} -C ports/esp32 BOARD=GENERIC_S2
        make ${MAKEOPTS} -C ports/esp32 BOARD=UM_TINYSS2
        make ${MAKEOPTS} -C ports/esp32 BOARD=UM_FEATHERS2
    fi
}
```

But I can't see any GENERIC-S2 builds on mp.org/downloads either, so I really don't understand how this all works, unless it's all done manually?

Anything I can do to make this easier for you @dpgeorge - just let me know, but I'm kind at a loss as to how it all works ;)

dpgeorge, 2021-06-20:
The scripts to build the firmware at `micropython.org/download` are not part of this repo.  I guess it makes sense to put them here though.

I'm happy to add all S2 builds.

UnexpectedMaker, 2021-06-20:
Thanks @dpgeorge !
I was just hoping I could do some of the work to make it easier for you, hence me looking for the actions.

robert-hh, 2021-06-20:
@dpgeorge May it be possible to add builds for Teensy 4.0 and Teensy 4.1 as well, even if that is already a VERY LONG list of builds? I started to extend the document files for the mimxrt tree, and at the point where it would have pointed to a firmware download place, I just can write at the moment, that the user has to build the firmware herself (himself). Users with a MIMXRT10xx_EVK board should be able to build the firmware themselves.

dpgeorge, 2021-06-21:
See #7436 for the scripts which build the firmware.

> May it be possible to add builds for Teensy 4.0 and Teensy 4.1 as well

Yes we can do that.

dpgeorge, 2021-06-23:
The build scripts are now committed into `tools/autobuild`.  So feel free to add boards to that and make a PR.

UnexpectedMaker, 2021-06-23:
Thanks @dpgeorge !

robert-hh, 2021-06-23:
Thanks. I glanced already in it to see, how that could be done. May I ask you to spare some time for the open mimxrt PRs, especially #7400.

UnexpectedMaker, 2021-06-23:
So this is all we need to do @dpgeorge ?
```
# build the versions
do_build esp32 GENERIC FROZEN_MANIFEST=$(pwd)/boards/manifest_release.py
do_build esp32spiram GENERIC_SPIRAM FROZEN_MANIFEST=$(pwd)/boards/manifest_release.py
do_build tinypico UM_TINYPICO
do_build tinys2 UM_TINYSS2
do_build featherS2 UM_FEATHERS2
```

dpgeorge, 2021-06-23:
@UnexpectedMaker yes that should be enough.  The names like "featherS2" will turn into something like "featherS2-20210701-v1.6-gxxxxxxx.bin".

UnexpectedMaker, 2021-07-16:
Wow, I `TOTALLY` forgot about this... sorry @dpgeorge :(

Here's the PR
https://github.com/micropython/micropython/pull/7544

closed 2021-08-02 by dpgeorge