micropython/micropython #7431: When can micropython support esp32c3?

SandroChen, 2021-06-20:
In early this year, Espressif published risp-v core microcontroller esp32c3 and I wonder if you plan to support micropython on this board and when can I use it ?

dpgeorge, 2021-06-20:
MicroPython uses cmake to build the esp32 port, and builds correctly against the latest IDF v4.4 master commit, so in theory it should be able to build for esp32c3.  It just needs someone with a board to test it.

schuschu, 2021-06-21:
After commenting out modules that fail I am left with mpcross which is very much beyond my ability to fix i guess

```
merged 1 ELF section
Generated /home/user/micropython/ports/esp32/build-GENERIC_C3/bootloader/bootloader.bin
[85/85] cd /home/user/micropython/ports/esp32/build-GENERIC_C3/bootloader/esp-idf/esptool_py && /home/user/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/user/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/user/micropython/ports/esp32/build-GENERIC_C3/bootloader/bootloader.bin
Bootloader binary size 0x3f00 bytes. 0x4100 bytes (103%) free.
[1059/1278] Generating ../../genhdr/mpversion.h
GEN /home/user/micropython/ports/esp32/build-GENERIC_C3/genhdr/mpversion.h
[1063/1278] Generating ../../genhdr/qstrdefs.collected.h
QSTR updated
[1066/1278] Generating ../../frozen_content.c
FAILED: frozen_content.c 
cd /home/user/micropython/ports/esp32/build-GENERIC_C3/esp-idf/main && /home/user/.espressif/python_env/idf4.4_py3.8_env/bin/python3.8 /home/user/micropython/tools/makemanifest.py -o /home/user/micropython/ports/esp32/build-GENERIC_C3/frozen_content.c -v MPY_DIR=/home/user/micropython -v MPY_LIB_DIR=/home/user/micropython/../micropython-lib -v PORT_DIR=/home/user/micropython/ports/esp32 -v BOARD_DIR=/home/user/micropython/ports/esp32/boards/GENERIC_C3 -b /home/user/micropython/ports/esp32/build-GENERIC_C3 -f-march=rv32imc /home/user/micropython/ports/esp32/boards/manifest.py
usage: /home/user/micropython/mpy-cross/mpy-cross [<opts>] [-X <implopt>] <input filename>
Options:
--version : show version information
-o : output file for compiled bytecode (defaults to input with .mpy extension)
-s : source filename to embed in the compiled bytecode (defaults to input file)
-v : verbose (trace various operations); can be multiple
-O[N] : apply bytecode optimizations of level N

Target specific options:
-msmall-int-bits=number : set the maximum bits used to encode a small-int
-mno-unicode : don't support unicode in compiled strings
-mcache-lookup-bc : cache map lookups in the bytecode
-march=<arch> : set architecture for native emitter; x86, x64, armv6, armv7m, armv7em, armv7emsp, armv7emdp, xtensa, xtensawin

Implementation specific options:
  emit={bytecode,native,viper} -- set the default code emitter
  heapsize=<n> -- set the heap size for the GC (default 2097152)
MPY neopixel.py
error compiling /home/user/micropython/ports/esp32/modules/neopixel.py:
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [Makefile:34: all] Error 2

```

dpgeorge, 2021-06-21:
See #7438 for a WIP of ESP32C3 support.  If you can test it, that would be great!

schuschu, 2021-06-21:
Tested on a ESP32-C3F (2MB external flash variant, partitions adapted from GENERIC_D2WD) https://docs.ai-thinker.com/en/esp32c3 soldered onto a slightly modified Wemos D1 Mini (reset/en bridged, pulldown on former GPIO15 removed, 1uF on reset instead of 100nF).
C3 does come with bluetooth but the module is disabled, so I did not test it. I assume mpycross is still set to xtensawin while C3 is based on riscv (rv32imc)

Initially loops on flash filesystem with the error, but I guess that is expected behaviour (don't forget make erase before deploy).
```
The filesystem appears to be corrupted. If you had important data there, you
may want to make a flash snapshot to try to recover it. Otherwise, perform
factory reprogramming of MicroPython firmware (completely erase flash, followed
by firmware programming).
```

Shell works fine (tested UART only, have not wired the new D+/D- from the USB-JTAG block out).
GPIO only go up to 7 (my module at least has IO19 as max, documentation has a few pointers where the flash is located)
```
Note that GPIO12-17 are usually used for SPI flash.
Note that GPIO 18 and 19 are used by USB-JTAG by default. In order to use them as GPIOs, USB-JTAG will be disabled by the drivers.
ESP32-C3 doesn’t have separate “RTC GPIO” support. However, GPIO0-5 keep the rtc feature, which can be used for power-management and analog subsystem.
```

Blinking a LED on GPIO5 works fine though (so machine.Pin and time.sleep work).

I will run some more tests with all 19 pins (sadly 18 and 19 are wired to my I2C breakout board, but then I should be able to test SSD1306 displays and maybe I find a stray neopixel later as well). Should I test WiFi, SPI etc. as well or is a "basic test" enough?

schuschu, 2021-06-21:
Neopixel on Pin 10 is not looking good. Same on Pin 5 though.
```
>>> import neopixel,machine
>>> np = neopixel.NeoPixel(machine.Pin(10), 1)
>>> np[0]=(255,0,0)
>>> np.write()
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
MEPC    : 0x40381fa0  RA      : 0x40381f8e  SP      : 0x3fcc3fe0  GP      : 0x3fc8fc00
TP      : 0x3fc8efac  T0      : 0x4005890e  T1      : 0x4202dbcc  T2      : 0xffffffff
S0/FP   : 0x000000c8  S1      : 0x00000038  A0      : 0x00000001  A1      : 0x3fc9c750
A2      : 0x00000038  A3      : 0x60004000  A4      : 0x00000000  A5      : 0x00000080
A6      : 0x00000000  A7      : 0x3c1019b0  S2      : 0x00000000  S3      : 0x00000400
S4      : 0x3fc9c753  S5      : 0x00000080  S6      : 0x3fc9c751  S7      : 0x00000004
S8      : 0x42009000  S9      : 0x00000001  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x3fc9b568  T4      : 0x00000002  T5      : 0x00000002  T6      : 0x00000001
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000018  MTVAL   : 0x0000e001
MHARTID : 0x00000000

Stack memory:
3fcc3fe0: 0x00000001 0x3fc9c978 0x00000001 0x00000001 0x3c1019ca 0x3fffffff 0x3fc98430 0x000000b0
3fcc4000: 0x000000c0 0x00000003 0x0000000a 0x4202dc0c 0x3fc9c9c2 0x3fcc4210 0x00000003 0x3fc9c750
3fcc4020: 0x000000c0 0x3fc9c750 0x00000003 0x00000001 0x3fc9c610 0x3fcc4204 0x3c1019c9 0x4201a86a
3fcc4040: 0x3fcc41f0 0x3c0fb0fc 0x3fcc4218 0x3fcc4214 0x0000001b 0x3fcc3fd0 0x3fcc4298 0x00000000
3fcc4060: 0x4201996e 0x3fcc41f0 0x00000030 0x00000000 0x3fc9c610 0x0000000c 0x00000044 0x3fcc420c
3fcc4080: 0x00000020 0x42009000 0x3fcc4298 0x00000000 0x4201996e 0x3fcc4250 0x00000000 0x3fcc41f0
3fcc40a0: 0x3fc9c610 0x00000005 0x00000030 0x3fc9c9c2 0x00000004 0x42009000 0x00000001 0x00000000
3fcc40c0: 0x00000000 0x3fcc4040 0x00000006 0x3fcc4410 0x3fc9c6cf 0x3fffffff 0x000013f2 0x3fc8f5d8
3fcc40e0: 0x0000008a 0x3fcc4118 0x00000011 0x42011c54 0x00000000 0x00000000 0x00000389 0x3c0f9f37
3fcc4100: 0x0000027e 0x0000027e 0x3fc8f5d8 0x42011da0 0x3fc9c6cf 0x3fffffff 0x00000000 0x00000000
3fcc4120: 0x0000027e 0x000013f2 0x3c0f5ee8 0x00000003 0x00000000 0x00000000 0x3fcc4220 0x42034b32
3fcc4140: 0x420100d0 0x420100bc 0x3fc9c190 0x00000000 0x00000000 0x00000001 0x42009000 0x3fcc4204
3fcc4160: 0x3fc9c919 0x00000040 0x00000008 0x4200c3d4 0x00000000 0x00000000 0x3fcc41f0 0x00000000
3fcc4180: 0x00000000 0x00000001 0x3c0f69b0 0x4200f056 0x3fc9c9bd 0x42015ca8 0x3fc9c214 0x00000004
3fcc41a0: 0x00000000 0x3fcc41fc 0x3fc9c7a0 0x42015134 0x00000000 0x00000001 0x42009000 0x00000000
3fcc41c0: 0x00000000 0x00000001 0x42009000 0x00000004 0x3fc9c9c2 0x00000030 0x00000005 0x3fc9c610
3fcc41e0: 0x3fcc41f0 0x00000000 0x3fcc4250 0x42010144 0x3fc9c730 0x3c1019c8 0x3fcc4200 0x00000005
3fcc4200: 0x3fc984c4 0x3fc8fcf0 0x3c0f33dc 0x3fc9c740 0x00000003 0x3fc9c210 0x3fcc4250 0x420100bc
3fcc4220: 0x000000c0 0x00000001 0x00000000 0x3fcc441c 0x000000c0 0x3fffffff 0x3fc98430 0x000000b0
3fcc4240: 0x000000c0 0x3fcc4418 0x3fc9c9c3 0x4201a922 0x3fcc4400 0x3c0fb0fc 0x3fcc4420 0x3fcc441c
3fcc4260: 0x00000032 0x3fc9cc60 0x00000001 0x00000002 0x3fc9cc74 0x3fcc43f0 0x00000000 0x00000000
3fcc4280: 0x000000aa 0x00000002 0x00000001 0x3fcc4414 0x00000020 0x00000000 0x3fcc4488 0x42039c04
3fcc42a0: 0x4201996e 0x3fcc4460 0x00000000 0x3fcc4400 0x3fc984c4 0x00000003 0x00000030 0x3c100000
3fcc42c0: 0x00000004 0x42009000 0x00000001 0x00000000 0x00000000 0x3fcc4250 0x20000008 0x00000000
3fcc42e0: 0x3c0f4578 0x3fc9c310 0x3fc9c391 0x00000000 0x00000020 0x00000080 0x00000011 0x00000002
3fcc4300: 0xffffffff 0x00000005 0x00000001 0x00000030 0x00000032 0x3fc9caa4 0x00000001 0x00000002
3fcc4320: 0x3fc9cab8 0x3fcc43f0 0x00000000 0x420383f4 0xffffffff 0x00000005 0x00000001 0x00000034
3fcc4340: 0x3fcc4480 0x3c108274 0x3fcc43f0 0x42039c04 0x3fc9cad0 0x3fc9cac0 0x3fcc43f0 0x00000001
3fcc4360: 0x00000000 0x3c10b126 0x00000011 0x00000001 0x3fc9cad0 0x00000000 0x3fcc43f0 0x00000000
3fcc4380: 0x00000000 0x00000000 0x00000010 0x4200c152 0x3fc9c8d0 0x3fc9cad0 0x00000000 0x4201030e
3fcc43a0: 0x3fcc4480 0x3c108274 0x3fc9c9b0 0x00000003 0x00000000 0x3fc9cad0 0x3fcc43f0 0x3fc9c9f0
3fcc43c0: 0x00000000 0x00000000 0x3fc9c960 0x00000000 0x00000000 0x00000001 0x42009000 0x00000004
```

schuschu, 2021-06-21:
I2C test for hw i2c0 and sw i2c (-1), ssd1306 (128x64) display works.

schuschu, 2021-06-21:
WiFi (tested only AP interface + socket + simple webserver) works.

schuschu, 2021-06-21:
ADC has no pins defined, reference states:
```
ADC1:
5 channels: GPIO0 - GPIO4
ADC2:
1 channels: GPIO5
```

schuschu, 2021-06-21:
PWM works (somewhere below 80Hz it does odd things (same goes for above 600kHz), I guess it was never designed to do that.

dpgeorge, 2021-07-18:
ESP32C3 is now supported, including most peripherals.  Any outstanding items/bugs/features can be addressed by individual issues.

closed 2021-07-18 by dpgeorge

QGB, 2022-05-08:
```
C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\share\openocd\scripts>openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Info : Hart 0 unexpectedly reset
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: Unable to execute JTAG queue
Error: Failed to write ESP32C3_TIMG1WDT_PROTECT (-4)!
Error: Failed to disable WDTs (-4)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed write at 0x17, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed write at 0x17, status=2
Error: Failed to enable EBREAKS handling (-4)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed write at 0x10, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: Failed to read dmstatus!
Error: Failed to resume core (-4)!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: failed read at 0x11, status=2
```