micropython/micropython #7433: RP2: Unexpected behaviour if allocation occurs in ISR

peterhinch, 2021-06-20:
Most platforms raise an exception if an ISR attempts allocation. RP2 fails silently.
```python
from machine import Pin, PWM,Timer
import time
a = 1.0
def ns(_):
    global a
    a *= 1.1

t1= Timer()
t1.init(freq =8000, mode = Timer.PERIODIC, callback =ns)

while True:
    print(a)
    time.sleep(1)
```
Result:
```
1.0
inf
inf
inf
inf
```
This behaviour is confusing: arguably an exception should be raised.

kadamski, 2021-06-22:
The frequency is just to big. With smaller frequency, the code will run correctly, even though there is an allocation. This is due to the fact that the machine.Timer uses mp_sched_schedule() to run the handler so it is just like you used the micropython.schedule() in your handler which means, you can allocate memory. This is why, you don't get the exception.

It seems there is no way to use hard irq for timers instead of the mp_sched_schedule() on RP2. In contrast to that, the timer on stm32 port is using hard IRQ and there seems to be no way to switch to using mp_sched_schedule() (other than calling the micropython.schedule() from the python). That's confusing.

kadamski, 2021-06-22:
@dpgeorge It seems that what is inconsistent with RP2 port is that the timer uses the mp_sched_schedule() internally for timers, while the users expect it to behave just like on stm32, where it is running in hard irq context. Shouldn't this inconsistency be also closed somehow?

peterhinch, 2021-06-22:
Good point about `freq`! I must have had an arithmetic bypass...

If rp2 uses soft IRQ's of necessity this needs to be documented. If so, it's a major drawback putting it in the same category as ESPx. In particular applications using the PIO are likely to expect a consistent fast response to IRQ's.

kadamski, 2021-06-22:
@peterhinch for Pin IRQ, you can choose the hard irq behaviour by using hard=True when creating an IRQ. In the example from  #7426 (gpio pulse on the GPIO as a response to the edge IRQ produced by PWM), it looks like this (yellow is the PWM signal, purple is the pulse created inside of an IRQ handler):

[image]

So the latency is ~25us.

With hard=false (so when using mp_sched_schedule(), it looks like this:

[image]

So the latency is ~35us. Not a huge difference.

BTW, the pulse width is ~8us.

Just for the completeness, here's the exact code that was used:

from machine import Pin, PWM
```python
def handler(x):
    v(1)
    v(0)

if __name__ == "__main__":
    pin = Pin(16, Pin.OUT)
    pwm = PWM(pin)
    pwm.freq(10)
    pwm.duty_u16(32512)
    cs = Pin(13, Pin.OUT, value=1)
    v = cs.value
    pin2 = Pin(14, Pin.IN)
    pin2.irq(trigger=Pin.IRQ_FALLING, handler=handler, hard=True)
```

robert-hh, 2021-06-22:
Since with pin.irq the option `hard=True` is supported, that could offer a choice for allowing the same for timers too, however this is implemented.
@kadamksi: You can reduce the latency drastically if you write it as:
```
def handler(x, v=cs):
    v(1)
    v(0)
```
You may have to create the cs beforehand.

kadamski, 2021-06-22:
@robert-hh hmm.. why would that change anything to the latency? I don't get it, don't I do basically the same already? What am I missing?

robert-hh, 2021-06-22:
Just try it. 
a) There is a video by Damien explaining why putting an object as pseudo-argument to a function speeds up the search for that object. 
b) pin(x) is faster than pin.value(x)

kadamski, 2021-06-22:
@robert-hh 

ad1 now I see your point - to make the v a name local to the function, instead of global. But honestly, in practice, the difference is almost non-measurable.. Well, maybe a tiny difference? In hard=True case, the latency changes between something like 23-27us in my original code while the fluctuations drops to more like 21-25us when using this trick. So maybe there's like 1-2us difference indeed, but for sure no drastic change.

ad2. if you look at the code I pasted in my comment, I already did use: v=cs.value, v(x) instead of cs.value(x). I could have used the cs(x) directly as well, indeed, but I don't think there is a difference between those two, as the dereference is already omitted anyway. BTW, If I use cs.value(x) instead of cs(x), the latency is something like 4-5us larger (so ~30us instead of ~25us), so, a noticeable difference but I also wouldn't call it drastic difference.

robert-hh, 2021-06-22:
Ok. Still the latency seems long. I did recently a similar test when setting up pin.irq for mimxrt. There, the latency for hard=True was between 1.5 and 5 µs, and for hard=False it was 5 to 10 µs. I retested mow removing the pin object from the argument, which increased latency by ~2µs. I had in mind that it was much more. I triggered the pin with an external generator, but that makes no difference I guess.

kadamski, 2021-06-22:
@robert-hh yes, 20-25us isn't low latency, for sure, if you look at my edit I did exactly when you posted your comment, you will see that even if the code is in C, the latency is 3us so its worse than you got on mimxrt with micropython!

But teensy 4.0 is Cortex-M7@600 MHz, while RP2 is Cortex-M0+@133 MHz so...

peterhinch, 2021-06-23:
I'll close this as hard IRQ's work as expected. I'd missed the fact that **machine** sets them soft by default.

closed 2021-06-23 by peterhinch