micropython/micropython #7434: Failing mpy-cross builds on aarch64-linux-gnu

andrewleech, 2021-06-21:
Hi, I've got an automated mpy-cross building project at https://gitlab.com/alelec/mpy_cross in which I added aarch64-linux-gnu support for v1.14

Unfortunately though these builds have all been failing ever since #6939 (`Implement NLR for AArch64` by @Jongy)

I use crossbuild docker image for building these like so:
```
docker run --rm -v $(pwd):/workdir -e CROSS_TRIPLE=aarch64-linux-gnu multiarch/crossbuild make -C mpy-cross
```
They're failing with:
```
make: Entering directory '/workdir/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/mpversion.h
CC ../py/mpstate.c
CC ../py/nlr.c
CC ../py/nlrx86.c
CC ../py/nlrx64.c
CC ../py/nlrthumb.c
CC ../py/nlraarch64.c
/tmp/ccHi3CVG.s: Assembler messages:
/tmp/ccHi3CVG.s:7: Error: operand 1 should be a floating-point register -- `stp lr,x9,[x0,#16]'
/tmp/ccHi3CVG.s:41: Error: operand 1 should be a floating-point register -- `ldp lr,x9,[x1,#16]'
make: *** [build/py/nlraarch64.o] Error 1
../py/mkrules.mk:77: recipe for target 'build/py/nlraarch64.o' failed
make: Leaving directory '/workdir/mpy-cross'
```

How are you building your aarch64 port @Jongy do you know if there would be a different compiler version/flag needed for this?

Jongy, 2021-06-21:
I used it in my linux-kernel port (#5482), to build it on Graviton machines.

Here's the command line it uses when I run `make -C mpy-cross V=1`:
```
...
CC ../py/nlraarch64.c
gcc -I. -Ibuild -I.. -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -std=gnu99 -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -Os   -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables -Os -c -MD -o build/py/nlraarch64.o ../py/nlraarch64.c
...
```

GCC:
```
$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Running native on AArch64:
```
$ uname -m
aarch64
```

This is from a build not based on master (it is based on some intermediate version of #5482), these are the relevant contents of `nlraarch64.c`:
```
#if MICROPY_NLR_AARCH64

// AArch64 callee-saved registers are x19-x29.
// https://en.wikipedia.org/wiki/Calling_convention#ARM_(A64)

// Implemented purely as inline assembly; inside a function, we have to deal with undoing the prologue, restoring
// SP and LR. This way, we don't.
__asm(
    "nlr_push:               \n"
    ".global nlr_push        \n"
    "mov x9, sp              \n"
    "stp lr,  x9,  [x0,  #16]\n" // 16 == offsetof(nlr_buf_t, regs)
    "stp x19, x20, [x0,  #32]\n"
    "stp x21, x22, [x0,  #48]\n"
    "stp x23, x24, [x0,  #64]\n"
    "stp x25, x26, [x0,  #80]\n"
    "stp x27, x28, [x0,  #96]\n"
    "str x29,      [x0, #112]\n"
    "b nlr_push_tail         \n" // do the rest in C
    );

NORETURN void nlr_jump(void *val) {
    MP_NLR_JUMP_HEAD(val, top)

    MP_STATIC_ASSERT(offsetof(nlr_buf_t, regs) == 16); // asm assumes it

    __asm volatile (
        "ldr x29,      [%0, #112]\n"
        "ldp x27, x28, [%0,  #96]\n"
        "ldp x25, x26, [%0,  #80]\n"
        "ldp x23, x24, [%0,  #64]\n"
        "ldp x21, x22, [%0,  #48]\n"
        "ldp x19, x20, [%0,  #32]\n"
        "ldp lr,  x9,  [%0,  #16]\n" // 16 == offsetof(nlr_buf_t, regs)
        "mov sp, x9              \n"
        "mov x0, #1              \n"  // non-local return
        "ret                     \n"
        :
        : "r" (top)
        :
        );

    MP_UNREACHABLE
}

#endif // MICROPY_NLR_AARCH64
```

What compiler version are you using?

andrewleech, 2021-06-21:
Hi thanks, yeah I'm not sure what version of gcc is used in the image. I thought they were all quite up to date because the multiarch/crossbuild GitHub page latest commit was pretty recent... but chasing it down the latest published release is 4 years old so that's probably the issue.

Looking around it appears dockcross is a newer project to achieve the same goal, I'll have to look at switching to that and see if that fixes the issue

closed 2022-01-11 by andrewleech