micropython/micropython #7435: Inconsistent behavior comparing binary data to arrays

andrewleech, 2021-06-21:
I'm got an application that deals with 16bit image data in a large binary array (1.2MB). It's typically held in a C object, but exposed via a raw buffer interface (`memoryview` etc).

For testing data content, I was previously doing something like 
```
pattern = [0, 0, 1, 1, 2, 2, 4, 4, 8, 8, 16, 16, 32, 32, 64, 64, 128, 128,
                       256, 256, 512, 512, 1024, 1024, 2048, 2048, 4095, 4095]
pattern_arr = array.array('H', pattern) # 16 bit unsigned short array

assert memoryview(image)[0: len(pattern) * 2] == pattern_arr
```

Even though the memoryview will be exposed as a U8, the binary comparison to the array was fine.  

Ever since ` py/objarray: Prohibit comparison of mismatching types. ` (https://github.com/micropython/micropython/commit/57365d855734142deb030ebcd00c10efcedf554b by @stinos) however this throws a `NotImplementedError` which I guess is the intent behind that commit? Somewhat annoying for me, but presumably safer?

However, while looking for a solution here (cast my image to U16 without copy?) I found:
```
image_row_bytes = uctypes.bytearray_at(uctypes.addressof(image), len(pattern) * 2)
assert image_row_bytes == pattern_arr
```
seems to work fine... so comparing to U16 array to memoryview is not allowed but U16 array compared to bytearray is allowed? 
Is this intended?

dpgeorge, 2021-06-21:
> seems to work fine... so comparing to U16 array to memoryview is not allowed but U16 array compared to bytearray is allowed?

I couldn't get that case working under uPy either.

This test:
```python
import array

raw = b'1234'
a = array.array('H', raw)
print(a)

print(a == bytearray(raw))
print(a == memoryview(bytearray(raw)))
print(a == bytearray(memoryview(bytearray(raw))))

print(bytearray(raw) == a)
print(memoryview(bytearray(raw)) == a)
print(bytearray(memoryview(bytearray(raw))) == a)
```
Succeeds for all cases on CPython (all fail for uPy), but the output is particularly interesting:
```
array('H', [12849, 13363])
True
False
True
True
False
True
```

andrewleech, 2021-06-21:
Interesting... But yes sorry I too should figure out how to replicate both issues I saw in a standalone unit test (without the image object) to properly replicate the issue.

andrewleech, 2021-06-21:
Ah yes, my `bytearray_at` "workaround" listed earlier was only working at 57365d8 but fails at current master, I missed I hadn't moved my tree back to HEAD this discrepancy must have been fixed already since. Turns out I'm going to need to find another workaround for this in my code.

But your finding @dpgeorge is quite true in that this new strictness in assay comparison does make micropython less consistent with cpython in this area. 

It is quite curious that cpython matching is inconsistent between bytearray and memoryview

stinos, 2021-06-21:
> does make micropython less consistent with cpython in this area

'less consistent' is probably not the correct term here: earlier all these things were indeed allowed, but they also gave different (incorrect) results than CPython so not consistent. Whereas now that is just forbidden.

andrewleech, 2021-06-21:
Yeah ok, sure. I don't particularly like the look of the cpython results anyway, they seem confusing too.
I don't know the use cases that were giving incorrect results I guess, I'm only aware of my case that was working, but is now denied. I do realise what I was doing was not obviously "correct"/working and sounds like I was using the one form of @dpgeorge 's test above that doesn't work on cpython anyway (for some reason).

I've "fixed" my case now by wrapping my U16 array with `byte()`so it can be compared to the memoryview of the image - this is a more explicit version of what it was doing implicitly before so that's somewhat better really, just with an extra copy from the array to the byte object. I could probably find a no-copy way to do the same thing.

Thanks for the feedback, I am inclined to agree that what's in master currently is better / safer than what was there before. Explicit support could be added in future to match cpython, but possibly isn't necessary if there are workarounds using some form of explicit typing like `byte()` or some functions of `uctypes`.

stinos, 2021-06-21:
> I don't particularly like the look of the cpython results anyway, they seem confusing too

Exactly, it's a bit strange.

> I don't know the use cases that were giving incorrect results I guess

Apart from the test dpgeorge shows which would give different results than CPython (because it would probably covert each of the U8 elements into an U16 then compare that), anything for which binary equality (since MicroPython uses memcpy) doesn't match per-element equality. Like comparing floating point against an integer of the same size, or comparing float agains float in the presence of NaN.

> I could probably find a no-copy way to do the same thing

Perhaps something simple like looping over `pattern`, for each element getting 2 bytes from your `image` and creating an U16 from that by LSH of the first or'ed with the second? That's still a copy though only of 2 elements at a time instead of the whole image

dpgeorge, 2021-06-21:
You can see why the original expression was not correct by:
```python
image = b'\x01\x00\x02\x00data'
print(array.array("H", image[0:len(pattern) * 2]))
print(array.array("H", memoryview(image)[0:len(pattern) * 2]))
```
output:
```
array('H', [1, 2])
array('H', [1, 0, 2, 0])
```
So a bytes/bytearray is treated as verbatim data, whereas a memoryview is treated as an array of elements.  I guess this is because a memoryview is more like an array than a bytearray and a memoryview has a type (a bytearray is just bytes).  So comparing a memoryview to an array is comparing them logical element wise.

You can try this for a copy-free solution:
```python
import uctypes

# like memoryview, but returns a bytearray by reference
def bytearray_of(buf, offset, length):
    return uctypes.bytearray_at(uctypes.addressof(buf) + offset, length)
    
assert bytearray_of(image, 0, len(pattern) * 2) == bytearray_of(pattern_arr, 0, len(pattern_arr) * 2)
```

Alternatively, just define `pattern_arr` as a bytes object from the beginning.

andrewleech, 2021-06-22:
Thanks! yeah I wanted to keep the initial definition as U16 numbers as they're easier to document why/where they come from, I converted that array to U8 with `uctypes.bytearray_at()` and that works well for me.

I'm happy to close this ticket, thanks for the input both of you! Making comparisons work the same as cpython might be worthwhile in future, but there's lots of other more critical things to work on in the mean time!

closed 2021-06-22 by andrewleech