micropython/micropython #7437: unable to implement a C-module located at py/ as per the example in official doc MP v1.16

mccwong, 2021-06-21:
@contributors,
Hi, wanted to report an issue (failure to implement the example shown in the official MPv1.16 doc "implementing a core module" at the following link [https://docs.micropython.org/en/latest/develop/library.html#implementing-a-core-module](url)

Development Platform : Raspberry Pi 4 (8GB ram) running official RaspPi OS and apt updated & upgraded 
MCU board : ESP32-WroverB on Dev-KitC-v4 
esp-idf : latest version 4.4 

Issue:
- added the modsubsystem.c file to micropython/py with the 3rd argument in the MACRO MP_REGISTER_MODULE hardcoded to 1.  
- added the line "modsubsystem.o \" in the appropriate place in the file py.mk  (I experimented and found out that unless this is done, the files modsubsystem.o and modsubsystem.h would not appear in the folder /mpy-cross/build/py/ 
- performed make -C mpy-cross at root folder of repository (this will build the 2 files named above from modsubsystem.c in py/ as well as those .o and .h from many other files from py/ folder.
- despite various efforts to amend the mpconfig.h files in py/ in ports/esp32/ directories, the subsystem module does not appear in the built system on the esp32  (I just used the default board setting = generic, without spi-ram) , using help('modules') or using import subsystem ...

May I know how I could implement a C-module located at py/ ??    Many many thanks.
(I chose to implement at location at py/ as it seems the easiest way to implement a C-module and without the restrictions of building in user-module/external-module.)

 Also, wanted to say thanks to all contributors and to Damien, micropython is magnificent way for software breadboarding!!

dpgeorge, 2021-06-21:
The approach you mention is not the recommended way to add user C modules, so it can't really be supported.

But probably all you need to do is add `modsubsystem.c` to `py/py.cmake`, the `MICROPY_SOURCE_PY` list (because esp32 uses cmake).

mccwong, 2021-06-21:
> The approach you mention is not the recommended way to add user C modules, so it can't really be supported.
> But probably all you need to do is add `modsubsystem.c` to `py/py.cmake`, the `MICROPY_SOURCE_PY` list (because esp32 uses cmake).

@Damien, thanks so much for your prompt reply, and help.   So sorry for crossing the line about the unorthodox usage.

For the record, this issue is non-existent (as the example in the official MP documentation v1.16 actually works), provided that modification are made to the files py/py.cmake (modification is simple) and py/objmodules.c (modification is more complicated).  Somehow, I also needed to modify the py/py.mk file (modification is simple).      

Unfortunately,  the build ended up in 2 modules of the same name (both are named "subsystem" in the REPL when type help('modules')).   But subsequent importing subsystem and then printing subsystem.info() works without issue.  
Tried many times to change various settings 😔😔😔 but still resulting in 2 subsystem modules being incorporated in the firmware (needs more help 😔😔).    If no further follow-ups on this topic.... may be can label it resolved or close it.  Great thanks indeed!!

dpgeorge, 2021-06-23:
> Unfortunately, the build ended up in 2 modules of the same name

This is probably because you added it to `py/objmodules.c`.  That shouldn't be necessary because it's automatically added to that list when you use `MP_REGISTER_MODULE` to register your module.

mccwong, 2021-06-24:
@damien, thank you very much for your help indeed.  
I understand the repository better now and for non-core features I will switch to implementing new modules using the USER_C_MODULES flag in the future.

closed 2021-06-24 by dpgeorge