micropython/micropython #7439: Win10 ESP32 build : fatal error: db.h: No such file or directory

LeMoussel, 2021-06-21:
On Win 10 (use ESP-IDF Tools Installer for Windows to install all the required tools),  For esp32 (ports/esp32) when I do `idf.py -D MICROPY_BOARD=GENERIC build`, I got this error: 
```
[3/221] Generating ../../genhdr/qstr.i.last
FAILED: genhdr/qstr.i.last
esp-idf\main\CMakeFiles\qstr.i.last-417609e.bat 951d66de95feced5
D:/Developpement/Python/micropython/extmod/modbtree.c:37:10: fatal error: db.h: No such file or directory
 #include <db.h>
          ^~~~~~
compilation terminated.
Traceback (most recent call last):
```

Hope to get your reply!

dpgeorge, 2021-06-21:
Try running `make submodules` in the `ports/esp32` directory.

LeMoussel, 2021-06-21:
`
D:\Developpement\micropython\ports\esp32>make submodules
'make' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.
`

The `make` command is not known

dpgeorge, 2021-06-23:
From the top level MicroPython directory (ie in `micropython/`) run:
```
git submodule update --init lib/berkeley-db-1.xx
```

LeMoussel, 2021-06-23:
OK. But I encountered a Unix/Win incompatibility with `sed` ('\' vs '/'). It seems that it is not possible to build under Windows. 
On the other hand I was able to build under WSL without any problems.

dpgeorge, 2021-06-23:
> But I encountered a Unix/Win incompatibility with `sed` ('' vs '/').

If you can give more details on the problem they maybe it could be fixed.

dpgeorge, 2024-03-15:
Closing due to inactivity.

closed 2024-03-15 by dpgeorge