micropython/micropython #7440: custom frozen board-specific main programs

oclyke, 2021-06-21:
Would the ability to specify board-specific main programs be considered? The benefit of doing this would be twofold:

* the ability to fire up devices that run the desired program out of the box (directly after flashing) rather than needing to access the filesystem and add the ```main.py``` script.
* the confidence knowing that the main program image is not affiliated with the filesystem and is therefore a little more immune to destruction by human error

I'll submit a very simple PR demonstrating this for the ESP32 port.

dpgeorge, 2021-06-23:
> the ability to fire up devices that run the desired program out of the box (directly after flashing) rather than needing to access the filesystem and add the `main.py` script.

It should already be possible to achieve this by freezing in `main.py`, ie specifying it in the `manifest.py` for the custom board.

> the confidence knowing that the main program image is not affiliated with the filesystem and is therefore a little more immune to destruction by human error

If `main.py` is frozen bytecode then it also satisfies these points.

In production systems, one can freeze `boot.py` and `main.py` and use `boot.py` to set `sys.path` to control the order of import searching, which allows frozen `main.py` to be overridden or not by the filesystem.

oclyke, 2021-06-23:
ahh excellent. thank you for answering with an elegant and well thought out solution.

closed 2021-06-23 by oclyke

dpgeorge, 2021-06-23:
This could benefit from some docs around best practices for doing this.  But then, there's always scope for more and better docs :)