micropython/micropython #7443: openmv port to esp32s3 block

YanMinge, 2021-06-22:
Hi,
I am creating an open source project, Part of it is to port openmv to esp32s3.
I have done most of the work, usb, camera, micropython...
But I was block here now
https://github.com/YanMinge/esp32_mpy/blob/master/project/aitest/omv/common/usbdbg.c
```
            if (!script_running && !gc_is_locked()) {
                vstr_add_strn(&script_buf, buffer, length);
                xfer_bytes += length;
                if (xfer_bytes == xfer_length) {
                    // Set script ready flag
                    script_ready = true;

                    // Set script running flag
                    script_running = true;

                    // Disable IDE IRQ (re-enabled by pyexec or main).
                    usbdbg_set_irq_enabled(false);

                    // Clear interrupt traceback
                    mp_obj_exception_clear_traceback(mp_const_ide_interrupt);
                    MP_STATE_VM(mp_pending_exception) = mp_const_ide_interrupt;
                    #if MICROPY_ENABLE_SCHEDULER
                        if (MP_STATE_VM(sched_state) == MP_SCHED_IDLE) {
                            MP_STATE_VM(sched_state) = MP_SCHED_PENDING;
                    }
                    #endif
                }
            }
```

I wan't break of repl or some micropython script. but the code can't achieve this purpose.

openmv do it like this
```
                    usbdbg_set_irq_enabled(false);
                    mp_obj_exception_clear_traceback(mp_const_ide_interrupt);
                    __set_BASEPRI(0);
                    pendsv_nlr_jump(mp_const_ide_interrupt);
```

dpgeorge, 2021-06-23:
It's hard to tell why it doesn't work without further details about the symptoms (what goes wrong).

But please use the `mp_sched_exception(mp_const_ide_interrupt)` function instead of setting `MP_STATE_VM()` state directly.

YanMinge, 2021-06-23:
thanks, I wrote the code as you suggested. Seems to be working perfectly

closed 2021-06-23 by dpgeorge