micropython/micropython #7466: How to make the LPUART working on STM32 -  Nucleo WB55  ?

YannickMarietti, 2021-06-26:
Hello

I am simply struggling to understand how to initialize the newly added LPUART on the Nucleo WB55 ...
I found no example about it.
I try things like this :

`
from pyb import UART # Classe pour gérer l'UART
from micropython import const # Pour déclarer des constantes entières

uart = UART(2) # Instance de l'UART numéro 2

# Initialisation à 9600 bauds, trames de 8 bits, sans contrôle de parité avec un stop bit à 1
uart.init(9600, bits=8, parity=None, stop = 1)
`

But it always fail with a message like : 

"ValueError: set baudrate XXXX is not within 5% of desired value"

Which UART number  shall I use ?

Any help would be greatky appreciated !

Best regards

Yannick

dpgeorge, 2021-06-27:
Try `pyb.UART('LP1')`

YannickMarietti, 2021-06-27:
Thank you !

With 

`from pyb import UART`
`uart = UART('LP1')`
`uart.init(9600, bits=8, parity=None, stop = 1)`

The problem remains the same, I still get the following message : 

`ValueError: set baudrate 0 is not within 5% of desired value`

Sorry :(

Yannick

dpgeorge, 2021-06-28:
What version of MicroPython are you running?  Commit 2ac09c2694f89b11544f4abf10413c64eeec2b15 did fix some issues around setting the baudrate on STM32WB55 MCUs.

YannickMarietti, 2021-06-28:
Hello Damien

Thank you again for your support !
I use this release : 
NUCLEO_WB55-20210625-unstable-v1.16-17-gb51ae20c0.dfu

dpgeorge, 2021-06-28:
Ok, I can confirm there is something wrong with the setting of the LPUART baudrate on the NUCLEO_WB55.

dpgeorge, 2021-07-26:
Should be fixed by fef21144047201d639b2692e3a55083c7180cb29

closed 2021-07-26 by dpgeorge