micropython/micropython #7471: Incorrect uasyncio behaviour: tasks are not correctly terminated

mattytrentini, 2021-06-29:
(First reported on the [forum](https://forum.micropython.org/viewtopic.php?f=2&t=10753))

The following snippet:

```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio

async def count():
    i = 0
    while True:
        print(i)
        i += 1
        await asyncio.sleep(1)

async def main():
    asyncio.create_task(count())
    await asyncio.sleep(5)

asyncio.run(main())
asyncio.run(main())
```

Produces the following in CPython:

```txt
0
1
2
3
4
0
1
2
3
4
```

But in MicroPython:

```
0
1
2
3
4
0
5
1
6
2
7
3
8
4
9
```

It appears that tasks are not correctly terminated (?).

A workaround that produces the correct output is to run the second task in a second event loop:

```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio
async def count():
    i = 0
    while True:
        print(i)
        i += 1
        await asyncio.sleep(1)
async def main():
    asyncio.create_task(count())
    await asyncio.sleep(5)
asyncio.run(main())
asyncio.new_event_loop() # Added line
asyncio.run(main())
```

Note that running either of the snippets _repeatedly_ causes more confusing output because all previous tasks are still running on the queue.

peterhinch, 2021-06-29:
By default **uasyncio** retains state, as [discussed here](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#72-uasyncio-retains-state). Arguably in the interests of CPython compatibility it shouldn't, but on occasion this behaviour can be useful. Currently the user has the choice of whether to allow retained state.

mattytrentini, 2021-06-29:
> By default **uasyncio** retains state, as [discussed here](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#72-uasyncio-retains-state). 

I did look through your docs to see if this was covered but missed this; I _knew_ you'd be aware of this @peterhinch!

> Arguably in the interests of CPython compatibility it shouldn't, but on occasion this behaviour can be useful. Currently the user has the choice of whether to allow retained state.

That behaviour was _very_ unexpected to me; not just because it was different to CPython but because it didn't make any logical sense. I'd strongly prefer to see CPython compatibility in this case. I'm also concerned that there will be subtle issues porting CPython libraries that may inadvertently stumble on this problem...

jimmo, 2021-07-05:
Related: It would be worth considering making asyncio.run() detect that it's been (incorrectly) called recursively (or worse, from an interrupt handler).