micropython/micropython #7473: esp8266 nonblocking ussl: `.read(N)` never returns any data

vshymanskyy, 2021-06-29:
MicroPython version: **esp8266-20210618-v1.16**

```py
s = usocket.socket()
s.connect(usocket.getaddrinfo(server, port)[0][-1])
conn = ussl.wrap_socket(s, server_hostname=server)
s.settimeout(0)          # or s.setblocking(False)
conn.write(b"some data")  # OK: server gets the data and replies
while True:
    data = conn.read(1) # increasing count doesn't help
    print('>', data)
```

Outputs:
```log
> None
> None
> None
> None
> None
> None
> None
> None
> None
> None
> None
> None
> b''
> b''
> b''
> b''
> b''
...
```
No data is received.

Observations:
- If I set `conn = s` (i.e. use plain insecure TCP), it works
- If I replace `s.settimeout(0)` with `s.settimeout(1)`, it works
- If I replace `s.settimeout(0)` with `s.settimeout(0.1)`, I get `OSError: -261` on `conn.read`
- Same code works on ESP32

dpgeorge, 2021-06-30:
ESP8266 uses axtls for the ssl library, and that does not fully support non-blocking mode.  It probably needs modifications to the axtls library to work.  This is not a high-priority item so feel free to look into it yourself.

vshymanskyy, 2021-06-30:
Ok, just wanted to document it for now.

dpgeorge, 2021-06-30:
Let's keep this issue open, because it describes it well and has a good reproduction of the problem.