micropython/micropython #7475: machine module: handling of unsupported args

peterhinch, 2021-06-30:
This showed up using pin IRQ's but the problem prompted contemplating the general issue. The docs state that the **hard** arg for **Pin.irq()** is not supported by all ports. Accordingly the line
```python
xirq = pin_x.irq(trigger=Pin.IRQ_RISING | Pin.IRQ_FALLING, handler=self._x_cb, hard=True)
```
fails on ESP32 with "TypeError: extra keyword arguments given". The obvious (ugly) way to make this portable is to trap the exception and try again without the offending arg. This could become unmanageable if there were multiple args with platform-dependent support.

The best solution is moot. One option would be to accept and silently ignore unsupported args, but this isn't very user friendly. Another would be to accept and ignore the arg, but throw (say) ValueError. Done in such a way as to show which arg is problematic, and in such a way that the handler can be null.

Given that the purpose of **machine** is to facilitate portability, I think this issue is worth review.

dpgeorge, 2021-07-02:
> The obvious (ugly) way to make this portable is to trap the exception and try again without the offending arg.

In this particular case (`hard=True` on pin IRQ), would the code work if hard IRQs were not supported?  And if it would, why not use hard=False by default?  (I think I know the answer, but would be good to hear the context here.)

> One option would be to accept and silently ignore unsupported args, but this isn't very user friendly.

IMO this is a bad way to handle it because then the user thinks it's working when it's not.  Although the case of `hard=True` is a bit different because you get the same functionality whether it's a hard or soft callback, it's just a difference in performance/latency.  So one could argue that `hard=True` can always be supported, it's just that it won't be any better than `hard=False` on some port/MCUs.

> Another would be to accept and ignore the arg, but throw (say) ValueError

I don't see how this is different to the current behaviour of `TypeError: extra keyword arguments given`, you still need to try/catch to support port with/without this feature?

peterhinch, 2021-07-03:
> would the code work if hard IRQs were not supported? And if it would, why not use hard=False by default?

The context is an asynchronous driver for control knobs which use quadrature encoding. This is to be used in a cross-platform library [micro-gui](https://github.com/peterhinch/micropython-micro-gui). On platforms like ESP32 with SPIRAM the interrupt latency means that sometimes changes may be missed - this seems unavoidable. On platforms which support hard IRQ's it seems logical to use them as the ISR code is fast and non-allocating.

> IMO this is a bad way to handle it because then the user thinks it's working when it's not.

Agreed, also with your comment about the special case of `hard`.

>I don't see how this is different to the current behaviour of TypeError: extra keyword arguments given, you still need to try/catch to support port with/without this feature?

In this particular case a workround was no problem because I knew which arg was the culprit - and, crucially, all other args are universally supported. My concern is the general case where more than one arg is supported on a subset of platforms. The `TypeError` does not indicate which arg(s) are causing the problem. In this case determining how to proceed would involve an unwieldy trial-and-error process. A lookup based on a table of possible platforms wouldn't be good, because platforms are regularly added.

But perhaps this is theoretical and `hard` is the only actual case?

dpgeorge, 2021-07-05:
> On platforms which support hard IRQ's it seems logical to use them as the ISR code is fast and non-allocating.

Yes, I suspected that was the reason, to support both hard/non-hard.

> My concern is the general case where more than one arg is only supported on a subset of platforms. The `TypeError` does not indicate which arg(s) are causing the problem

Yes I understand.  We could improve the error message to print (at least one) of the extra args given that are not supported.

> But perhaps this is theoretical and `hard` is the only actual case?

Perhaps... more feedback from users would be needed to know.  In the end I guess the way to minimise the issue is to minimise the number of additional args that any given port can add.  Ideally they shouldn't add any and maybe that is a good goal, except for constructors which can have some differences among ports.

jimmo, 2021-07-06:
> On platforms like ESP32 with SPIRAM the interrupt latency means that sometimes changes may be missed - this seems unavoidable. On platforms which support hard IRQ's it seems logical to use them as the ISR code is fast and non-allocating.

@peterhinch 

At some stage I'd love to move modmachine entirely into extmod (i.e. same machine.X bindings for all ports), and doing this will require solving how to deal with this problem more generally. The two obvious options are:

a) Use preprocessor macros for a port to configure which "features" are available in the bindings (in this case `MP_MACHINE_PIN_SUPPORTS_HARD_IRQ`), which would gate the kwarg being available.

b) Make the kwarg always available, but raise a "not implemented" error at the port-specific implementation if you attempt to use it.

In this particular case, it sounds like you're saying is that you want "prefer hard, but ok to fall back to soft". But I'm a bit nervous about that in general.

> In this case determining how to proceed would involve an unwieldy trial-and-error process. A lookup based on a table of possible platforms wouldn't be good, because platforms are regularly added.

Agreed! But it's true that only the driver really knows how to proceed. What if there was a way for a driver to ask about capabilities?

So perhaps a third option:

c) Replace all bool fields in constructors / init (or perhaps everywhere in the machine API) with enums:
 - No
 - Yes
 - Prefer yes, ok to fall back to no
 (where unset now means the third option)

Now the driver can be explicit about what it needs, but because of bool<->int conversion, the API is backwards compatible.

dpgeorge, 2021-07-06:
> At some stage I'd love to move modmachine entirely into extmod

See related #7450, which does this for `machine.PWM`.

peterhinch, 2021-07-06:
> In this particular case, it sounds like you're saying is that you want "prefer hard, but ok to fall back to soft". But I'm a bit nervous about that in general.

No. I'm entirely happy with this particular driver trapping the error and handling it. In some cases the correct response would not be to fall back to soft, but to raise a "sorry mate, not on this platform you can't" exception. Or something else - perhaps fall back to a blocking mode using polling.

My concern is for the general case: ensuring that there is a sensible way for a user written device driver to determine the capabilities available.

> But it's true that only the driver really knows how to proceed. What if there was a way for a driver to ask about capabilities?

Exactly. Or ensure that a thrown exception provides enough information to enable the driver to figure out what went wrong.

I'm unsure about your enum idea. It depends on the number of cases exist where fallback actually makes sense, and whether the issue of unsupported args is restricted to boolean constructor args. I can't claim a wide enough knowledge of **machine** to make this judgement.

Maybe some cases apply to numeric args where fallback might be appropriate - but the driver needs to be able to determine whether this has occurred.

Along with https://github.com/micropython/micropython/issues/5564 this is a plea for a design approach where a rational approach to writing portable applications is provided.

jimmo, 2021-07-06:
@peterhinch OK, thanks, I see what you're saying... it's not enough to fall back, you also might need to know that it's fallen back.

If capabilities were to be implemented it would be great if it could be evaluated as a constant expression (so that the compiler could eliminate unused code)...

peterhinch, 2021-07-06:
@jimmo That would be fine. I'd also be happy with no automatic fallback so long as the user code could determine exactly which arg was causing the problem and take appropriate action. This might be a fallback or something quite different.