micropython/micropython #7476: RP2 board specific I2C macros

mcauser, 2021-07-01:
Should we add board specific macro support to the rp2 machine_i2c.c:
https://github.com/micropython/micropython/blob/master/ports/rp2/machine_i2c.c#L36

Like we did for esp32:
https://github.com/micropython/micropython/blob/master/ports/esp32/machine_i2c.c#L35

So rp2 board defs can set preferred I2C pins:
```
#define MICROPY_HW_I2C0_SCL (12)
#define MICROPY_HW_I2C0_SDA (13)

#define MICROPY_HW_I2C1_SCL (14)
#define MICROPY_HW_I2C1_SDA (15)
```

eg. https://github.com/micropython/micropython/blob/master/ports/esp32/boards/UM_TINYPICO/mpconfigboard.h#L4

So you can construct an I2C instance without having to always specify pins:
```
i2c0 = I2C(0)
i2c0 = I2C(0, scl=Pin(12), sda=Pin(13))
```

mcauser, 2021-07-01:
On it.

kevindawson, 2021-07-06:
@mcauser why would you not use the default pins for `i2c0`**?**

[Pico pin_out](https://datasheets.raspberrypi.org/pico/Pico-R3-A4-Pinout.pdf)

`i2c0 = I2C(0, scl=Pin(4), sda=Pin(5))`

`i2c1` is cool ;)

mcauser, 2021-07-07:
@kevindawson I'm not recommending changing any existing defaults, just allowing at a board level macros so makers can define their own boards with their favourite pins to match onboard peripherals.

_Macro all the things!_

kevindawson, 2021-07-07:
@mcauser The point I was trying to make was that the 

> Pico Pin out

 is using different defaults to 

>  machine_i2c.c#L36 

they can not both be correct

They should be the **same**

wherever you ask this question the answer should be the **same**.

which one do you think young python coders will find...

jimmo, 2021-07-08:
@kevindawson I agree this is confusing. Unfortunately I think this is a historical accident (@dpgeorge might be able to confirm) -- the MicroPython defaults were chosen (last year) before Raspberry Pi decided what the defaults were (looks like this happened in March this year) and subsequently updated the pinout diagrams (I have an older one here that doesn't have the defaults).

Changing this would be a breaking change for code currently using `I2C(0)`, not to mention it would flow on to the defaults for UART and SPI (i.e. UART1 currently defaults to pins 4 & 5, which is now the I2C0 default).

What @mcauser is proposing is independent of this though -- it just allows boards that embed the RP2040 directly (and don't care at all about the Pico) to specify their default.

If we decide in the future to update the defaults for the Pico board (using the newly added defaults in the SDK in boards/pico.h), then boards/PICO/mpconfigboard.h is the right place to do it (so it doesn't affect the other board definitions) and so this change is a good step towards that.

dpgeorge, 2021-07-08:
The "Raspberry Pi Pico Python SDK" document has I2C defaults which match what is in the code in `machine_i2c.c`.  So they can't really change.  See section 3.6

kevindawson, 2021-07-08:
Thank You.

- @jimmo 
- @dpgeorge 

I will go and Knock on a **Raspberry Pi Pico Door**

mcauser, 2021-07-08:
UART macros in: https://github.com/micropython/micropython/pull/7503

closed 2021-07-19 by mcauser