micropython/micropython #7479: USB transmission gradually slowing down (in sawtooth pattern)

FilipDominec, 2021-07-01:
I am trying to (mis)use Rpi Pico as a scientific data acquisition platform, imagine a 500 kSpS cheap oscilloscope. It is easy to set up communication with μPython: writing data to stdout on the Rpi Pico, and reading data from /dev/ttyACM0 (or other quasi-serial port)  in the computer. 

Its direct USB support would promise decent speed up to 1 MB/s, which would be just fine for the task. In practice, there are tighter bottlenecks described below. I made a little script for Rpi Pico, flooding the USB with data, and measuring the speed.

My code for μPyhon in Rpi Pico:

    from machine import Pin
    import time, machine, usys, gc, micropython 
    machine.freq(125000000*2)

    ctrl = machine.Pin(25, machine.Pin.OUT)
    ctrl.value(1); time.sleep(.5); ctrl.value(0) # no resets detected

    while True:
        for a in range(1000):
            try:
                usys.stdout.buffer.write('x'*10000)  
            except:
                ctrl.value(1); time.sleep(.2); ctrl.value(0) # no errors detected
        time.sleep(1) # try allowing USB stack to flush (no effect on later speed)

My code to receive the data in the computer:

    #!/usr/bin/python3  
    import sys, serial, time
    port = serial.Serial('/dev/ttyACM0') 
    c = 0
    while True: 
        t0=time.time()
        raw = port.read(10000)
        t1= time.time()
        tooktime = t1-t0
        print(f'{c:5d} receiving 10 kB now took', tooktime )
        c+=1

**First observation** is that on Rpi Pico clocked at 250 MHz, one has to avoid using ```print()``` as it is slow (some 30−50 kB/s). Calling ```usys.stdout.write()``` is better (up to 200 kB/s), but ```usys.stdout.buffer.write()``` can initially transmit 10kB in 0.015 s, getting close to the USB "Full Speed" throughput limit (i.e. ~600 kB/s). So I use the latter function for data. 

**Observation 2**: The trouble is that this speed drops continously - after few seconds of transmission, transmitting 10kB takes 0.050 s (i.e. only 200 kB/s). Then suddenly the transmit time returns to its initial value and starts growing linearly. This repeats with the transmit times forming a sawtooth pattern.

Here I paste part of the printout on my computer where the drop in transmit time happened:

  116 receiving 10 kB now took 0.051924705505371094
  117 receiving 10 kB now took 0.05239605903625488
  118 receiving 10 kB now took 0.05241680145263672
  119 receiving 10 kB now took 0.05560755729675293
  120 receiving 10 kB now took 0.04527783393859863
  121 receiving 10 kB now took 0.015183687210083008
  122 receiving 10 kB now took 0.015230655670166016
  123 receiving 10 kB now took 0.01515960693359375
  124 receiving 10 kB now took 0.015021562576293945
  125 receiving 10 kB now took 0.015263795852661133

Complementary observations: 

**2a)** Without overclocking @125 MHz, the transmit speeds are roughly half of what I wrote above, so the Rpi Pico is the bottleneck here. 

**2b)** When transmit speed goes high, I thought the chip resets, but it is not the case (LED does not blink). 

**2c)** If I put a random ```time.sleep(1)``` after some hundreds of transmission cycles, the transmit speed does not change much after the delay and continues to grow steadily when delay expires (so it is probably not a USB buffer error).

**2d)** Between transmitting cycles, I observed no significant changes in the ```micropython.mem_info()``` printout. So it is probably not a memory issue either.

**2e)** The period of main transmission loop is proportional to the time required to receive data in computer (as visible on oscilloscope). So no packets are probably lost. 

Achieving fast USB transmission is a generally useful task and micropython on Rpi Pico does surprisingly well. At least initially. Can anybody give me a hint to get constantly high USB throughput, please?

veni-vidi-code, 2021-07-01:
Have you measured the temperatur of your pico while doing so? Is it more steady when not overclocked?

FilipDominec, 2021-07-01:
@veni-vidi-code Thanks for the tip, but surface temperature is well below 35 °C all the time. No throttling expected, even when overclocking.

FilipDominec, 2021-07-01:
I could also plot the sawtooth pattern... 

[image]

robert-hh, 2021-07-01:
One critical line slowing down the process is this one:
```
                usys.stdout.buffer.write('x'*10000)  
```
It creates the data object to be sent every time it is called. That will lead to garbage collectikon after a few iterations. It is more effective to create that data buffer beforehand. like:
```
    data = b'x'*10000
    while True:
        for a in range(1000):
            try:
                usys.stdout.buffer.write(data)  
            except:
                LED.value(1); time.sleep(.2); LED.value(0) # no errors detected
        time.sleep(1) # try allowing USB stack to flush (no effect on later speed)

FilipDominec, 2021-07-01:
@robert-hh Thanks, this improved speed by few %. 

Interestingly, the speed cycling occurs at the same period with regards to amount of data. I am now quite confident the cycling occurs at each 418 chunks, i.e. 4180 kB transmitted. 

[image]

robert-hh, 2021-07-01:
At least there seems not much difference between sys.stdout and sys.stdout.buffer any more, unless you changed the plotting style. Can you test, whether these delays are caused by the RP2 or by your PC?

FilipDominec, 2021-07-01:
@robert-hh I changed the plotting style. In this second picture, both datasets use ```sys.stdout.buffer```, but the red is with a string prepared beforehand as you suggest. ```sys.stdout.write()``` is rather slow and I don't try it any more. 

[image]

The delays are roughly doubled when Rpi runs at 125 MHz. So I guess the bottleneck is somewhere in Rpi - not sure whether it is micropython or the USB interface...

FilipDominec, 2021-07-01:
I also noticed that the transmission speed does not depend on the size of ```data```, be it 100 bytes or 10 kB. (Naturally there is some limit that entirely clogs USB of Rpi Pico and requires ```flash_nuke.uf2```, so I didn't test larger chunks than 50 kB.)

dpgeorge, 2021-07-02:
It might be worth trying this with the latest (stable) pico-sdk and tinyusb, see #7474.

FilipDominec, 2021-07-02:
@dpgeorge I agree - either I can confirm the bug persists in latest version (which is good), or I find that it is no more there (which is even better). So far I used ```rp2-pico-20210618-v1.16.uf2```, but on the [download page](https://micropython.org/download/rp2-pico/) I can see there is a fresh version. Cannot wait to try it out!

robert-hh, 2021-07-02:
No change here with MicroPython v1.16-48-g0b3332c8e. Times between 0.0126 and 0.084 seconds.

FilipDominec, 2021-07-02:
@robert-hh I can confirm, no change whatsoever. 

[image]

Why is the sawtooth repeated each 418 or 419 chunks? Why are there 10 steps? 

Note these curves were measured today on a different computer, and they sligtly differ in shape from the previous ones.

FilipDominec, 2021-07-02:
I also changed the computer script so that it receives 1kB chunks instead of 10kB. To make it comparable, I normalized the time to 10kB also in this case. 

We can see finer structure of the delays, perhaps some mismatch of RP2 and computer loops. But the general trend of transmission slowing down in still obvious. 

[image]

lurch, 2021-07-02:
> I am trying to (mis)use Rpi Pico as a scientific data acquisition platform, imagine a 500 kSpS cheap oscilloscope.

Have you looked at https://github.com/raspberrypi/pico-examples/tree/master/pio/logic_analyser ?
Using a compiled language (like C) instead of an interpreted language (like MicroPython) will be faster, and you should also see more "consistent" performance, as there's no memory garbage-collection being done in the background.

FilipDominec, 2021-07-02:
@lurch Sure I did. It would be interesting to see if the problem persists - AFAIK the logic analyser uses the ```tinyusb``` for communication, doesn't it?

<rant>
There are many options to do the job. Every project is a tradeoff between spending time to try out different ways, and between wasting time to fix one chosen way. This week I thought micropython on RP2 could be the right choice. The previous two weeks I wasted time almost finishing my project in compiled C for ESP32 -- only to find out its USB communication is slow (limited to 921600 baud) and not reliable (spitting random 0xFF bytes in the data stream). One should never give up.  
</rant>

lurch, 2021-07-02:
I'm sure @kilograham and @hathach would be very appreciative of any stress-testing you can give to the USB stack :grinning: 

And as a bonus, you get to narrow down whether the problem is with MicroPython or with TinyUSB :wink: 
Hopefully at least _some_ of your C code that you wrote for ESP32 will be usable (with tweaks) on RP2040? Good luck.

FilipDominec, 2021-07-02:
Now I must admit that the problem seems to be beyond my hacking skills, and beyond my time dedicated for making the RP2/computer communication fast. 

Anyway, initial data rate is 10kB in 0.012 s, that is, 830 kB/s. This is very promising, but after transmitting 320 kB, the speed starts dropping as shown above. This must be some software trouble and I still believe more skilled coders will be able to find its cause. 

While currently I cannot dig through micropython + TinyUSB codebase, I have few ideas that might inspire somebody:
 
 * Since we observe some unknown event that makes the speed high again every 4180 kB transmitted, couldn't this mysterious event be activated more often? For instance, is there an equivalent of ```sys.stdout.flush()``` in micropython? (I observed ```sys.stdout.close()``` after ```write()``` changed nothing.)
 * Similar issue is mentioned here: https://github.com/hathach/tinyusb/issues/920 Is the call to ```usys.stdout.buffer.write()``` making any data copy?
 * Does the slowdown manifest itself in the TinyUSB speed tests proposed here: https://github.com/hathach/tinyusb/issues/189 ?

dhalbert, 2021-07-02:
I have seen things run slower due to the heap filling up and finding a free block taking longer. Once garbage collection happens, the cycle starts over. Here is a non-USB example of an analogous situation: https://github.com/adafruit/circuitpython/issues/2602.

FilipDominec, 2021-07-02:
@dhalbert I also guessed this will be the correct explanation, but ...
 * it is not clear to me why speed cycling occurs each 4180 kB transmitted (i.e. 20× more data than the RAM in RP2 is),
 * why would any extra object be allocated in the cycle at all,
 * why calling ```gc.collect()``` in each cycle has no effect on the speed cycling (only slows the transmission down by a constant, as I checked now).
 
Aside of this, it is rather likely (?) that the cycle in my example allocates no extra object on heap explicitly, this garbage would have to be created somewhere in the implementation of ```usys.stdout.buffer.write()```. Is it correct?

FilipDominec, 2021-07-03:
I experimented further, changing the size of ```data``` down to few bytes. Like this. 

```python
from machine import Pin
import time, machine, usys, gc, micropython 
machine.freq(125000000*2)

data = b'x'*14
buf = usys.stdout.buffer
while True:
    buf.write(data)  
```

Naturally transmitting in such short bursts slows the transmission down due to overhead of the program loop. What interested me more is its impact on the sawtooth function. 

1. Transmitting byte arrays of 13bytes or shorter makes the sawtooth repeat more frequently. 
2. Transmitting byte arrays between 14 and some 30 bytes shows a very similar sawtooth as previously used large chunks of 10kB, but there are no more steps.

[image]

Can this inspire somebody?

lurch, 2021-07-03:
@FilipDominec Do you have an FTDI232 or PL2303 or similar USB-UART adaptor? Might be interesting to repeat your experiment, but sending the data over the RP2040's UART instead of USB? Obviously the maximum data-rate will be _much_ reduced, but it would be interesting to see if it stays constant or shows a similar slowdown over time?

robert-hh, 2021-07-03:
Using UART, the pattern looks different- For a 10000 bytew block at 921600 baud it oscillates between 0,10767 and  0,1131 seconds.

[image]

FilipDominec, 2021-07-03:
Using [online tips](https://www.tomshardware.com/how-to/program-raspberry-pi-pico-with-arduino-ide) and getting inspired from the "Adafruit TinyUSB/CDC/cdc_multi" example in Arduino studio, it is not hard to flood TinyUSB from C, either:

```C
#include <Adafruit_TinyUSB.h>
#include "pico/stdlib.h"

void setup() {
  set_sys_clock_khz(250000, false);
  Serial.begin(115200);
}

void loop() {
  Serial.printf("xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx");
}
```

And the result is beatiful, sustained throughput at 770 kB/s for this simple C program:

[image]

So there is either a bug in micropython, its interface to TinyUSB, or in the way I am using it to transmit data to computer.

tjvr, 2021-07-03:
Does MicroPython have any profiling tools we could use to figure out where it’s spending time?

hathach, 2021-07-04:
Could you try to increase `CFG_TUD_CDC_EP_BUFSIZE` to larger value such as 128/256 to see if it improves the speed. The latest tinyusb has also implemented USB double buffered for rp2040 (IN only) so it may help as well https://github.com/hathach/tinyusb/pull/891 . There is more approach mentioned in my github issue, though, it may need to take a bit of time to write it in portable way to work with all mcus (therefore takes time).

Note CFG_TUD_CDC_RX_BUFSIZE/CFG_TUD_CDC_TX_BUFSIZE` may also need to be increased as well.

robert-hh, 2021-07-04:
`CFG_TUD_CDC_RX_BUFSIZE/CFG_TUD_CDC_TX_BUFSIZE` were already at 256. Setting `CFG_TUD_CDC_EP_BUFSIZE`to 256 as well improves the speed, in that the 10kB times then vary between 0.09 and 0.19 seconds. Setting all buffer sizes to 512 moves the range further down to 0.08 to 0.13 seconds.
The PR you mention seems not merged into MicroPyhton.

dpgeorge, 2021-07-05:
Using the Pico & PC code in the first post of this issue, I was able to reproduce the "saw tooth" plots.

Using PR #7474 seems to fix the problem completely, I can get a sustained ~7Mbits/sec transfer rate from Pico to PC.  I'll merge that PR shortly.

hathach, 2021-07-05:
> Using the Pico & PC code in the first post of this issue, I was able to reproduce the "saw tooth" plots.
> 
> Using PR #7474 seems to fix the problem completely, I can get a sustained ~7Mbits/sec transfer rate from Pico to PC. I'll merge that PR shortly.

That is great to hear :tada: :tada:

dpgeorge, 2021-07-05:
> That is great to hear

I'm not sure where the issue was and what the fix was but it should be somewhere in the pico-sdk and-or tinyusb.  I don't have time to do a git-bisect on these to find out :slightly_smiling_face:

hathach, 2021-07-05:
yeah, to be honest, tinyusb isn't speed optimized at the moment. Since I am more focused on the portability. I am glad that micropython could pull off a great throughput 7Mbps which will give me a lot more time to work on the speed feature :)

robert-hh, 2021-07-05:
Data rate confirmed. Setting CFG_TUD_CDC_EP_BUFSIZE to 256 as suggested by @hathach increases the data rate further by 35% to about 9.5 Mbit/sec.

dpgeorge, 2021-07-05:
> Setting CFG_TUD_CDC_EP_BUFSIZE to 256 as suggested by @hathach increases the data rate further by 35% to about 9.5 Mbit/sec.

Since the rp2040 has quite a bit of RAM it's probably worth making this the default (or at least bump it to 128).

dpgeorge, 2021-07-05:
> > Setting CFG_TUD_CDC_EP_BUFSIZE to 256 as suggested by @hathach increases the data rate further by 35% to about 9.5 Mbit/sec.
> 
> Since the rp2040 has quite a bit of RAM it's probably worth making this the default (or at least bump it to 128).

Actually, testing this (using 256 for CFG_TUD_CDC_EP_BUFSIZE) shows no difference in data rate (for Pico -> PC) for me.  @robert-hh what test script to you use to get 9.5Mbit/sec?

robert-hh, 2021-07-05:
I'm using stripped down versions of the initial script:
Send:
```
import time, machine, usys
machine.freq(250_000_000)

data = b'x'*10000
while True:
    for a in range(1000):
        usys.stdout.buffer.write(data)  
```
Receive:
```
#!/usr/bin/python3  
import sys, serial, time
port = serial.Serial('/dev/ttyACM0')
c = 0
while True: 
    t0=time.time()
    raw = port.read(10000)
    t1= time.time()
    tooktime = t1-t0
    print(f'{c:5d} receiving 10 kB now took', tooktime )
    c+=1
```

FilipDominec, 2021-07-05:
Good job, everybody! The remaining task is to provide micropython newbies (like me) an official tutorial for interfacing with the computer over USB. As a matter of fact, the popular and seemingly preferred way, ```printf()```, is lamentably slow. 

Remember the words of a prophet:

    There should be one-- and preferably only one --obvious way to do it.
    Explicit is better than implicit.
    In the face of ambiguity, refuse the temptation to guess.

dpgeorge, 2021-07-06:
> I'm using stripped down versions of the initial script:

Ok, thanks.  Using that I can also achieve around 9MBit/sec (with 256 endpoint size).  And I get a similar rate using the same test on a PYBD-SF2.

dpgeorge, 2021-07-06:
> As a matter of fact, the popular and seemingly preferred way, `printf()`, is lamentably slow.

`print()` was slow because it was not optimised.  #7491 makes it much faster, almost as fast as `sys.stdout.buffer.write`.

But note that `print()` is only going to be fast for printing strings (ascii and/or utf8).  If you need to print raw byte data then `sys.stdout.buffer.write` is the way to do it (even in CPython).

dpgeorge, 2021-07-06:
> Since the rp2040 has quite a bit of RAM it's probably worth making this the default

See #7492 for this.

lurch, 2021-07-06:
> But note that print() is only going to be fast for printing strings (ascii and/or utf8). If you need to print raw byte data then sys.stdout.buffer.write is the way to do it (even in CPython).

There's also the obvious difference that `print` has to do any variable interpolations that might be necessary (e.g. `print("Value is %d" % value)`) whereas `sys.stdout.buffer.write` can just squirt out the raw data as fast as possible without having to check for special-characters first.

But good to hear that you've managed to speed up `print()` anyway :slightly_smiling_face:

dpgeorge, 2021-07-07:
> There's also the obvious difference that `print` has to do any variable interpolations that might be necessary

That's not technically true: `print()` itself doesn't do the interpolations, it's just passed a pre-formatted string.  And one could actually do `sys.stdout.buffer.write(b"%d" % 123)`.  The main difference is that `print()` renders objects (which for strings is converting `\n` to `\r\n`) while `sys.stdout.buffer.write()` just writes out a bytes object.

lurch, 2021-07-07:
Ahh, I must have been muddling up Python's `print` with C's `printf` :wink:

dpgeorge, 2021-07-08:
This issues should be fully resolved now.  See related commits 20a6044952281a4f55a5ee83bcaca88ec1b870f9 and 0613d3e3561a82fffa36594647ef916b19bc912b

closed 2021-07-08 by dpgeorge

derdudee, 2022-08-16:
Is there also a possibility to use this in C?
printf is too slow for me

lurch, 2022-08-16:
@derdudee For questions regarding the C SDK, you ought to be asking at https://github.com/raspberrypi/pico-sdk/issues

FilipDominec, 2022-08-16:
@derdudee Without questioning the note of lurch, I would like to point you to https://github.com/FilipDominec/rp2daq/blob/main/rp2daq.c#L157 where I commented the issue. Do not forget to set ```stdio_set_translate_crlf(&stdio_usb, false);``` to transmit binary data correctly.

0i9j, 2022-08-18:
@FilipDominec Are you able to read data from you COM Port? Or where is fwrite printing to?