micropython/micropython #7480: 'mpremote help' would be nice

lurch, 2021-07-01:
I just pip-installed mpremote and was surprised to see:
```
$ mpremote --help
mpremote: '--help' is not a command
$ mpremote -h
mpremote: '-h' is not a command
$ mpremote h
mpremote: 'h' is not a command
$ mpremote help
mpremote: 'help' is not a command
```

Simply printing out the
```
mpremote connect <device>        -- connect to given device
                                    device may be: list, auto, id:x, port:x
                                    or any valid device name/path
mpremote disconnect              -- disconnect current device
mpremote mount <local-dir>       -- mount local directory on device
mpremote eval <string>           -- evaluate and print the string
mpremote exec <string>           -- execute the string
mpremote run <file>              -- run the given local script
mpremote fs <command> <args...>  -- execute filesystem commands on the device
                                    command may be: cat, ls, cp, rm, mkdir, rmdir
                                    use ":" as a prefix to specify a file on the device
mpremote repl                    -- enter REPL
                                    options:
                                        --capture <file>
                                        --inject-code <string>
                                        --inject-file <file>
```
text (which I've just copied from https://github.com/micropython/micropython/blob/master/tools/mpremote/README.md ) would probably be sufficient?

Also, looking at the command-list above, perhaps it makes sense for `mpremote list` to do the same thing as `mpremote connect list` ?

dpgeorge, 2021-07-02:
Yes, help text would be great.  The above sounds good to me.

> Also, looking at the command-list above, perhaps it makes sense for `mpremote list` to do the same thing as `mpremote connect list` ?

I have an alias in my `.config/mpremote/config.py` file for exactly this:
```python
commands = {
    "list": "connect list",
    ...
}
```

lurch, 2021-07-02:
> I have an alias in my .config/mpremote/config.py file for exactly this:

I'd argue that it's probably worth adding that directly to https://github.com/micropython/micropython/blob/master/tools/mpremote/mpremote/main.py#L28 ?

dpgeorge, 2021-07-02:
> I'd argue that it's probably worth adding that directly to

Yes, good idea.  The only concern I'd have is that it's very close to "ls" which lists files on the target board... maybe there's a better alias for "connect list" that's not as similar?

lurch, 2021-07-02:
Yeah, I thought of that but I think people with a command-line background will be familiar enough with `ls` to know the difference? (and luckily `list` and `ls` are both "read-only" commands, i.e. no side-effects if used accidentally)
You could add something like `mpremote list_connections` or `mpremote list_devices` but that's then not any easier to type than `mpremote connect list` :wink:

davehylands, 2021-07-02:
Since `connect list` shows you devices you can connect to, you could use the command `devices` to list the devices.

dpgeorge, 2021-07-05:
> Since `connect list` shows you devices you can connect to, you could use the command `devices` to list the devices.

How about just `devs` for short?  So:
```
$ mpremote devs
```
which is a shortcut for `connect list`.

lurch, 2021-07-05:
Seems reasonable. As long as people wouldn't expect:
```
$ mpremote devs
@dpgeorge and others
```
:wink:

dpgeorge, 2021-08-12:
See #7647 to add a "devs" shortcut.

dpgeorge, 2021-12-29:
A help command was added in 74e995dfd26c7c9304392b2073ec4558173b699c

closed 2021-12-29 by dpgeorge