micropython/micropython #7481: mpremote lists VID and PID backwards?

lurch, 2021-07-01:
I just noticed that `mpremote connect list` seemed to be displaying the PID _before_ the VID, so I had a look at the source and saw that https://github.com/micropython/micropython/blob/master/tools/mpremote/mpremote/main.py#L169 says
```python
                        p.device, p.serial_number, p.pid, p.vid, p.manufacturer, p.product
```
which seems a bit odd, as in my experience the USB VID (vendor id) is **always** displayed before the PID (product id).
I won't submit a PR though, in case this was a deliberate decision rather than just a typo?

dpgeorge, 2021-07-02:
I tried to make the output match `lsusb`, so this is indeed a typo :)

dpgeorge, 2021-07-02:
Fixed by d1decdfa937368bc83adf6829eba6743d75136d0

closed 2021-07-02 by dpgeorge