micropython/micropython #7485: mpremote with mount fails on ESP32

peterhinch, 2021-07-04:
Re-installed mpremote today - tail end of pip3:
```
Successfully installed mpremote-0.0.5
```
ESP32 works OK, with ctrl-d resetting the ESP uneventfully if there is no mount.

With a mount, ctrl-d kills mpremote. In the following, when the REPL appeared I pressed ctrl-d:
```python
>>> [adminpete@capybara]: ~/temp
$ mpremote mount .
Local directory . is mounted at /remote
Connected to MicroPython at /dev/ttyUSB0
Use Ctrl-] to exit this shell
>
MicroPython v1.16 on 2021-06-28; ESP32 module with ESP32
Type "help()" for more information.
>>> 
MPY: soft reboot
Traceback (most recent call last):
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 452, in main
    do_repl(pyb, args)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 332, in do_repl
    do_repl_main_loop(
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 259, in do_repl_main_loop
    pyb.soft_reset_with_mount(console_out_write)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboardextended.py", line 612, in soft_reset_with_mount
    self.exec_(fs_hook_code)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 465, in exec_
    ret, ret_err = self.exec_raw(command, data_consumer=data_consumer)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 456, in exec_raw
    self.exec_raw_no_follow(command)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 453, in exec_raw_no_follow
    raise PyboardError("could not exec command (response: %r)" % data)
mpremote.pyboard.PyboardError: could not exec command (response: b'R\x01')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/adminpete/.local/bin/mpremote", line 8, in <module>
    sys.exit(main())
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 455, in main
    do_disconnect(pyb)
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/main.py", line 208, in do_disconnect
    pyb.umount_local()
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboardextended.py", line 620, in umount_local
    self.exec_('uos.umount("/remote")')
  File "/home/adminpete/.local/lib/python3.8/site-packages/mpremote/pyboard.py", line 467, in exec_
    raise PyboardError("exception", ret, ret_err)
mpremote.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nOSError: [Errno 22] EINVAL\r\n')
```

dpgeorge, 2021-07-05:
I have seen this issue before (on esp32) but can no longer reproduce it... using current mpremote (not from pip, but also tried that).  Maybe it has something to do with boot.py/main.py on the device?

peterhinch, 2021-07-05:
The files `boot.py` and `main.py` both exist but all lines are commented out with `#` at the LHS. Hardware is a TTGO T_Display which is an ESP32 without SPIRAM. Firmware is
```python
MicroPython v1.16 on 2021-06-28; ESP32 module with ESP32
```
The fault is consistent on that board. However I have just installed the same firmware on the reference board, ensured it has similar null `boot.py` and `main.py` files and I can't reproduce it. So this is a puzzler.

jdtsmith, 2021-09-01:
FYI, my ESP32 board with the crash-on-soft-boot-when-FS-mounted issue is also a TTGO T-DISPLAY.

andrewleech, 2022-02-04:
Ah, this traceback looks the same as an issue I just fixed the other day https://github.com/micropython/micropython/pull/8240

The inter-byte timeout during flush wasn't quite long enough during the boot after soft reset, it stopped writing for data then before the board was actually ready.

I would see it every time on one of my esp32 boards, but it never occurred on a second one which had an external different brand usb-serial.

dpgeorge, 2022-02-07:
This is most likely fixed by a combination of 5a86d8dc424b5923d5d1a7c1352378ef328847d6 and fecfd5269632e86f86c6b31a9bc2ec22876f8934, so I'll close the issue.  Please reopen/comment if the issue crops up again.

closed 2022-02-07 by dpgeorge